fwupd/firmware-lenovo #156: Lenovo T470 not showing correct Intel ME Version

Roxxor91, 2021-09-22:
**Describe the bug**
fwupd shows that only 184.70.3626 is installed. I installed 184.86.3909 without issues via fwupd. The BIOS shows 184.86.3909. All other firmwares are correctly displayed and also updated.

**Steps to Reproduce**
Upgrade Intel ME, shown version stays the same

**Expected behavior**
184.86.3909 is installed and should displayed accordingly.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
client version:	1.5.7
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.7

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
apt, debian testing

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
20HES01100
                     get-devices --show-all-devices
20HES01100
│
├─Embedded Controller:
│     Device ID:          efc88c8524a5d0ada629ba38b89ac722dc3d5efb
│     Current version:    0.1.36
│     Minimum Version:    0.0.1
│     Vendor:             DMI:LENOVO
│     GUIDs:              4b37c365-ecb2-4d89-8e15-e93c27c6ce12
│                         897f3781-238c-56a0-95fa-e2612102db0e
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─HD Graphics 620:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    02
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              164325d0-cc40-51e0-9935-b1fabe2a686b
│                         943b8145-a3aa-5e38-9a26-6f3dd2bb1f0f
│                         2886e312-afd8-5510-8993-12d568a85e00
│                         38f6c009-a25c-5b50-b3b5-fef4f9c6b846
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─INTEL SSDPEKNW020T8:
│     Device ID:          c82980cd3fe66f7d0a8bb83b353cbc271177efd4
│     Zusammenfassung:    NVM Express Solid State Drive
│     Current version:    002C
│     Vendor:             Intel Corporation (NVME:0x8086)
│     GUIDs:              c5fe8b70-dc9a-5c3b-9634-659091d29812
│                         1122104f-b10a-5f32-bc13-7a1ac0f52ea2
│                         c6cd9ab0-8f20-512e-9e1f-1af55b8454b9
│                         82741c78-f5dc-5c23-a152-00de5799edc8
│                         7c0505be-b609-566e-bf94-561e441a9171
│     Device Flags:       • Internal device
│                         • Updatable
[... 429 lines cut ...]
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initrd.img-5.10.0-7-amd64
├── initrd.img-5.10.0-8-amd64
├── lost+found [error opening dir]
├── System.map-5.10.0-7-amd64
├── System.map-5.10.0-8-amd64
├── vmlinuz-5.10.0-7-amd64
└── vmlinuz-5.10.0-8-amd64

```

**Additional questions**

- Operating system and version: Debian Testing
- Have you tried rebooting? Yes
- Is this a regression? No
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? I guess not.

hughsie, 2021-09-22:
@kmauleon @mrhpearson do we know what/where actually updates the value in the ESRT table for the ME firmware? This seems to be something we keep getting wrong, and maybe it's something I can verify at LVFS upload time?

kmauleon, 2021-09-22:
hi @hughsie ESRT table is managed by BIOS but BIOS dev does not want to further investigate unless the issue is recreated at Lenovo.  sorry i've been trying to recreate the issue myself for months now but still unlucky T_T

hughsie, 2021-09-22:
@kmauleon can you expand a bit (over email is fine if this is too public) -- e.g. how does the system bios look at the ME file system? Or does the system bios have some kind of map like:

    if me_checksum == "8dfa3c58ae9bd685152dff928b2981a7"
         set esrt_version = "1.2.4.5"
    elif me_checksum == "bdb4d43b8478bd927615e5022693c940"
         set esrt_version = "1.2.4.6"

If that's what happens then that's fine, but we need to put a requirement on the ME firmware to depend on the system firmware that knows "what it is".

Roxxor91, 2021-09-22:
If I can test/help something, let me know! Right now I do not know what you are talking about :D

kmauleon, 2022-11-16:
Closing this as BIOS ESRT fix has been rolled out already.  Please reopen if you still see the problem.  Thank you very much.

closed 2022-11-16 by kmauleon