fwupd/firmware-lenovo #159: Missing BIOS update (1.53) for X1 Carbon 6th gen

LordChaos73, 2021-10-01:
**Describe the bug**
There seem to be no 1.53 BIOS update available, but it is available on Lenovo's support website

**Steps to Reproduce**
No BIOS update available after running ``fwupdmgr refresh && fwupdmgr get-updates``

**Expected behavior**
A BIOS update should be available

**fwupd version information**
Please provide the version of the daemon and client.

```shell
[ericd]% fwupdmgr --version
client version:	1.6.3
compile-time dependency versions
	gusb:	0.3.7
daemon version:	1.6.3
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Came with the distro (pacman).

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
20KHCTO1WW
│
├─Core™ i7-8550U CPU @ 1.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000ea
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         561403e8-143a-5071-ab09-bf5e4c146983
│                         54bdc429-0179-5019-9cb6-8a343324c97c
│     Device Flags:       • Internal device
│
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUIDs:              3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│                         7090c94a-ee40-593e-91bc-8cfd5a8c0a5d
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    184.86.3909
│     Minimum Version:    184.86.3909
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUIDs:              9c9d9769-32fa-4841-b550-ea998e754e99
│                         1a4eea87-292a-5d99-82bd-fd244d4e0194
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
[... 201 lines cut ...]
                          c528df4b-7972-5880-8cb1-330415e2dc6a
                          137520ce-3603-53e6-9165-56694ed744e7
                          ff2112dc-038c-596d-90ca-d43c5077c6ec
                          63af1441-fb24-5368-b450-886fbc98610b
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

**Additional questions**

- Operating system and version:

Manjaro (Arch Linux)

- Have you tried rebooting?

Yes

- Is this a regression?

No that I'm aware of

LinuxOnTheDesktop, 2021-10-05:
I wonder whether the reason that this update is missing, and this other one (#160) too, is as follows. Other updates for the X1CG6 are shown on fwupd but do not actually install. (See e.g. #142.) That is: perhaps Lenovo is aware of that latter problem and perhaps that awareness is holding Lenovo back. Still, the problems could be distinct.

LordChaos73, 2021-11-25:
Upgraded to 1.54 successfully.

closed 2021-11-25 by LordChaos73