fwupd/firmware-lenovo #160: Missing Thunderbolt firmware updates for ThinkPad X1 Carbon Gen 6 (Machine types: 20KH, 20KG)

johannbg, 2021-10-02:
There are seemingly two missing Thunderbolt updates for the X1 Carbon Gen 6 and it looks like the Thunderbird controller has not recived updates on Linux since version 43 for whatever reason.

One `46` that was released on 2021/03/17
`n23th07w   17.0.0.18     N23TF23W        46   F407.06.31    01    2021/03/17`
and fixes a thunderbolt vulnerability

`Version 17.0.0.18-N23TF23W

[Important updates]
- Fixed Thunderbolt vulnerability.

[New functions or enhancements]
- Intel NVM version update to 46.
- Updated FW update tool version to 17.0.0.18.
- Updated PD FW to F407.06.31

[Problem fixes]
  Nothing.`

And another one `45` that was released on 2020/06/02
`n23th06w   17.0.0.12     N23TF22W        45   F407.06.29    01    2020/06/02`
and most notably fixes docking compatibility issues

[Important updates]
  Nothing.

[New functions or enhancements]
- Updated PD Firmware version to 6.29.

[Problem fixes]
- Fixed dock compatibility.
- Fixed an issue where Thunderbolt Firmware Update tool shows "FILE NOT FOUND"
  error on Turkish language.
`

Linux users that own a Gen 6 and can afford of potentially loose it, can do the following ( as a workaround ) to close the security vulnerability until Lenovo get's their act together and make the firmware update available to Linux users.

1. Extract the Windows EXE file provided by Lenovo via wine, p7zip or whatever floats people boat.
2. dd if=/tmp/Thunderbolt/TPS65988_F407.06.31.bin of=/sys/bus/thunderbolt/devices/0-0/nvm_non_active0/nvmem
3. echo 1 > /sys/bus/thunderbolt/devices/0-0/nvm_authenticate ( the firmware update may take ca 30s to finish, with LBL and flickering screen )

Validation if the firmware flashing was succefull can be seen by running
`cat /sys/bus/thunderbolt/devices/0-0/nvm_version` ( it should say 46.0 ) 

Now one thing worth mentioning is that the [fwupd website](https://fwupd.org/) needs some organasation and improvements in it's on site search engines since it's somewhat pain in the ass finding all firmware updates for a spesific hw type/model from vendors, then having to drill down into the firmware update and then compare it to what's available on Lenovo's website for that model.

mrhpearson, 2021-10-04:
I have a ticket open with the FW teams on missing LVFS updates for TBT FW (it's a number of different platforms)
For my reference: LO-1371
Mark

johannbg, 2021-10-09:
@mrhpearson have you identified the bottleneck already I mean what is the root problem for the missing firmware updates which in somecases are even available on Lenovo's website for the Linux platform. Is it usability issues with LVFS, workflow issues inside Lenovo in uploading them to LVFS, something else?

johannbg, 2021-10-09:
@hughsie does lvfs provide something similar to the update-testing path in Fedora where vendors employees and users can test firmware updates prior to them being "officially" pushed as updates?

johannbg, 2021-10-16:
@mrhpearson I suspected that much since I have had a climpse into some of Lenovo's processes in the past anyways that was few years back and typical internal process workflow should have workflow steps which identifies where the bottleneck is and or steps that prevent you to move the status of the task from open to closed without completing that tasks so you might want to review the workflow in use and have that revisited, re-structured and or identify if it's the same employee(s) doing those mistakes ( which requires them either to be retrained or replaced ), that is if it's within your power to do so.

johannbg, 2021-11-07:
@mrhpearson well it's getting close to a month so any news on this issue. Is the update somehow failing in testing?

johannbg, 2021-12-07:
@mrhpearson another month has passed, any update on this matter?

johannbg, 2022-02-20:
@mrhpearson it's been close to half a year since you opened LO-1371 what's the status of the issue?

amichuda, 2022-03-27:
Want to second this, is there any update on this?

mrhpearson, 2022-04-05:
Sorry for the lack of reply - this is a bit of a mess and has taken a lot of back and forth to figure out.

We got a lot of platforms updated (there are 34 in total we're tracking for the update) - but not all have been done. I still have 8 remaining and X1C6 is one of them (the remaining ones are all of that older generation).

For these older machines, the product requirements around LVFS for the platform teams weren't particularly clear back then. The product FW team were under the understanding they were delivering BIOS and EC updates on LVFS and Thunderbolt wasn't part of their requirements - they weren't funded for it. It's taken quite a bit of back and forth to get to the stage where that was made clear.

I've kicked off an internal request to get TBT added for these remaining platforms. I don't know for sure yet if it will be approved but I hope it will be - I've highlighted that TBT FW updates are critical. The only potential wrinkle is if it's hard to update this FW from Linux for some reason (it doesn't look like it's using UEFI capsules, which makes life a lot easier). I can't make any promises at this point.

My recommendation is (and I know how much this sucks) is to install Windows, get the FW update, and then go back to Linux. I think some folk have had success with WindowsToGo...but I've not tried it myself.

I'll update when I get confirmation on the next steps - but it might take a while based on my previous experiences of this process

hughsie, 2022-04-05:
> if it's hard to update this FW from Linux for some reason

I think it _should_ be easy with the existing thunderbolt plugin; that's at least what the other OEMs are using. Thanks for the update Mark, it's incredibly refreshing to see a vendor be so open talking about all of this commercial stuff.

johannbg, 2022-04-05:
@mrhpearson thanks for your dedication to resolving this issue and sharing the inner processes. I for one would have never imagine that a fw team for a hw vendor would require special funding ( which sound's like it's being outsourced to a third party ). 

Presumably this is standard approach by vendors so it will be interesting to see how this fits EU's right to repair and the new generation of ecodesign rules since I suspect in the future fw will also be part of those. 

An example would be like AMD's Platform Secure Boot (PSB) which tries to provides a hardware root of trust (RoT) and is probably locked to vendor’s firmware signature key, propably violates those RtR/Ecodesign rules and risks the vendor being fined by EU if the vendor adopts and there is not an off switch in the firmware ( but having an off switch defeat's the purpose of having it in the firstplace ).

LinuxOnTheDesktop, 2022-12-18:
> My recommendation is (and I know how much this sucks) is to install Windows, get the FW update, and then go back to Linux. I think some folk have had success with WindowsToGo...but I've not tried it myself.

WindowsToGo no longer exists - or at least is no longer provided by Microsoft. That has been the case for some years now. (I failed to find an exact date on the web.) I think it was already the case when @mrhpearson wrote the sentences that I quoted above.

persicsb, 2022-12-18:
>  I'll update when I get confirmation on the next steps - but it might take a while based on my previous experiences of this process

It seems it will never happen. More than half a year passed, and no updates on this issue. Lenovo has given up on Thinkpad users with Linux?

persicsb, 2025-02-19:
More than 3 years passed, yet nothing changed.

LinuxOnTheDesktop, 2025-02-19:
One thing that has changed is that I gave up on Lenovo - well, I had to: I had multiple motherboards fail on me, one after the other - and bought a laptop by Framework, which has customer support, and Linux support, that is north of abysmal.