fwupd/firmware-lenovo #161: ThinkPad X1 Nano gets stuck on EC 1.42 update

bmreading, 2021-10-04:
Using the latest Fedora 34, I'm currently on version 1.40, and tried to update to the 1.42 version of the embedded controller firmware for my X1 Nano three times. Each time, it reboots the system, and gets stuck on the screen showing the Lenovo logo with white text under it saying "Installing firmware update..."

[image]

The device requires holding the reset button on the underside of the case in order to reboot. From there, it boots normally, but I'm stuck with the unapplied EC update in GNOME Software.

Here is the output of `fwupdmgr get-devices`:
```
20UN0007US
│
├─ELAN0670:00 04F3:3150:
│     Device ID:          e196e632c32c20ba7361f8903f61942e6b96133d
│     Summary:            Elan Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0001
│     Vendor:             HIDRAW:0x04F3
│     GUIDs:              ac0d9b27-06f8-52a6-a92a-af67db424dfd
│                         d29c9b88-4c94-50f0-88c0-33a928b55718
│                         b496e068-7916-55a4-8b89-14880267fc62
│                         15e96dd6-4f5e-5d56-a991-2e63e82d1bd2
│                         2ae001f7-bcfd-5ff2-aec0-8f6294a49d1f
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─Embedded Controller:
│     Device ID:          00109eba01b4e290a438214141e2d26c379f96c0
│     Current version:    0.1.40
│     Minimum Version:    0.1.40
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUIDs:              9cba69ff-4c8b-4398-aede-7c39450174a3
│                         7b748899-8133-5527-b4be-d98bfdf1a8e9
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Integrated Camera:
│     Device ID:          3fa281ddf80d8a06b8ee5d8beb48d38ee95a9627
│     Current version:    58.18
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     GUIDs:              23083b0e-ca90-5dff-82d8-a6a515efb165
│                         30bb6882-12ea-5e20-84a6-29bb797a7b76
│     Device Flags:       • Updatable
│   
├─Prometheus:
│ │   Device ID:          0d5d05911800242bb1f35287012cdcbd9b381148
│ │   Summary:            Fingerprint reader
│ │   Current version:    10.01.3273255
│ │   Vendor:             Synaptics (USB:0x06CB)
│ │   Install Duration:   2 seconds
│ │   GUIDs:              09e8bf16-3e69-50f5-bb66-c7a040248352
│ │                       8088f861-6318-5b1e-9ce4-fbddbedb09ac
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:        73e606488fec47b3e3f9288094f66fded0051446
│       Current version:  0029
│       Minimum Version:  0029
│       Vendor:           Synaptics (USB:0x06CB)
│       GUIDs:            7c5a1e62-38fa-5859-9337-09dbac6377e4
│                         8a12a1dd-5e54-5cb8-9607-c3a4e3e9a7e4
│       Device Flags:     • Updatable
│                         • Supported on remote server
│     
├─System Firmware:
[... 240 lines cut ...]
  │     Lenovo ThinkPad X1NanoGen1 Embedded controller Firmware 1.42
  │     
  │     Some new functionality has been added:
  │     
  │     • Support shutdown system by AC detach. [SP20210616_0003_001]
  │     • Limit input current to 1.8A on T24m-10 and LG 34UC98.
  │     • FnLock LED will not change when Quick Clean Application is running.
  │   
  └─ThinkPad X1NanoGen1:
        New version:      0.1.41
        Remote ID:        lvfs
        Summary:          Lenovo ThinkPad X1NanoGen1 Embedded controller Firmware
        License:          Proprietary
        Size:             767.4 kB
        Created:          2021-07-08
        Urgency:          High
        Vendor:           Lenovo
        Flags:            is-upgrade
        Description:      
        Lenovo ThinkPad X1NanoGen1 Embedded controller Firmware 1.41
        
        This stable release fixes the following issues:
        
        *Update thermal table for ThermalTable_(Zeus)(Ver.03)
```

lsde, 2021-10-04:
I can confirm that the same thing happens also on X1 Carbon Gen 9 (same hardware as X1 Nano Gen1) with EC 0.1.27->0.1.29 update.
Also Fedora 34

mrhpearson, 2021-10-05:
Do you have secure boot enabled? If it's enabled can you try the update with SB disabled - there is an issue with shim that is impacting FW updates with SB enabled.
I realise it's not particularly helpful - but I've done both those updates succesfully (Fedora 34 on the X1 Nano, Debian 11 on the X1 Carbon 9)
Thanks
Mark

bmreading, 2021-10-05:
> Do you have secure boot enabled? If it's enabled can you try the update with SB disabled - there is an issue with shim that is impacting FW updates with SB enabled. I realise it's not particularly [...]

Hi Mark, thanks for the response. Good to hear that someone has been successful, but I do not have Secure Boot enabled. I've included some screenshots of some other stuff in the UEFI settings in case they are helpful:

[image]
[image]
[image]
.

lsde, 2021-10-05:
Hi, secure boot is disabled on my X1 Carbon Gen 9.

bmreading, 2021-11-07:
Hi, I just wanted to check back and see if there was perhaps a workaround for this issue? I have updated to Fedora 35, and still cannot update the embedded controller.

lsde, 2021-11-07:
@bmreading Just download the latest BIOS Update from lenovo support which also includes the embedded controller update.
Then flash it with the standard method, and it will flash both components.
(do not forget to use 'geteltorito' command if you're flashing the ISO to USB pendrive, otherwhite it won't boot)

bmreading, 2022-01-05:
@lsde Thanks. This worked for me. Closing this now.

closed 2022-01-05 by bmreading

lsde, 2022-05-14:
Just a small follow-up - the firmware now updates successfully with fwupdmgr on Fedora 36 (v1.51 -> v1.52)

hai-nguyen-van, 2024-02-19:
Doing a manual installation of the BIOS solved the issue on my ThinkPad X1 Nano Gen 1 ✅ Thanks @lsde