fwupd/firmware-lenovo #163: Thinkpad X1 Carbon 6th gen BIOS/UEFI updates missing (LVFS has 1.51, latest version is 1.53)

atopia, 2021-10-13:
The Lenovo download page lists a [BIOS/UEFI update](https://pcsupport.lenovo.com/de/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-6th-gen-type-20kh-20kg/20kh/20kh006jge/pf1e2xr1/downloads/ds502281-bios-update-utility-bootable-cd-for-windows-10-64-bit-linux-thinkpad-x1-carbon-type-20kh-20kg) to version 1.53, however the latest version in LVFS is 1.51.

mrhpearson, 2021-11-24:
Looks like 1.54 is up https://fwupd.org/lvfs/firmware/8945
OK if I close the ticket?

hughsie, 2021-11-25:
>  OK if I close the ticket?

Dude, of course, better to ask forgiveness than permission. :)

closed 2021-11-25 by hughsie

LordChaos73, 2021-11-25:
Just upgraded to 1.54, works fine! Thank you for your hard work.

hughsie, 2021-11-25:
Indeed, @mrhpearson is a shining star from my point of view.