fwupd/firmware-lenovo #165: T14/P14s gen 2 AMD lagging behind on system updates

mvdan, 2021-11-01:
1.11 was released on the 11th of October: https://pcsupport.lenovo.com/gb/en/products/laptops-and-netbooks/thinkpad-p-series-laptops/thinkpad-p14s-gen-2-type-21a0-21a1/21a0/21a0cto1ww/downloads/driver-list/component?name=BIOS%2FUEFI

LVFS only has 1.09 from early September: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1MET.firmware

It's possible and likely that other bits of firmware for this line of machines is also out of date; I haven't checked.

I couldn't find a thread for the gen2 AMD variants of these machines, so I think a new one is reasonable. I'll update this thread if fwupdmgr ever tells me there's an upgrade available.

mrhpearson, 2021-11-02:
I'll check and find out. (For my reference LO-1419)

As a note - we haven't officially released Linux support on the AMD G2 platforms due to WLAN driver issues. I know that sounds like a wonky reason - but the FW team use our certified Linux images for their testing so that may have had an impact on why the updates aren't available yet or are slower to be released (and they did manage to get 1.09 up).

mvdan, 2021-11-02:
Thanks for the very quick update! Much appreciated.

I swapped the WLAN card on my laptop for that reason. It seems like Linux 5.15 was released just this week with native support for its original Realtek card.

bdaase, 2021-11-04:
Looks like Marks rechecking worked and a new version is available now: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1MET.firmware

mvdan, 2021-11-06:
Interesting, thanks! I just tried upgrading to it, but it did not work. The error I got is https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot. My secure boot and boot order lock BIOS settings are off.

bdaase, 2021-11-07:
Interesting, for me it updated just fine.

mvdan, 2021-11-08:
I just tried again, same result. The following boot goes to a dark screen and shows a few lines of output for a split second, but it doesn't give me enough time to take a picture of it.

I also double-checked that `fwupdmgr update` does add a boot entry as expected:

```
# efibootmgr -v
BootNext: 0001
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0017,0018,0019,001A,001B,001C,0001
Boot0000* Arch Linux	HD(1,GPT,3157e685-1c71-7b46-9759-cdc8f80516a9,0x800,0xaf000)/File(\vmlinuz-linux)r.o.o.t.=.P.A.R.T.U.U.I.D.=.8.4.0.1.7.c.6.2.-.9.e.a.8.-.d.8.4.d.-.a.c.4.8.-.7.5.2.b.0.7.0.b.d.7.e.3. .r.w. .i.n.i.t.r.d.=.\.a.m.d.-.u.c.o.d.e...i.m.g. .i.n.i.t.r.d.=.\.i.n.i.t.r.a.m.f.s.-.l.i.n.u.x...i.m.g. .m.i.t.i.g.a.t.i.o.n.s.=.o.f.f. .n.o.w.a.t.c.h.d.o.g.
Boot0001* Linux-Firmware-Updater	HD(1,GPT,3157e685-1c71-7b46-9759-cdc8f80516a9,0x800,0xaf000)/File(\EFI\arch\fwupdx64.efi)
[...]
```

mrhpearson, 2021-11-09:
I usually take a video using the phone and then can pause that to check the error message - can you check that please and see if you can get any clues?
Only obvious one I can think of is make sure you're plugged in and battery charged > 50% (I think)

mvdan, 2021-11-09:
That's a clever idea :) I'll give that a go.

mvdan, 2021-11-10:
It flashed for just a few frames, but I was able to capture it:

> WARNING: Could not query capsule capabilities: Unsupported
> WARNING: Could not apply capsules: Unsupported

This is with the device plugged in with ~80% battery. It's plugged in to a usb-c dock, but I don't imagine that changes anything. The only other interesting bit of context is that I swapped the Realtek wireless card for an Intel card, so I'm not sure if that's throwing off the system update. The machine and BIOS otherwise work perfectly fine after the swap.

mvdan, 2021-11-10:
Issues like https://github.com/fwupd/fwupd/issues/1190 and https://github.com/fwupd/firmware-lenovo-thinkpad/issues/59 seem to point towards my current BIOS being too old to work with fwupd. I am on 0.1.9 at the moment, as it came when it shipped in August. I'll try updating via an alternative method.

mvdan, 2021-11-10:
Downloading 0.1.11 from the Lenovo site and burning it to a usb with geteltorito and dd worked - I'm now on that version. I'm not entirely sure why fwupd has 0.1.12 when that doesn't show up on the Lenovo firmware site. I'll try the regular fwupd upgrade again.

tpopela, 2021-11-11:
I do see that same as @mvdan - can't update firmware through fwupd.

mvdan, 2021-11-11:
I'm more than happy to move this to the fwupd issue tracker, by the way - but as far as I can tell, fwupd is doing its job correctly to download the update and boot it. Perhaps there's some difference between the files uploaded to LVFS and the ISO I obtained from lenovo.com - beyond that it's 0.1.12 versus 0.1.11.

tpopela, 2021-11-23:
@hughsie any idea how to debug this further?

mrhpearson, 2021-11-25:
Hi @mvdan 
Main comment from the FW team was "confirm battery is more than 25%" - which I know you've noted above so I will feed that back to them; but seems for them that message is usually battery related.

Mostly for the heck of it can you try without using the dock and being direct plugged into a power source. Just because I've had some notes from users getting power from monitor hubs who have strange things - I'd like to rule it out.

There are some updates coming shortly too (1.13) so we might also wait for that and see if it magically fixes it...

tpopela, 2021-11-25:
If it matter @mrhpearson I've tried it now - fully charged and connected to the charger - and still the same.

happyalu, 2021-12-16:
i noticed that 1.13 is released on the lenovo support page, but it is [listed](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1MET.firmware) with status "testing" rather than "stable".

XenGi, 2021-12-23:
Didn't check from which version I came but I just successfully flashed to 0.1.12 through fwupd on my P14s Gen2 AMD.

mvdan, 2021-12-25:
@mrhpearson I just tried upgrading again, this time to 0.1.13 and with the official charger plugged in, and I still get the same split-second error as in https://github.com/fwupd/firmware-lenovo/issues/165#issuecomment-964993004.

I'll upgrade manually once again :) Just letting you know that, as far as I can tell, it's still not working.

tpopela, 2022-01-03:
For me the same as @mvdan .

mrhpearson, 2022-01-07:
Thanks @mvdan and @tpopela - I've forwarded the details to the FW team and asked for suggestions on how to debug further.

mrhpearson, 2022-01-10:
Hi - request from the FW team - could you provide the output from:
fwupdmgr get-devices
Thanks
Mark

mvdan, 2022-01-10:
@mrhpearson anywhere I can email the output? It has serial numbers and other unique strings, so I'd rather not post them publicly. Thanks!

mvdan, 2022-01-10:
On second thought, the fwupd issue tracker asks for this info for bug reports, so I guess there isn't any harm :)

<details>

```
# fwupdmgr --version
client version:	1.7.3
compile-time dependency versions
	gusb:	0.3.8

daemon version:	1.7.3
# fwupdmgr get-devices
21A0CTO1WW
│
├─SKHynix HFS001TDE9X081N:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express solid state drive
│     Current version:    41720C20
│     Vendor:             SK hynix (NVME:0x1C5C)
│     Serial Number:      ANA4N82411010525N
│     GUIDs:              34f9d6fa-e1d6-5d7d-909e-0406644298c5 ← NVME\VEN_1C5C&DEV_174A&SUBSYS_1C5C174A&REV_00
│                         34e06406-f620-5b9d-a9cd-7f3a4e648633 ← NVME\VEN_1C5C&DEV_174A&SUBSYS_1C5C174A
│                         4406b8dc-20c5-5f3a-bc9c-8afdcee48684 ← NVME\VEN_1C5C&DEV_174A&REV_00
│                         0ee57466-ca9d-5f10-a005-ec6bacf6fb82 ← NVME\VEN_1C5C&DEV_174A
│                         867711b5-67ef-56e4-8085-0aa777d4de72 ← SKHynix_HFS001TDE9X081N
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.13
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              8fb7e1f9-447b-41fb-8832-ae5ef955665f
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │                       2f587fcf-b814-56ef-8496-db1a4a256e5e ← UEFI\RES_{8FB7E1F9-447B-41FB-8832-AE5EF955665F}
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  267
│       Minimum Version:  267
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            1e9c3d61-4bd9-514e-ad87-0a9b61b9e77d ← UEFI\CRT_156645FC005F8617669DF76E1DE6FAE02B3856D24B624E47D97BD2B8ADFFB6E0
│                         304a9ede-aff2-58f6-b5e5-0e651b065f1a ← UEFI\CRT_156645FC005F8617669DF76E1DE6FAE02B3856D24B624E47D97BD2B8ADFFB6E0&ARCH_X64
│                         c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│     
├─UEFI Device Firmware:
│     Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│     Summary:            UEFI ESRT device
│     Current version:    6409
│     Vendor:             DMI:LENOVO
[... 57 lines cut ...]
│     Minimum Version:    65549
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUIDs:              4bea12df-56e3-4cdb-97dd-f133768c9051
│                         31bfb296-9f8b-5519-94ad-04dadf19d202 ← UEFI\RES_{4BEA12DF-56E3-4CDB-97DD-F133768C9051}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:          f65c03fdcbd4f1d376a477d7ef6b99f012f2894b
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR24.11_B0036
      Bootloader Version: BOT03.02_B0009
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:       • Updatable
```

</details>

mrhpearson, 2022-01-10:
Thanks! Forwared to the FW team.
For future reference you're welcome to send them to my email if you're uncomfortable posting (markpearson at lenovo.com)
Mark

mvdan, 2022-02-09:
This thread is now about why firmware updates only work without fwupd, and no longer about a delay in versions, so I've adjusted the title.

hughsie, 2022-02-09:
> only work without fwupd

Is the non-fwupd mechanism also using UpdateCapsule?

mvdan, 2022-02-09:
I am not sure; for both 1.11 and 1.13, the fwupd route failed, but what worked was downloading the bootable CD ISO from the lenovo firmware site, and using geteltorito and dd to put it on a bootable USB. Then selecting that device in the BIOS boot menu and following the instructions, letting the machine reboot itself a couple of times.

tpopela, 2022-02-25:
@mrhpearson looks like 1.14 is missing in LVFS.

mrhpearson, 2022-02-25:
It's there - but still under testing
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1MET.firmware
Mark

mvdan, 2022-03-12:
I am simiarly now confused by how LVFS has 1.15, but the Lenovo site only has 1.14 :) Given that both versions are tagged as "high" urgency, I also wonder how long it will take for either of them to come out of testing.

bdaase, 2022-04-09:
By the way, I just noticed that Lenovo has, for whatever reason, pulled back the 1.14 and 1.15 updates from their website: https://pcsupport.lenovo.com/de/de/products/laptops-and-netbooks/thinkpad-p-series-laptops/thinkpad-p14s-gen-2-type-21a0-21a1/downloads/DS550363.

lissyx, 2022-04-13:
> By the way, I just noticed that Lenovo has, for whatever reason, pulled back the 1.14 and 1.15 updates from their website: https://pcsupport.lenovo.com/de/de/products/laptops-and-netbooks/thinkpad-p-series-laptops/thinkpad-p14s-gen-2-type-21a0-21a1/downloads/DS550363.

I'm wondering about the same, especially since I have some issues over the past days with WiFi firmware being very unstable (whatever the version of rtw89 driver and firmware I use, even versions that used to be quite stable)

bdaase, 2022-04-15:
> By the way, I just noticed that Lenovo has, for whatever reason, pulled back the 1.14 and 1.15 updates from their website: https://pcsupport.lenovo.com/de/de/products/laptops-and-netbooks/thinkpad-p-series-laptops/thinkpad-p14s-gen-2-type-21a0-21a1/downloads/DS550363.

@mrhpearson Do you know what's going on there?

mvdan, 2022-04-28:
Looks like both are on 1.17 now, so at least the inconsistency is settled.

bdaase, 2022-04-28:
@mvdan Indeed, thanks. It doesn't work for me with an "invalid capsule header" error though :/

AndrewKvalheim, 2022-05-02:
I just tried 0.1.9 → 0.1.17 with confusing results. During the reboot I saw only a black screen for several minutes, then a self-healing. fwupd reports “failed to run update on reboot”, but also shows that the current version is now 0.1.17 and the boot screen agrees.

mrhpearson, 2022-06-07:
I was checking the internal release notes for this to see if there were any known issues for this update and they don't mention any.

It would be worth trying with SB disabled - there is the shim issue that impacts updates and I don't believe has been released yet.

I'd also see if you can do it in smaller steps instead of the jump from 13 to 17. I think I can find the cab files for the older releases if you want to try that.

Mark

lissyx, 2022-06-07:
> It would be worth trying with SB disabled - there is the shim issue that impacts updates and I don't believe has been released yet.

As I mentonned earlier, 100% works when disabling SecureBoot.

tpopela, 2022-06-07:
> It would be worth trying with SB disabled - there is the shim issue that impacts updates and I don't believe has been released yet.

I can definitely try that, but that might solve the situation for me and those who will be able to find this issue, but not for all the users. Do you have a link to the shim issue Mark?

> I'd also see if you can do it in smaller steps instead of the jump from 13 to 17. I think I can find the cab files for the older releases if you want to try that.

I already tried that in the past before jumping to 0.1.13 (using the "iso" way).

mrhpearson, 2022-06-07:
Hi,
RH bug is: https://bugzilla.redhat.com/show_bug.cgi?id=1955416

tpopela, 2022-06-08:
> Hi, RH bug is: https://bugzilla.redhat.com/show_bug.cgi?id=1955416

I don't think that I'm affected by this bug as nothing gets stuck in my case. I get a "blink" and then I'm greeted with the LUKS password entry.

mrhpearson, 2022-06-10:
Can you try with SB disabled to confirm? I think the shim/UEFI updater would all be called before the LUKS password entry so I wouldn't discount it.

tpopela, 2022-06-13:
@mrhpearson so it worked with SB disabled. I will try to follow up with Jared internally to see what can be done about it. Thanks!

mrhpearson, 2022-06-13:
Awesome! They have the fix for shim - it's just the lengthy process to get it released in every distro but I'm hoping it gets out soon and that's not really fwupd specific.

I think we're good to close this bug - if anybody objects we can re-open :)

closed 2022-06-13 by mrhpearson

tpopela, 2022-06-16:
For Fedora users that have some firmware update pending - here is a shim update that "should" fix the problems - https://bodhi.fedoraproject.org/updates/FEDORA-2022-98830efc68

mvdan, 2023-07-28:
> It flashed for just a few frames, but I was able to capture it:
> 
> > WARNING: Could not query capsule capabilities: Unsupported
> > WARNING: Could not apply capsules: Unsupported

I finally found what was causing this: my fat32 boot partition was missing the `boot` and `esp` flags, so somehow that was confusing either fwupd or the BIOS. The firmware updates were put in (presumably) the right place, and the update was attempted, but it failed. Fixing this quirk now makes all my updates suddenly work :)

Hopefully this helps someone else in the future. I honestly don't recall why those flags weren't set up when I installed Arch years and years ago. It was the first time I installed with UEFI, so I probably did something slightly wrong.

hughsie, 2023-07-28:
> my fat32 boot partition was missing the boot and esp flags

If you try fwupd 1.9.3 it'll warn you about that now.

mvdan, 2024-07-24:
I wanted to circle back to say thanks again - all Thinkpad BIOS updates have gone just fine in the past year :)