fwupd/firmware-lenovo #166: P1G3 and X1E3 TBT FW v62 update fails with invalid version format

mrhpearson, 2021-11-05:
P1G3 TBT FW update fails with:

`Firmware version formats were different, device was 'pair' and release is 'plain'
`

Looks like metadata in FW is incorrect:
`<value key="LVFS::VersionFormat">plain</value>
`

Issue raised with Lenovo FW team. Tracked with internal ticket # LO-1424.
(Raising this issue so others are aware it's being looked at if they hit it)

superm1, 2021-11-06:
@hughsie at least for the thunderbolt protocol maybe worth enforcing on LVFS side that it needs to be set to version format of pair and block promotions if it's wrong?

hughsie, 2021-11-06:
> needs to be set to version format of pair

We already have that set; I'm wondering what happened here. I think the LVFS only uses the protocol verfmt as a fallback in case the component verfmt is not set, rather than a "problem" on mismatch.

Etn40ff, 2021-11-24:
Any update on this?

mrhpearson, 2021-11-24:
Afraid not - I'll go and nag

ChiWei-Chen, 2021-11-30:
Hello All, the FW team pushed the new FW 08W with correct metadata to LVFS stable. (thanks, hughsie )
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2VTF.firmware

Please let me know if this not work for you.