fwupd/firmware-lenovo #168: Lenovo X1 Carbon 6th gen UEFI update 1.53 isn't available via fwupd

aeckstein, 2021-11-08:
Hi,
apologies if this is not the right venue to bring this up, but the latest UEFI update for Lenovo X1 Carbon 6th gen UEFI has been out for six weeks or so but isn't offered via 'fwupd get-updates' (neither of the variants as far as I can tell).
Is there some known reason for this? If not, can anyone tell me if there is a person at Lenovo I can poke to make it happen? Thanks!

iyanmv, 2021-11-11:
Where do you see 1.53 is out? It [looks](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-yoga-6th-gen-type-20xy-20y0/downloads/driver-list/component?name=BIOS%2FUEFI) like 1.48 is the latest update, no?

LordChaos73, 2021-11-25:
1.54 is online, please check.

johannbg, 2021-12-07:
I can confirm 1.54, is available for the gen6 and the update worked flawlessly on fwupd-1.7.x on Fedora 35. 
I probably should mention that I had previously manually updated to 1.53 ( via Lenovo's provided cab file on Lenovo's website ) and run Fedora without any trace of grub2 installed on this particular laptop ( I use sd-boot ) so depending on the distribution in use and it's setup, the update experience might vary but the update is available for the gen6 so this issue probably can be closed.

aeckstein, 2021-12-07:
Thanks guys, I'll give it a whirl!