fwupd/firmware-lenovo #170: Last two UEFI BIOS updates missing for Thinkpad X1 Carbon Gen 9 / Thinkpad X1 Yoga Gen 6

iyanmv, 2021-11-15:
[Latest UEFI BIOS version](https://download.lenovo.com/pccbbs/mobiles/n32ul13w.txt) for Lenovo Thinkpad X1 Carbon Gen 9 and Thinkpad X1 Yoga Gen 6 is 1.49. It was released on 11/11/2021. Previous version, 1.48 was released on 28/10/2021.

However, the latest update that I received via fwupdmgr was 1.47.

mrhpearson, 2021-11-15:
I've flagged to the FW team - Internal ticket # LO-1439

iyanmv, 2021-11-15:
Thanks! Also, if I download the files from the support page and try to install them with `fwupdmgr install *.cab` as mentioned in the instructions, they fail to install because they cannot be verified.

iyanmv, 2021-11-29:
Any news about this? Right now, the only workaround is to use Windows to install the updates.

t-8ch, 2021-11-29:
1.49 seems to be available now

iyanmv, 2021-11-29:
I confirm it is. Updated without errors.

closed 2021-11-29 by iyanmv