fwupd/firmware-lenovo #171: Latest Embedded Controller Firmware update missing for Thinkpad X1 Carbon Gen 9 / Thinkpad X1 Yoga Gen 6

iyanmv, 2021-11-15:
Version [1.30 (N32HT50W)](https://pcsupport.lenovo.com/ch/en/downloads/ds548819-bios-update-utility-bootable-cd-for-windows-10-64-bit-thinkpad-x1-carbon-9th-gen-x1-yoga-6th-gen) is available, but latest on LVFS is [1.29](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN32HT.firmware).

fkru, 2021-11-15:
Lenovo's update policy for Linux seems to be very lousy. For example, the latest firmware for my ThinkPad T14 Gen 1 was published on 27 Aug on Lenovo's website, but was available on LVFS on 10 Nov.

mrhpearson, 2021-11-15:
I've flagged to the FW team - Internal ticket # LO-1439

t-8ch, 2021-11-29:
0.1.30 seems to be available now.

iyanmv, 2021-11-29:
This one, too. I confirm it is available now. Updated without errors.

closed 2021-11-29 by iyanmv