fwupd/firmware-lenovo #172: ThinkPad L13 Gen 2 on LVFS?

nre, 2021-11-23:
Hello @mrhpearson

Will the ThinkPad L13 Gen 2 be on LVFS?

My 20VH-CTO1WW isn't Yoga but they seem to share the same firmware updates. [The Yoga is certified for Ubuntu 20.04](https://ubuntu.com/certified/202011-28408) although [it isn't on Lenovo's list](https://support.lenovo.com/us/en/solutions/pd031426#l). I did get an [Ubuntu driver update for L13 Gen 2](https://bugs.launchpad.net/oem-priority/+bug/1910952).

Thanks!

Previous issues: #101.

mrhpearson, 2021-11-23:
Hi,
L13 G2 should be on the list...it has completed it's certification. Not sure how that got missed.
I'll go and bug the firmware team as to why the updates aren't on LVFS - they are supposed to be.

Mark

nre, 2022-01-18:
@mrhpearson System firmware for [vPro](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1PET.firmware) and [non-vPro](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1FET.firmware) are in testing. Thank you!

Non-vPro versions >= 1.10 have a hard requirement on ME firmware 15.0.23.1706 but I don't see `<requires>` in the [testing metadata](https://cdn.fwupd.org/downloads/firmware-testing.xml.gz).

I understand that it is still in testing. But the Vantage app (which I have no experience with because I don't have Windows) has not always respected this hard requirement and the result has been BSODs and at least one brick in [July-August](https://forums.lenovo.com/t5/ThinkPad-L-R-and-SL-series-Laptops/L13-Gen2-BSOD-after-install-of-BIOS-1-10/m-p/5088961), December [[1](https://forums.lenovo.com/t5/ThinkPad-L-R-and-SL-series-Laptops/BSOD-on-L13-Gen-2-after-Bios-update-to-v1-13-via-vantage-app/m-p/5114283)][[2](https://forums.lenovo.com/t5/ThinkPad-L-R-and-SL-series-Laptops/L13-Stuck-in-Crash-Loop-After-BIOS-Update/m-p/5116589)] and [two days ago](https://forums.lenovo.com/t5/ThinkPad-L-R-and-SL-series-Laptops/Anybody-know-where-to-download-a-old-bios/m-p/5120672).

I think Linux users are even less likely to have the required ME version because the available installer is a non-capsule exe. (I installed it using WinPE on a thumb drive.)

The requirement is documented in the [README](https://download.lenovo.com/pccbbs/mobiles/r1fpuj01wd.txt): "Important! R1FET36W (1.10) or above requires MEFW 15.0.23.1706(R1FRG03W) update first, or system might run in problem."

hughsie, 2022-01-18:
@mrhpearson if you want me to demote the firmware back to embargo say the word.

mrhpearson, 2022-01-18:
Yeah - let's put it back to embargo and I'll send them a note on why. Thanks for flagging this.

hughsie, 2022-01-18:
> let's put it back to embargo

Done.

mrhpearson, 2022-01-19:
Thanks - it was apparently a mistake for it to be in Testing in the first place, they had meant to put it in Embargo whilst they updated the metadata.
We're expecting this to be ready next week

nre, 2022-02-09:
@mrhpearson The [non-vPro ME firmware](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1FRN.firmware) 15.0.23.1706 (R1FRN17W) is now in testing without `<requires>`.

The [README](https://download.lenovo.com/pccbbs/mobiles/r1frg03w.txt) for package R1FRG03W (ME 15.0.23.1706 and other firmware updates) says "non Vpro system BIOS should be R1FET35W(v1.09) or later". (The complete dependency hell is described near the end: "Requires BIOS R1FET35W(v1.09) to update. Also, R1FET36W or above requires r1frg03w (MEFW version 15.0.23.1706) to update first.")

Is it worth checking that the requirement doesn't apply to this ME alone?

pshem, 2022-03-18:
Windows 10 just updated my UEFI without updating ME firmware first. Any tips for how I can update ME firmware now to fix it? 
Linux(Ubuntu 21.04 specifically) kernel panics on booting, Windows crashes within a minute of booting, and the ISO on a USB stick (from https://support.lenovo.com/gb/en/downloads/ds541927-bios-update-utility-bootable-cd-for-windows-10-64-bit-thinkpad-l13-l13-yoga) doesn't get detected.

nre, 2022-03-18:
@pshem That link is to BIOS for L13 Gen 1 but R1FET42W is for non-vPro L13 Gen 2. I suggest that you downgrade to [L13 Gen 2 non-vPro BIOS v1.09](https://pcsupport.lenovo.com/gb/en/products/laptops-and-netbooks/thinkpad-l-series-laptops/thinkpad-l13-gen-2-type-20vh-20vj/downloads/ds554195-bios-update-utility-bootable-cd-for-windows-11-10-64-bit-thinkpad-l13-gen-2-l13-yoga-gen-2) (scroll down to see the table with that BIOS) before updating ME. There is more [info in Lenovo's forums](https://forums.lenovo.com/t5/ThinkPad-L-R-and-SL-series-Laptops/thinkpad-l13-gen-2-immediate-bluecreens-after-bios-update-reverse-not-possible/m-p/5133910) and it is better to ask for help there.

pshem, 2022-03-18:
> @pshem That link is to BIOS for L13 Gen 1 but R1FET42W is for non-vPro L13 Gen 2. I suggest that you downgrade to [L13 Gen 2 non-vPro BIOS [...]

Thank you, that explains why I couldn't get the numerical and letter versions of the BIOS version to match between the website and what I saw in the BIOS. I was able to follow https://askubuntu.com/questions/651281/write-bootable-bios-update-iso-to-usb-stick and downgrade to  BIOS to 1.09 as you advised :D

xscd, 2022-03-21:
Hello, apparently the non-vpro (https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1FRN.firmware) is on stable and I see the update when I run fwupdmgr update. 

[image]

However I can't apply the update it just doesn't work, after reboot I see the LENOVO logo with something "applying firmware update" for a split of second but then it just reboots into fedora and I see the update is still available. 

Thinks I have tried:
*disable secure boot and try
*return bios to defaults and try
*run fwupdmgr with sudo

One more thing is I think I already have the latest update so this could be also a mismatch of the version algorithm? 
Here is my BIOS and ME version:
[image]

In the screenshot of fwupdmgr the update shows as 0.23.1706 to 240.0.23.1706 but I already have 15.0.23.1706 on my BIOS. So I am confused what these "0", "15" and "240" is on front of 23.1706 (https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1FRN.firmware also mentions the 15 and 240 numbers) so I got a little confused of whats going on and why the update can't be applied (or if its applied but there is a problem somewhere that still shows an update available)

xscd, 2022-03-22:
This seems related to #89

mrhpearson, 2022-03-22:
Hold off on the ME FW update - that is still under testing and it does look like they have something wonky going on with the numbers (I'll flag it to the FW team)
The BIOS FW is now in stable: https://fwupd.org/lvfs/firmware/9503 but I'm seeing a failed report against that. I don't have that system to test against unfortunately. I'm flagging that to the team as well.
What a mess :(

tjyrinki, 2022-05-05:
Just stating (the probably obvious) that I've a non-VPro L13 Gen 2 laptop which is on R1FET40W and no upgrades are being offered. Reading above, I'll restrain myself from experimenting and simply keep on R1FET40W until fwupdmgr tells me otherwise some sunny day... mostly the power related fixes in R1FET42W would be probably beneficial.

xscd, 2022-05-26:
Thanks for the update @mrhpearson however the links you provided aren't valid? Or they aren't visible yet since its in testing? Its the https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1FRN.firmware 15.0.35.1951 update I assume.

[image]
[image]

xscd, 2022-05-26:
Also anyone in this thread wild enough to try it out? Let us know how it went :)

xscd, 2022-05-26:
[image]

I successfully updated it! 

But there seems an issue with versioning. fwupdmgr thinks there is an update but I just applied it:
[image]

xscd, 2022-06-06:
Hi Mark, I actually posted it already above :laughing:  I hope this is it? If you need any other info let me know. Happy to help! 

[image]

mrhpearson, 2022-06-07:
Doh - sorry...not sure how I missed that.

OK - so good news is this looks like something wonky in how the ME is reporting the version to the ESRT tables. We've had a ton of issues where it hasn't been updating it so this is new that it's putting in a bad value...sigh.

Could you check what is reported in the esrt tables so we can confirm that is what is wrong. I think if you do this:
`find /sys/firmware/efi/esrt/entries -name fw_version -exec cat {} \; `
that should dump the values and I can figure it out from there.

Thanks!

xscd, 2022-06-07:
```
[tree@fedora ~]$ sudo find /sys/firmware/efi/esrt/entries -name fw_version -exec cat {} \;
16777218
1
65554
1
0
1
4028827551
1409356129
10001
1
1
2295711

```

mrhpearson, 2022-06-07:
@hughsie - think I might need your help interpreting numbers here...not sure if this is fwupd or metainfo issue or something else.
Would you mind double checking my logic please?

4028827551 = 0xF023079f
0x79f = 1951 (correct)
0x23 = 35 (correct)
0xF0 = 240 (correct for triplet format - and = 15.0 for the ME's weirdo quad format)

So I'm not sure fwupdmgr in the above thinks the version is 0.35.1951. That seems wrong.

I checked the metainfo has the format stored as triplet so am not sure why the top byte seemingly gets dropped. I haven't looked at code yet I'm afraid :)

Mark

MartijnVdS, 2022-06-08:
For some reason, this (Lenovo) update popped up on my MSI Summit E13 Evo Flip, which seems incorrect and potentially dangerous.

I have not tried to install the update, as I need this machine to work.

```
# fwupdmgr get-upgrades
Devices with no available firmware updates: 
 • Micron 3400 MTFDKBA1T0TFH
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
MS-13P2
│
└─Intel Management Engine:
  │   Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
  │   Summary:            UEFI ESRT device
  │   Current version:    0.30.1776
  │   Vendor:             Lenovo (DMI:Micro-Star International Co., Ltd.)
  │   Update State:       Success
  │   GUID:               19aef08f-2701-4db3-b90b-d4bc1412ade7
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad L13 Gen 2:
        New version:      240.35.1951
        Remote ID:        lvfs
        Release ID:       13426
        Summary:          Lenovo ThinkPad L13 Gen 2 Consumer ME Firmware
        License:          Proprietary
        Size:             4,1 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=R1FRN18W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        description (en_US): Lenovo ThinkPad L13 gen2 Consumer ME Firmware
```

hughsie, 2022-06-08:
> this (Lenovo) update popped up on my MSI Summit E13 Evo Flip

Totally unexpected. Can you attach a `fwupdmgr get-devices --json` please. The vendor ID will stop the update being applied (phew), but one of the vendors is doing a Bad Thing (tm).

hughsie, 2022-06-08:
@mrhpearson in the meantime, can you check if the Lenovo ThinkPad L13 ME GUID is actually generated, rather than using some kind of default value please. Two vendors generating the same GUID seems hugely implausible.

hughsie, 2022-06-08:
It also looks like that GUID is used on a Asus TUF Dash F15 FX516PM Laptop too. @mrhpearson I think you probably need to escalate and file an issue internally.

MartijnVdS, 2022-06-08:
> > this (Lenovo) update popped up on my MSI Summit E13 Evo Flip
> 
> Totally unexpected. Can you attach a `fwupdmgr get-devices --json` please. The vendor ID will stop the update being applied (phew), but one of the vendors is doing a Bad Thing (tm).

Here it is:
[fwupdmgr.json](https://github.com/fwupd/firmware-lenovo/files/8863904/fwupdmgr.json.txt)

mrhpearson, 2022-06-08:
Ack - and confirming with the FW team.

hughsie, 2022-06-08:
Fwiw, we can DMI lock the L13 firmware if needed, but we'd need to coordinate with the ME team as all future updates for that ID would need DMI requirements too.

hughsie, 2022-06-09:
@achimfrase has a Intel® NUC 11 Pro Kit NUC11TNKi5 -- so I suspect the ME GUID is the reference one from Intel...

compumatco, 2022-06-09:
> @achimfrase has a Intel® NUC 11 Pro Kit NUC11TNKi5 -- so I suspect the ME GUID is the reference one from Intel...

Also offered for another NUC11 model - this one a PABi7.  Something is seriously wrong with this picture.  I know not to install this because I have no idea what the ramifications are, since Intel themselves don't have a similar version number for NUC11's, but I can just imagine that it's something bad.  Any other user might install this and brick their whole computer right out of their warranty (firmware update issues are excluded from warranty service).  The update needs to be pulled entirely and fwupd needs to have its device categorization routines examined more thoroughly so this never happens again in the future.

hughsie, 2022-06-10:
>  Any other user might install this and brick their whole computer

The update won't be signed using the correct key, so even if it was deployed it wouldn't be used. I don't even think it can be deployed for new fwupd updates as the vendor ID is wrong.

>  The update needs to be pulled entirely

It's been moved out of the stable remote while we work out what happened.

> fwupd needs to have its device categorization routines examined more thoroughly

The update says "I install onto 19aef08f-2701-4db3-b90b-d4bc1412ade7" and it seems at least 4 different machines seem to share this "_Globally Unique_" ID, so there have been some problems along the chain somewhere. We've got quite a few layers of protection, so please don't worry.

xscd, 2022-06-12:
> It's been moved out of the stable remote while we work out what happened.

Its still in stable FYI or was there some progress that fixed appearing on wrong vendors?

hughsie, 2022-06-13:
> that fixed appearing on wrong vendors

Ya, we added a CHID requirement for Lenovo systems, and updated the LVFS so we always require a CHID for those GUIDs.

tjyrinki, 2022-06-16:
Same as xscd, the ME update shows up on (actual) L13 Gen 2 even though the version is the "same" (sans 0/240) already.

And after "update", it both allows sending a report about the successful update and doing the update yet again - and I can guess, looping forever.

nre, 2022-06-17:
fwupd reads the versions correctly. Another device has the full version that we expect for ME:
- "Intel Management Engine": Version 1509034 == 0x1706aa == 0.23.1706
- "UEFI Device Firmware": Version 4028040874 == 0xf01706aa == 240.23.1706 == 15.0.23.1706
 <details>

```
 FwupdDevice:
  DeviceId:             a45df35ac0e948ee180fe216a5f703f32dda163f
  Name:                 Intel Management Engine
  Guid:                 19aef08f-2701-4db3-b90b-d4bc1412ade7
  Guid:                 950e547f-0090-5a43-a74f-497902f980a9
  Summary:              UEFI ESRT device
  Plugin:               uefi_capsule
  Protocol:             org.uefi.capsule
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot|usable-during-update
  Vendor:               Lenovo
  VendorId:             DMI:LENOVO
  Version:              0.23.1706
  VersionFormat:        triplet
  VersionRaw:           0x001706aa
  Icon:                 audio-card
  Created:              2022-06-17
  UpdateState:          success

FwupdDevice:
  DeviceId:             a083ebc5138e5e071ef7270cc9a8280722cc7adf
  Name:                 UEFI Device Firmware
  Guid:                 bc7f67ec-c74a-424a-9575-305f278a37bd
  Guid:                 aa3c65b2-df2a-595f-956e-43b4d16b1f3a
  Summary:              UEFI ESRT device
  Plugin:               uefi_capsule
  Protocol:             org.uefi.capsule
  Flags:                internal|updatable|require-ac|registered|needs-reboot|usable-during-update
  VendorId:             DMI:LENOVO
  Version:              4028040874
  VersionLowest:        1
  VersionFormat:        number
  VersionRaw:           0xf01706aa
  VersionLowestRaw:     0xf01706aa
  Icon:                 audio-card
  Created:              2022-06-17
  UpdateState:          success

FuDevice      using a45df35ac0e948ee180fe216a5f703f32dda163f for /sys/firmware/efi/esrt/entries/entry0
FuPlugin      emit added from uefi_capsule: a45df35ac0e948ee180fe216a5f703f32dda163f
XbSilo        ignoring for OR statement: indexed string '950e547f-0090-5a43-a74f-497902f980a9' was unfound
FuEngine      HWID provided 6de5d951-d755-576b-bd09-c5cf66b27234
XbSilo        ignoring for OR statement: element name suggests is unknown in silo
XbSilo        ignoring for OR statement: element name X-categories is unknown in silo
FuDevice      a45df35ac0e948ee180fe216a5f703f32dda163f device overwriting name value: UEFI Device Firmware->Intel Management Engine
FuDevice      changing verfmt for a45df35ac0e948ee180fe216a5f703f32dda163f: number->triplet
FuDevice      changing version for a45df35ac0e948ee180fe216a5f703f32dda163f: 1509034->0.23.1706
FuPlugin      fu_plugin_device_registered(dell_dock)
FuPlugin      fu_plugin_device_registered(pci_bcr)
FuPlugin      fu_plugin_device_registered(thunderbolt)
FuPlugin      fu_plugin_device_registered(tpm)
FuPlugin      fu_plugin_device_registered(uefi_pk)
FuPlugin      fu_plugin_device_registered(usi_dock)
FuPlugin      fu_plugin_device_registered(msr)
FuPlugin      fu_plugin_device_registered(uefi_capsule)
FuDeviceList  ::added a45df35ac0e948ee180fe216a5f703f32dda163f

FuDevice      using a083ebc5138e5e071ef7270cc9a8280722cc7adf for /sys/firmware/efi/esrt/entries/entry8
FuPlugin      emit added from uefi_capsule: a083ebc5138e5e071ef7270cc9a8280722cc7adf
XbSilo        ignoring for OR statement: indexed string 'bc7f67ec-c74a-424a-9575-305f278a37bd' was unfound
FuPlugin      fu_plugin_device_registered(dell_dock)
FuPlugin      fu_plugin_device_registered(pci_bcr)
FuPlugin      fu_plugin_device_registered(thunderbolt)
FuPlugin      fu_plugin_device_registered(tpm)
FuPlugin      fu_plugin_device_registered(uefi_pk)
FuPlugin      fu_plugin_device_registered(usi_dock)
FuPlugin      fu_plugin_device_registered(msr)
FuPlugin      fu_plugin_device_registered(uefi_capsule)
FuDeviceList  ::added a083ebc5138e5e071ef7270cc9a8280722cc7adf

/sys/firmware/efi/esrt/entries/entry0
capsule_flags: 0x0
fw_class: 19aef08f-2701-4db3-b90b-d4bc1412ade7
fw_type: 2
fw_version: 1509034
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 0

/sys/firmware/efi/esrt/entries/entry8
capsule_flags: 0x8010
fw_class: bc7f67ec-c74a-424a-9575-305f278a37bd
fw_type: 2
fw_version: 4028040874
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 1
```
 </details>

xscd, 2022-07-22:
thinkpad l13 users getting titanium firmware updates. cc @mrhpearson 

[image]

ganchair, 2022-07-28:
Can someone here try to apply the latest BIOS update and see if they will encounter an error like this: https://forums.lenovo.com/t5/ThinkPad-L-R-and-SL-series-Laptops/Intel-L13-GEN-2-20VJ-Wrong-BIOS-Image-Is-Being-Distributed-On-The-Support-Page/m-p/5161220

It may related to Intel ME firmware so I am a little stuck and need some additional info. Would be good that someone with 15.0.23.1706 Intel ME firmware can update their BIOS with 1.19 (https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-l-series-laptops/thinkpad-l13-gen-2-type-20vh-20vj/downloads/driver-list/component?name=BIOS%2FUEFI)

nre, 2022-09-24:
[Non-vPro BIOS 1.20](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1FET.firmware) was released on LVFS stable three days ago but it `<requires><firmware compare="ge" version="240.35.1951">19aef08f-2701-4db3-b90b-d4bc1412ade7</firmware></requires>`.

[ME firmware 240.35.1951](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1FRN.firmware) has been in LVFS testing since July and has never been on https://pcsupport.lenovo.com for L13 gen 2 so I presume that Windows users never got it (unless Vantage has it).

BIOS 1.20 has been on https://pcsupport.lenovo.com for the past month for Windows users (but not other users because [the ISO version is faulty](https://forums.lenovo.com/topic/findpost/1304/5161220/5718015)).

@mrhpearson 

* Does BIOS 1.20 really require ME 240.35.1951 if Windows users don't have that?
* Is it safe to release ME 240.35.1951 now that the [GUID problem](https://github.com/fwupd/firmware-lenovo/issues/172#issuecomment-1150037718) is fixed?
* Does BIOS 1.20 fix the [ME-version problem](https://github.com/fwupd/firmware-lenovo/issues/172#issuecomment-1158753329) or will ME 240.35.1951 fix that when it is released?

ganchair, 2022-09-26:
hi @nre I have ME firmware 240.35.1951 installed yet I don't see the non-vpro bios update from fwupd. about the faulty bios image there is a new v3 release but that also failed to install with a different error. https://forums.lenovo.com/t5/ThinkPad-L-R-and-SL-series-Laptops/Intel-L13-GEN-2-20VJ-Wrong-BIOS-Image-Is-Being-Distributed-On-The-Support-Page/m-p/5161220?page=1#5752369

ganchair, 2022-09-26:
about the 240.35.1951 firmware if I remember right there was a short period of time where it was listed in pcsupport lenovo but then taken down.

ganchair, 2022-09-26:
tried the V3 image with defaulting my BIOS led to a stuck bios update session (waited over 30 mins) had to shutdown my device. nothing bad happened nothing updated guess that V3 image is also borked. 

[image]

mrhpearson, 2022-09-26:
Afraid I don't know the answers to those questions. @ChiWei-Chen - do you?

Mark

nre, 2022-09-26:
@ganchair If fwupd thinks your ME firmware version is 0.35.1951 (because of the [ME-version problem](https://github.com/fwupd/firmware-lenovo/issues/172#issuecomment-1158753329)) then it will also think that doesn't meet the requirement of >= 240.35.1951.

Otherwise check that you are on mains power instead of battery and that you don't have Boot Order Lock enabled in BIOS. I think `fwupdmgr get-devices -vvvv` showed me those problems when I had them.

ganchair, 2022-09-26:
yes it thinks like that.

```
(fwupdmgr:37354): FuMain-DEBUG: 23:17:02.323: FwupdDevice:
  DeviceId:             a45df35ac0e948ee180fe216a5f703f32dda163f
  Name:                 Intel Management Engine
  Guid:                 19aef08f-2701-4db3-b90b-d4bc1412ade7
  Summary:              UEFI ESRT device
  Plugin:               uefi_capsule
  Protocol:             org.uefi.capsule
  Flags:                internal|updatable|require-ac|registered|needs-reboot|usable-during-update
  Vendor:               Lenovo
  VendorId:             DMI:LENOVO
  Version:              0.35.1951
  VersionFormat:        triplet
  VersionRaw:           0x0023079f
  Created:              2022-09-26
  UpdateState:          success

```

ChiWei-Chen, 2022-09-27:
@nre , @mrhpearson I'm afraid I don't have this info, I'll give you back once I heard any back from FW team.

ChiWei-Chen, 2022-09-29:
Hi @nre we heard some feedback from FW team, could you help to check with it? Thanks.

> 1. The actual required version should be 15.0.23.1706. For Windows user, greater or equal than 1706 will do.
>           As for Linux users, the version 1951 should be more stable, so we set is as the requirement.
>     Back when version 1951 is released, we suffered from the BIOS control of previous issues and were checking how it should be release on web or vantage. 
> As now it should be ready, the version 2142 is also ready.  So we plan to release 2142 version instead.
> 2. With the HWID and GUID both modified, it should be safe for us release it. 
>           The previous issue happened due to other project did not set the HWID correctly.
>           Leading them to download L13 Gen2 instead. I will double check it again and set it to stable.
> 3. Do you mean the problem where user seeing their system version with something like 0.23.1706 instead of 240.23.1706?

nre, 2022-09-29:
@ChiWei-Chen 
1. I have found ME 240.23.1706 to be less stable than the previous version (e.g. errors at shutdown) so I am glad to hear that 1951 is more stable and that 2142 is also ready.
2. Thank you for checking and releasing 1951.
3. Yes, that is the problem. BIOS 1.20 is released now on LVFS but I believe no one can install it because fwupd thinks that ME 0.35.1951 does not meet the requirement of ME >= 240.35.1951. I don't want Lenovo to think that no installations means L13 gen 2 has no Linux users.

ChiWei-Chen, 2022-09-30:
@nre , thanks, I'll keep you posted once I get any FW team's reply.

ganchair, 2022-10-15:
looks like firmware 15.0.41.2142 has been released on driver page (https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-l-series-laptops/thinkpad-l13-gen-2-type-20vh-20vj/downloads/driver-list/component?name=BIOS%2FUEFI)

ganchair, 2022-12-22:
ping :eyes: any news on this? would be great to get those CVEs fixed + seems like charging improvements are in the latest firmware (also the new BIOS iso is lacking too)

ChiWei-Chen, 2022-12-23:
@ganchair I haven't got update from FW team yet, I'll need to follow up again.
BTW, it looks the latest BIOS 1.20 is stable already, could you give it a try? Thanks!
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1FET.firmware

ganchair, 2022-12-23:
> It's under internal process and planning to release within this couple weeks.

Thanks a lot for the update.

> BTW, it looks the latest BIOS 1.20 is stable already, could you give it a try? Thanks!

My fwupd doesn't show it as an update. Not sure why maybe a similar version mismatch checking like firmware has? Can I manually get the *.cab file and flash it via fwupd somehow? Would that work?

ChiWei-Chen, 2022-12-23:
@ganchair 

> My fwupd doesn't show it as an update. Not sure why maybe a similar version mismatch checking like firmware has? Can I manually get the *.cab file and flash it via fwupd somehow? Would that work?
Yes, please five it a try, if it doesn't work for you, please help to provide error message, output of `fwupdmgr --version` and `fwupdmgr -get-devices`. Thanks!

ganchair, 2022-12-23:
[image]

ganchair, 2022-12-23:
fwupdmgr --version
```
compile   org.freedesktop.fwupd         1.8.8
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.2
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.2
runtime   org.freedesktop.fwupd         1.8.8
runtime   org.kernel                    6.0.14-300.fc37.x86_64
```

fwupdmgr get-devices
```
LENOVO 20VJS0HB00
│
├─11th Gen Intel Core™ i7-1165G7 @ 2.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000a4
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         2347ad97-b545-5bae-b6b3-b70a9553cbba ← CPUID\PRO_0&FAM_06&MOD_8C
│                         1df99df5-22f4-5505-98cc-22dcc53b132a ← CPUID\PRO_0&FAM_06&MOD_8C&STP_1
│     Device Flags:       • Internal device
│   
├─Intel Management Engine:
│     Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│     Summary:            UEFI ESRT device
│     Current version:    0.35.1951
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               19aef08f-2701-4db3-b90b-d4bc1412ade7
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─KXG6AZNV512G TOSHIBA:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express solid state drive
│     Current version:    5108AGLA
│     Vendor:             Toshiba Corporation (NVME:0x1179)
│     Serial Number:      31DF33TRFDM3
│     GUIDs:              94eae9cd-5ee6-5b80-9a3e-ffa2db88cd87 ← NVME\VEN_1179&DEV_011A
│                         de51500a-8b48-5f13-b0c1-0bbded7c54c1 ← NVME\VEN_1179&DEV_011A&REV_00
│                         141102bc-3608-5bc8-a700-983f47d476f4 ← NVME\VEN_1179&DEV_011A&SUBSYS_11790001
│                         9da14130-275e-52fc-8119-63be32bd58dd ← NVME\VEN_1179&DEV_011A&SUBSYS_11790001&REV_00
│                         7907bb58-84c2-5473-bcca-255d8f33a347 ← KXG6AZNV512G TOSHIBA
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─System Firmware:
│ │   Device ID:          11a7fbdd09ce583e58b899660fe10f147fd155a2
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.18
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              78ad74e1-d048-4abd-b566-9e4e7fbd6e7c
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
[... 170 lines cut ...]
│     Minimum Version:    1
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               bc7f67ec-c74a-424a-9575-305f278a37bd
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
      Summary:            UEFI ESRT device
      Current version:    16777218
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               c5e36a4d-f132-419f-a8ce-638404c9b87a
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update

```

tjyrinki, 2022-12-23:
The problem with the 0.xx.xxxx vs 240.xx.xxxx vs 15.0.xx.xxxx has been around for at least 9 months, and I don't see there having been discussion on that front since June when @mrhpearson and @hughsie discussed it - but the discussion then prioritized the separate GUID problem.

Meanwhile, I had one L13 Gen 2 with several problems updated manually to 1.20 from the ISO via geteltorito conversion to bootable USB, and it has been reportedly working fine (at least not worse than before). Before upgrading that machine had ME 15.0.23.1706 (that is, the pre-requirement for further updates) and UEFI 1.14.

I still have occasional access to several L13 Gen 2 machines with Lenovo's pre-installed Ubuntu waiting for the actual fwupdmgr offered update. It sounds it would work fine, if the version number prefix problem would be figured out.

ravine1802, 2022-12-23:
> I had one L13 Gen 2 with several problems updated manually to 1.20 from the ISO via geteltorito conversion to bootable USB

What was your firmware version while doing this? 15.0.23.1706? I had issues upgrading the BIOS manually with bootable USB possible reason is probably my firmware being 15.0.35.1951 (see https://forums.lenovo.com/t5/ThinkPad-L-R-and-SL-series-Laptops/Intel-L13-GEN-2-20VJ-Wrong-BIOS-Image-Is-Being-Distributed-On-The-Support-Page/m-p/5161220?page=2#5752390)

ganchair, 2022-12-23:
> manually to 1.20 from the ISO via geteltorito conversion to bootable USB

What was your firmware version while doing this? 15.0.23.1706? I had issues upgrading the BIOS manually with bootable USB possible reason is probably my firmware being 15.0.35.1951 (see [forums.lenovo.com/t5/ThinkPad-L-R-and-SL-series-Laptops/Intel-L13-GEN-2-20VJ-Wrong-BIOS-Image-Is-Being-Distributed-On-The-Support-Page/m-p/5161220?page=2#5752390](https://forums.lenovo.com/t5/ThinkPad-L-R-and-SL-series-Laptops/Intel-L13-GEN-2-20VJ-Wrong-BIOS-Image-Is-Being-Distributed-On-The-Support-Page/m-p/5161220?page=2#5752390))

ganchair, 2022-12-23:
@tjyrinki just tried with 2 different sticks & verifying hashes + from two usb ports and the same thing happened waited over 15 mins for each. I am highly suspecting either its the firmware preventing it or my BIOS version 1.18 has issues. (BIOS settings were default)

tjyrinki, 2022-12-23:
Ok, that at least rules out some things, and maybe it's about the ME 15.0.35.1951 after all - which is, according to https://download.lenovo.com/pccbbs/mobiles/r1frg04w.txt - something that wasn't "officially" published, only 15.0.23.1706 rev 01 and 02 and the 15.0.41.2142.

So maybe the (recalled?) ME 15.0.35.1951 is something that causes problem, but then again it should not be a problem for upgrading UEFI for people who have ME 15.0.23.1706.

[109 more replies]

closed 2024-09-19 by kmauleon