fwupd/firmware-lenovo #173: Firmware 1.11 for X1 Extreme Gen4 (20Y5) missing

ginkel, 2021-11-27:
Hi there,

the [Lenovo support page for the X1E4](https://pcsupport.lenovo.com/de/de/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-extreme-4th-gen-type-20y5-20y6/downloads/driver-list/component?name=BIOS) lists firmware 1.11 while LVFS only has 1.09 available.

Thanks,
Thilo

ginkel, 2021-12-05:
s/1.11/1.13/

mrhpearson, 2021-12-06:
1.11 has been up on LVFS for 2 months - https://fwupd.org/lvfs/firmware/8519 so the orignal bug you're raising seems invalid. Let me know if you weren't able to update to 1.11

1.13 only landed on the Lenovo site 3 days ago. It always takes a while for the update to happen on LVFS as they do extra testing. Just the way it is.

ginkel, 2021-12-06:
The most recent version I see on LVFS is 1.09: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN40ET.firmware

mrhpearson, 2021-12-06:
Ooops - sorry, I was logged in and I have access to some internal pages that aren't generally accessible. My bad.
Turns out I'm having a bad day - I mixed up EC and BIOS firmware - I was looking at https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN40HT.firmware :(

I've raised an internal ticket: LO-1471 and will get this updated

ginkel, 2021-12-07:
Excellent, thanks a lot!

closed 2021-12-07 by ginkel