fwupd/firmware-lenovo #174: TBT 3 WS G2 dock update failing with latest fwupd

mrhpearson, 2021-12-03:
Trying to install https://fwupd.org/lvfs/firmware/8790 on TBT3 WS G2 dock. Using Fedora 35 on P1 G3 and it is failing with fwupd version  1.7.1

Attached are the logs from fwupd and fwupdmgr. Failure message reported is:
`failed to get device before update attach: failed to wait for detach replug: device 3281dd62e64cfca4eec0492d675e1d4deb1b1cea did not come back
`
I believe this is the MSP430.

[log.txt](https://github.com/fwupd/firmware-lenovo-thinkpad/files/7650633/log.txt)
[dock-update.txt](https://github.com/fwupd/firmware-lenovo-thinkpad/files/7650634/dock-update.txt)

This works with older versions of fwupd (1.5.11). The below log is from a system running Ubuntu, where it worked with 1.5.11 and then failed when I used the 1.7.1 on the same system/OS
[log_1.5.11.txt](https://github.com/fwupd/firmware-lenovo-thinkpad/files/7650635/log_1.5.11.txt)

Let me know if there is anything else I can help with for debugging

hughsie, 2021-12-04:
Notes for me: The MSP430 device auto-reboots when the firmware write gets to 100% and doesn't require attach. Will continue debugging on Monday.

hughsie, 2021-12-06:
So the weirdness seems to be that the USB context seems to stop emitting hotplug events, but only after the MSP430 device is updated. You can even insert a device on a different parent to the dock and it's ignored completely. Continuing to debug.

hughsie, 2021-12-06:
That would do it:

	==184942== Invalid read of size 8
	==184942==    at 0x4DF9261: UnknownInlinedFun (gusb-context.c:334)
	==184942==    by 0x4DF9261: g_usb_context_idle_hotplug_cb (gusb-context.c:390)
	==184942==    by 0x4CC352A: g_idle_dispatch (gmain.c:5897)
	==184942==    by 0x4CC733E: UnknownInlinedFun (gmain.c:3381)
	==184942==    by 0x4CC733E: g_main_context_dispatch (gmain.c:4099)
	==184942==    by 0x4D1C2B7: g_main_context_iterate.constprop.0 (gmain.c:4175)
	==184942==    by 0x4CC49E2: g_main_context_iteration (gmain.c:4240)
	==184942==    by 0x41CF7F: fu_device_list_wait_for_replug (fu-device-list.c:886)
	==184942==    by 0x421C43: fu_engine_get_device (fu-engine.c:2803)
	==184942==    by 0x4229B6: fu_engine_attach (fu-engine.c:3035)
	==184942==    by 0x4229B6: fu_engine_install_blob (fu-engine.c:3302)
	==184942==    by 0x422EAB: fu_engine_install_release (fu-engine.c:2517)
	==184942==    by 0x423600: fu_engine_install (fu-engine.c:2749)
	==184942==    by 0x4237FC: fu_engine_install_tasks (fu-engine.c:2167)
	==184942==    by 0x4182BC: fu_main_authorize_install_queue (fu-main.c:802)
	==184942==  Address 0x6b19cf8 is 120 bytes inside a block of size 136 free'd
	==184942==    at 0x48440E4: free (vg_replace_malloc.c:872)
	==184942==    by 0x4CCC0BC: g_free (gmem.c:199)
	==184942==    by 0x4CE66CF: g_slice_free1 (gslice.c:1183)
	==184942==    by 0x4C4CF7B: g_type_free_instance (gtype.c:2001)
	==184942==    by 0x4DF92EE: UnknownInlinedFun (gusb-context.c:375)
	==184942==    by 0x4DF92EE: g_usb_context_idle_hotplug_cb (gusb-context.c:396)
	==184942==    by 0x4CC352A: g_idle_dispatch (gmain.c:5897)
	==184942==    by 0x4CC733E: UnknownInlinedFun (gmain.c:3381)
	==184942==    by 0x4CC733E: g_main_context_dispatch (gmain.c:4099)
	==184942==    by 0x4D1C2B7: g_main_context_iterate.constprop.0 (gmain.c:4175)
	==184942==    by 0x4CC49E2: g_main_context_iteration (gmain.c:4240)
	==184942==    by 0x41CF7F: fu_device_list_wait_for_replug (fu-device-list.c:886)
	==184942==    by 0x421C43: fu_engine_get_device (fu-engine.c:2803)
	==184942==    by 0x4229B6: fu_engine_attach (fu-engine.c:3035)
	==184942==    by 0x4229B6: fu_engine_install_blob (fu-engine.c:3302)
	==184942==  Block was alloc'd at
	==184942==    at 0x484186F: malloc (vg_replace_malloc.c:381)
	==184942==    by 0x4CCF788: g_malloc (gmem.c:106)
	==184942==    by 0x4CE71B4: g_slice_alloc (gslice.c:1072)
	==184942==    by 0x4CE781D: g_slice_alloc0 (gslice.c:1098)
	==184942==    by 0x4C522DA: g_type_create_instance (gtype.c:1901)
	==184942==    by 0x4C39C9C: g_object_new_internal (gobject.c:1939)
	==184942==    by 0x4C3B68A: g_object_new_valist (gobject.c:2295)
	==184942==    by 0x4AB80ED: g_initable_new_valist (ginitable.c:244)
	==184942==    by 0x4AB81BC: g_initable_new (ginitable.c:162)
	==184942==    by 0x431424: fu_usb_backend_setup (fu-usb-backend.c:51)
	==184942==    by 0x48BAFE0: fu_backend_setup (fu-backend.c:125)
	==184942==    by 0x429143: fu_engine_load (fu-engine.c:6911)

hughsie, 2021-12-06:
@mrhpearson can you compare the libgusb versions between _working_ and _not-working_? https://github.com/hughsie/libgusb/pull/55 seems to fix the issue for me.

hughsie, 2021-12-06:
also be careful that libgusb gets built as a subproject if not new enough.

mrhpearson, 2021-12-06:
Ubuntu was using 0.3.4 and Fedora 0.3.8
I built the patched version of libgusb and installed (and did 'sudo ldconfig'). I'm still seeing the issue - do I need to rebuild fwupd as well?
Thanks!
mark

hughsie, 2021-12-06:
@mrhpearson can you try the libgusb here please: https://people.freedesktop.org/~hughsient/fedora/32/x86_64/ -- you can install the fwupd build to if that makes things easier too.

hughsie, 2022-02-07:
I can reproduce the same failure; mea culpa. https://github.com/fwupd/fwupd/pull/4255 fixes it for me.

mrhpearson, 2022-02-07:
Sweet :) Thanks - I'll give that a spin