fwupd/firmware-lenovo #175: ThinkPad Thunderbolt 4 Dock is missing

j5k, 2021-12-04:
Hello

I have problems with flickering displays and want to update my `ThinkPad Thunderbolt 4 Dock`. Can you add this device to LVFS?

``` bash
├─ThinkPad Thunderbolt 4 Dock:
│     Device ID:          bab866fd47fa89b8d33a266062db1c9869dc01b7
│     Current version:    30.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0108, TBT:0x0108)
│     GUIDs:              f1bdfa07-026c-5973-89f9-b7bd8e42ff9d ← THUNDERBOLT\VEN_0108&DEV_2031&REV_00
│                         dc3a1b4e-176f-589a-949c-d058324bb6a7 ← THUNDERBOLT\VEN_0108&DEV_2031
│                         350ec5bb-25c0-543e-8d4c-72c30d2b8df0 ← TBT-01082031
│                         2e927863-03be-5488-aeba-b0b9b3895fb5 ← TBT-01082031-controller1-1
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Device stages updates
```
https://support.lenovo.com/uu/en/solutions/pd500503-thinkpad-universal-thunderbolt-4-dock-overview-and-service-parts

hughsie, 2021-12-05:
The ODM Lenovo is using for this dock is working on the required plugin to fwupd. I don't have an ETA I'm afraid.

j5k, 2021-12-05:
Nice to hear that someone is working on it. Thanks.

ma773o, 2021-12-11:
I'm also experiencing display flickering issues when using the dock with my Thinkpad T14 Gen 1, with Fedora workstation 34.
I've a setup with two 4k monitors connected to the dock via DisplayPort 1.4 + the laptop display on (FullHD)
Flickering, or better "not stable signal" that causes the external monitor to turn black, happens only on one of the two monitors. This flickering is mitigated if I temporary set the resolution to 2k instead of the native 4k, and then back to 4k; also, it happens more frequently when the external monitor is configured to go 60 Hz instead of 30 Hz.

@hughsie, Thanks for looking into this.

@j5k, what's the shell program you've used to output the dock info? Thanks

j5k, 2021-12-16:
@ma773o: `fwupdmgr get-devices`

XenGi, 2021-12-23:
Any updates on this?
I'm trying to get this dock to work with my P14s Gen2 AMD and it seems like there is a firmware update that could solve my issues. I can't flash it via the firmware update tool because the check and install buttons stay grey and unclickable. I tried with my T480 and windows 10.

hughsie, 2021-12-29:
We added the code side of this a few days ago: https://github.com/fwupd/fwupd/tree/main/plugins/usi-dock although there's nothing on the LVFS yet.

ejawhary, 2022-01-11:
I have an issue with this dock and my ThinkPad X1 Carbon gen 9. I use gnome and in the settings I have set the power button to suspend the pc when pressed. The pc power button suspends the pc as it should however the dock power button shuts down the pc after a short press. Any ideas??

hughsie, 2022-01-13:
We merged the usi-dock plugin in 1.7.4 and the ODM is currently testing the payload on the LVFS.

j5k, 2022-03-08:
To date, no firmware update is available in LVFS. Do we know where it is stuck? The screen problem is quite annoying...

mrhpearson, 2022-03-08:
It's stuck on the Lenovo docking team delivering it :(
If you can borrow a Windows machine to update the dock I would do that rather than wait. I'm running out of ways to escalate this.

j5k, 2022-03-09:
Thanks for the information. I think the original issue of this thread is fixed and I hope, the Lenovo docking team works on it.

closed 2022-03-09 by j5k

katsar0v, 2022-04-15:
I also have T14 Gen 1 AMD and the screen flickering is killing me. Does updating via Windows fix this issue? I use Pop Os 20.04...

j5k, 2022-04-15:
Maybe. I can't update because of this problem: https://forums.lenovo.com/t5/Displays-Options-and-Accessories/Thunderbolt-4-dock-firmware-update-problem/m-p/5133318

Hope Lenovo will fix this problem soon.

XenGi, 2022-04-16:
My dock is still on version 1.0.0.7. Their is an update to 1.0.0.9, but only for Windows. I'm stuck with the flickering until this is also available for Linux over fwupd. Can someone clarify if this update finally fixes the issue?

j5k, 2022-04-18:
> My dock is still on version 1.0.0.7. There is an update to 1.0.0.9, but only for Windows. I'm stuck with the flickering until this is also available for Linux over fwupd. Can someone clarify if this update finally fixes the flickering issue?

I can't, since I'm not able to update the dock (no firmware in the LVFS yet and problem with Windows update). It's quite annoying. You can also put some pressure on Lenovo to fix this issues.

XenGi, 2022-04-19:
I updated my dock to version 1.0.09 yesterday with a friends windows notebook. Today the monitor flickered two times but for now it seems stable. I had these lucky days before too. I'll update my observations on the behavior in a few days.

GefaketHD, 2022-06-29:
Is there any update to this issue? My screen is connecting and disconnecting about every 2 seconds…I can’t work with that at all - and at this price point I would expect these issues to be solved quite fast.

j5k, 2022-06-30:
I'm now on the newest firmware. But the flickering is not fixed for me. This is really annoying!

mrhpearson, 2022-06-30:
For the origins of this ticket - I tested the (under embargo) FW update on my TBT G4 dock succesfully this morning. Went smoothly. Interestingly it's only up to 10.7 - so I think it's taken long enough to get everything working that we're behind now and I'll have to go follow up on that. Hopefully the FW update gets released soon - the team are currently testing it.

For the flickering - I wasn't seeing that on my setup (P32u Monitor using HDMI). I switched it to using thunderbolt and I did some some infrequent flickers so dropped the frequency from 60Hz to 50Hz and they went away. I'll do some investigation on that one - but let me know if that matches what you're seeing or not.

lohedges, 2022-09-15:
Apologies for hijacking this issue thread, but I am also interested in the availability of firmware for this dock within the LVFS. I am running Arch Linux on a P1 Gen 2 and the dock is mostly working other than the external displays (connected to the dock via DisplayPort) not being detected on boot or following suspend/resume. The dock itself is detected and USB peripherals attached to it work as expected. The displays are detected after disconnecting and reconnecting the TB cable from my laptop, which is my current workaround. I am also experiencing intermittent drop-out and flickering, but this is fairly minimal.

For anyone who has updated, via Windows or otherwise, does the latest firmware solve the problem that I am experiencing. Having searched Reddit and the Lenovo forum it seems like the issue isn't Linux specific. I also have access to the older TB3 workstation dock, which works fine.

For reference, my current firmware as reported by `fwupdmgr get-devices`:

```
├─ThinkPad Thunderbolt 4 Dock:
│ │   Device ID:          691357701be9715ae7179d965f4fbfefdeb6e828
│ │   Current version:    10.7
│ │   Vendor:             Lenovo (USB:0x17EF)
│ │   Serial Number:      1S40B0ZDZG00FB
│ │   GUIDs:              8d30b09f-bcc5-5379-bc65-9ccceeece1f4 ← USB\VID_17EF&PID_30B4
│ │                       2e4ffb60-b2e2-5f2f-9ea6-60931eed758b ← USB\VID_17EF&PID_30B4&REV_0100
│ │                       275e4695-9b4e-5263-835e-8681bec8cd1a ← USB\VID_17EF&PID_30B4&CID_40B0
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Device stages updates
│ │                       • Signed Payload
│ │
│ └─Dock Management Controller Information:
│       Device ID:        70835e1192ddb447c964de775916a4cd505ec3b3
│       Current version:  10.7
│       Vendor:           Lenovo (USB:0x17EF)
│       Serial Number:    1S40B0ZDZG00FB
│       Update Error:     Use the MCU to update the DMC device
│       GUIDs:            697a6a95-464e-5b90-a46e-73beab89316e ← USB\VID_17EF&PID_30B5
│                         f962de03-d70f-5440-9870-f9bbe628381a ← USB\VID_17EF&PID_30B5&REV_1007
│
├─ThinkPad Thunderbolt 4 Dock:
│     Device ID:          6df782fd9f7375a3c0752b474ebda2e130731e3a
│     Current version:    35.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0108, TBT:0x0108)
│     Update Error:       firmware update inhibited by [usi_dock] plugin
│     GUIDs:              dc3a1b4e-176f-589a-949c-d058324bb6a7 ← THUNDERBOLT\VEN_0108&DEV_2031
│                         f1bdfa07-026c-5973-89f9-b7bd8e42ff9d ← THUNDERBOLT\VEN_0108&DEV_2031&REV_00
│                         350ec5bb-25c0-543e-8d4c-72c30d2b8df0 ← TBT-01082031
│                         3e36d06f-3e44-5eb8-87c7-de0b723b1595 ← TBT-01082031-controller0-1
│     Device Flags:       • System requires external power source
│                         • Device stages updates
│                         • Updatable
│                         • Signed Payload
```

bearpebble, 2022-10-31:
@mrhpearson any idea on whether this is still being worked on?

lohedges, 2022-10-31:
According to [this](https://github.com/fwupd/firmware-lenovo/issues/273#issuecomment-1292598135) post, they are hoping to release around the beginning of November.

lawa42, 2023-01-12:
Any news here. I see the same behaviour as @lohedges 

```
├─ThinkPad Thunderbolt 4 Dock:
│     Device ID:          0bc61a5474de11264bfe1a2eb13d4b426d232326
│     Current version:    35.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0108, TBT:0x0108)
│     Update Error:       firmware update inhibited by [usi_dock] plugin
│     GUIDs:              dc3a1b4e-176f-589a-949c-d058324bb6a7 ← THUNDERBOLT\VEN_0108&DEV_2031
│                         f1bdfa07-026c-5973-89f9-b7bd8e42ff9d ← THUNDERBOLT\VEN_0108&DEV_2031&REV_00
│                         350ec5bb-25c0-543e-8d4c-72c30d2b8df0 ← TBT-01082031
│                         2e927863-03be-5488-aeba-b0b9b3895fb5 ← TBT-01082031-controller1-1
│     Device Flags:       • System requires external power source
│                         • Device stages updates
│                         • Updatable
│                         • Signed Payload
│   
├─ThinkPad Thunderbolt 4 Dock:
│ │   Device ID:          b28166ffadecd8d55b9c6d34e057ae848191debf
│ │   Current version:    1.0
│ │   Vendor:             Lenovo (USB:0x17EF)
│ │   GUIDs:              8d30b09f-bcc5-5379-bc65-9ccceeece1f4 ← USB\VID_17EF&PID_30B4
│ │                       2e4ffb60-b2e2-5f2f-9ea6-60931eed758b ← USB\VID_17EF&PID_30B4&REV_0100
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Device stages updates
│ │                       • Signed Payload
```

lohedges, 2023-01-12:
I couldn't resolve the issues so ended up selling the dock and going back to the Thunderbolt 3 variant, which works perfectly with my setup. I posted a question in the Arch forum where I explain things in more detail ([here](https://bbs.archlinux.org/viewtopic.php?id=280771)), but didn't get any help their either. I tracked the issue to the modesetting option of the NVIDIA driver, i.e. I could get the displays to be detected on boot if modesetting was disabled. (Resume still failed, though.) However, this resulted in graphical glitches and drop-out, so wasn't really a workable solution. I did find a few other posts on the NVIDIA forums mentioning similar issues caused by modesetting, but those weren't specifically related to this dock.

Cheers.

lohedges, 2023-01-12:
For reference, the issues that I describe above were _after_ having applied the firmware update via Windows (V1.0.12). A new version has been released as of today. However, the CHANGELOG doesn't mention anything related to the flickering or display detection issues.

lawa42, 2023-01-12:
Thanks for the update. I saw the new firmware update but then it disappeared and I am stuck with the blocked by USI message. I see if I can update the dock using a windows computer. I am using two monitors via DP and the standards intel graphics card. I see glitches and black screens on the monitor that is connected to the DP above the HDMI. The other one is having no issues.

lohedges, 2023-01-12:
I've ready reports of the DP above the HDMI picking up noise from the pass-through power cable. This can be resolved by using one DP and one HDMI, or by removing the pass-through cable and powering the laptop with an independent power supply, i.e. its own charger. The Thunderbolt 3 dock has a different arrangement for the ports, so isn't affected by this particular issue.

lawa42, 2023-01-12:
Good point I try this next week.

Gregor

-- 

Dipl.-Ing. Dr. Gregor Hlawacek
Group leader---Ion induced nanostructures
Ion Beam Microscopy

Helmholtz-Zentrum  Dresden-Rossendorf (HZDR)
Institute of Ion Beam Physics and Materials Research
Ion Beam Center
Office: 710/209

Bautzner Landstr. 400 | 01328 Dresden | Germany
phone: +49 (0)351 260 3409/3352

http://www.hzdr.de/

Board of Directors: Prof. Dr. Sebastian M. Schmidt, Dr. Diana Stiller
Company Registration Number VR 1693, Amtsgericht Dresden

lawa42, 2023-01-12:
Good point. I try this next week.

jankratochvil, 2023-02-13:
I have made a successul update of Lenovo Thunderbolt 4 Docking station with `fwupd-1.8.6-1.fc36.x86_64`:
```
# fwupdmgr update
Devices with no available firmware updates:
 • (null)
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • SKHynix HFM512GD3HX015N
 • System Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade ThinkPad Thunderbolt 4 Dock from 10.7 to 10.13?                      ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Before continuing, ensure the following:                                     ║
║                                                                              ║
║ • Your computer battery life if over 25%.                                    ║
║ • Do not unplug the dock during update.                                      ║
║ • Do not put your computer into sleep or hibernate mode during update.       ║
║ • This update typically may take up to 15 minutes depending on your          ║
║ notebook and dock firmware version.                                          ║
║ • Please ensure fwupd version is 1.8.6 or above.                             ║
║                                                                              ║
║ ThinkPad Thunderbolt 4 Dock and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
Downloading…             [***************************************]
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Waiting…                 [***************************************]
Writing…                 [***************************************]
Waiting…                 [***************************************] Less than one minute remaining…
Successfully installed firmware
Waiting…                 [***************************************]
Idle…                    [-                                      ]
Idle…                    [***************************************]
Successfully uploaded 1 report
```
I forgot to run `get-devices` before but now it shows:
```
├─ThinkPad Thunderbolt 4 Dock:
│ │   Device ID:          0b3c94fad31628e8adddaad723b8b6b5e24a9665
│ │   Current version:    10.13
│ │   Vendor:             Lenovo (USB:0x17EF)
│ │   Serial Number:      1S40B0ZDZA0KGN
│ │   GUIDs:              8d30b09f-bcc5-5379-bc65-9ccceeece1f4 ← USB\VID_17EF&PID_30B4
│ │                       2e4ffb60-b2e2-5f2f-9ea6-60931eed758b ← USB\VID_17EF&PID_30B4&REV_0100
│ │                       275e4695-9b4e-5263-835e-8681bec8cd1a ← USB\VID_17EF&PID_30B4&CID_40B0
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Device stages updates
│ │                       • Signed Payload
```
I had some off/on (is it a flickering?) problem with external display before, I had to switch the display 60->50Hz which did fix it. Currently I tried to switch it back to 60Hz and I see no problem so far. But it is true I haven't tried to reproduce the problem right before the firmware update.