fwupd/firmware-lenovo #176: Lenovo: Missing firmware for Lenovo V14/15IGL and IdeaPad 3 14/15IGL05 (82C3/LNVNB161216)

paulmenzel, 2021-12-06:
The laptop (compatible to Lenovo V14/15IGL and IdeaPad 3 14/15IGL05) has firmware

    DMI: LENOVO 82C3/LNVNB161216, BIOS DVCN17WW 09/29/2020

and the newer version DVCN21WW from October 9th, 2021 is [available on Lenovo’s Web site](https://pcsupport.lenovo.com/de/de/products/laptops-and-netbooks/lenovo-v-series-laptops/v15-igl/82c3/downloads/ds544169-bios-update-for-windows-10-64-bit-v14-igl-v15-igl-ideapad-3-14igl05) but not LVFS.

Lenovo said to seamlessly support GNU/Linux on *all* their devices, so it’d be great, if they delivered on that promise.

paulmenzel, 2022-01-10:
@mrhpearson, do you have an idea what to do about it? My impression was, that Lenovo would get GNU/Linux support by default on all Lenovo devices.

paulmenzel, 2022-01-10:
Thank you for clearing that up. Too bad, I misremembered the press coverage from 2020, but the article [*Lenovo is now offering full certification and Linux preinstalled hardware.*](https://www.linux-magazine.com/Online/News/Lenovo-Upping-Their-Linux-Support) clearly states, that IdeaPads are not part of it.

Though, I do not really understand the problem. The firmware updates are published on Lenovo’s Web site, can’t they be uploaded to LVFS nevertheless, and fwupdmgr be extended to show a big fat warning, that it’s unsupported?

mrhpearson, 2022-01-10:
We can't really do it in an unsupported way - the potential backlash would be too big if there were issues and systems bricked.

While LVFS is amazing and makes doing FW updates on Linux so much easier than it used to be, it's still not trivial either. The FW teams would need to put in the work to prepare the FW for uploading to LVFS (creating the right metadata, doing the upload, testing) and then would need to support it after.

Hopefully we get to expand the Linux program and eventually include the Ideapads - we've still got a way to go though.

Mark