fwupd/firmware-lenovo #178: TPM PCR0 differs from reconstruction

PhilippWendler, 2021-12-17:
**Describe the bug**
fwupd reports that the TPM PCR0 differs from reconstruction and an issue should be created with this [output of fwupdtpmevlog](https://github.com/fwupd/fwupd/files/7729107/fwupdtpmevlog.txt).

Btw.: The [wiki page](https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction#analysis) has a typo in the tool name: `fwupdtpmevelog` should be `fwupdtpmevlog`.

**Steps to Reproduce**
`fwupdmgr security --force`

**Expected behavior**
No such warning, because this is a brand-new system with a fresh installation of Ubuntu 20.04, so neither a hardware failure nor malware are likely. Furthermore, I have several identical machines and it occurs on both machines where I checked it.

**fwupd version information**
1.5.11 (from Ubuntu 20.04 apt package)

**fwupd device information**
This is a Lenovo ThinkCentre M70s with Intel Core i7-10700.

<details>
<summary>Output of "fwupdmgr get-devices --show-all-devices"</summary>

```
11DBSAPP00
│
├─Display controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    05
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              768306a9-1c54-5ad2-866c-106768410c94 ← PCI\VEN_8086&DEV_9BC5&SUBSYS_17AA3168&REV_05
│                         93ddcacf-bf57-5d67-9293-ac7c93455565 ← PCI\VEN_8086&DEV_9BC5&SUBSYS_17AA3168
│                         f677e5ac-41ac-5587-b2ef-1322dc207d18 ← PCI\VEN_8086&DEV_9BC5&REV_05
│                         a8dc77a9-e1b4-5ff1-9b0e-72dd2a64d3df ← PCI\VEN_8086&DEV_9BC5
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Embedded Controller:
│     Device ID:          230c9f798927716eeb2c5a2fc8e9304cc17cc68d
│     Current version:    65547
│     Minimum Version:    65547
│     Vendor:             DMI:LENOVO
│     GUIDs:              a7f3a1a9-fd4e-4df9-8c9b-0ce37d1a7a6e
│                         0bbacb87-7824-54ac-b4f1-5422a26aa041 ← UEFI\RES_{A7F3A1A9-FD4E-4DF9-8C9B-0CE37D1A7A6E}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          1b283e8c07f679415e2fe0172ce86d5f233aad3b
│     Current version:    1612776549
│     Minimum Version:    1612776549
│     Vendor:             DMI:LENOVO
│     GUIDs:              6159fb9c-e80b-4d13-9018-bb00bd2f353a
│                         acb3838e-98f7-5a8f-aacc-b10362ce23b5 ← UEFI\RES_{6159FB9C-E80B-4D13-9018-BB00BD2F353A}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel(R) Core™ i7-10700 CPU @ 2.90GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000ec
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         a45b0522-5722-54bd-b802-86cd044262df ← CPUID\PRO_0&FAM_06&MOD_A5
│                         538440cb-7cac-5185-bc01-d93a1e474058 ← CPUID\PRO_0&FAM_06&MOD_A5&STP_5
[... 108 lines cut ...]
✔ Intel BootGuard verified boot: Valid    
✔ Intel DCI debugger:            Locked                                              
✘ IOMMU:                         Not found
✘ TPM PCR0 reconstruction:       Invalid                                             
                                          
HSI-3                                                                                
✔ Intel BootGuard error policy:  Valid                                               
✘ Intel CET Enabled:             Not supported
✘ Pre-boot DMA protection:       Disabled                                            
✘ Suspend-to-idle:               Disabled 
✘ Suspend-to-ram:                Enabled                                             
                                          
HSI-4                                                                                
✔ Intel SMAP:                    Enabled                                             
✘ Encrypted RAM:                 Not supported
                                          
Runtime Suffix -!                         
✔ Linux kernel:                  Untainted                                           
✔ Linux kernel lockdown:         Enabled  
✔ Linux swap:                    Disabled                                            
✔ fwupd plugins:                 Untainted                                                                                                                                
                                                                                     
The TPM PCR0 differs from reconstruction.                                                                                                                                 
 » https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
```

superm1, 2021-12-17:
> If you file it here https://github.com/fwupd/firmware-lenovo-thinkpad/issues/ I can try and find the right person. @mrhpearson might also know a better way to file this internally.

Magic transfer button!

PhilippWendler, 2021-12-17:
Thank you very much!

PhilippWendler, 2022-08-30:
Interestingly, the situation has changed compared to what was described above.
I am now running fwupd 1.8.4 from the snap package and Ubuntu's 5.4.0-125-generic kernel (same Ubuntu 20.04 as above). I did not install any firmware upgrade.

Now `fwupdtool security` claims that "TPM PCR0 reconstruction" is **valid**, so it would seem this issue is fixed.
The log message that was requested above now reads
```
FuPluginTpm          comparing TPM 750c738b99a9258e935f0bf00a5ff901bacef8a36f7968691662998559cea54d and EVT 750c738b99a9258e935f0bf00a5ff901bacef8a36f7968691662998559cea54
```

I am hesitating to close this issue as fixed without further confirmation, however, because in https://github.com/fwupd/firmware-lenovo/issues/178#issuecomment-996698917 it seemed that the SHA256 hash `750c738b99a9258e935f0bf00a5ff901bacef8a36f7968691662998559cea54d` for PCR0 was surprising, and this hash is still the same.

Furthermore, the outputs of `tpm2_pcrread` and `tpm2_eventlog` for PCR0 are still the same as in https://github.com/fwupd/firmware-lenovo/issues/178#issuecomment-996850185, i.e., they *differ* (`tpm2_pcrread` shows `0x750C738B99A9258E935F0BF00A5FF901BACEF8A36F7968691662998559CEA54D` and `tpm2_eventlog` shows `0x51c3ff2e82c72fe474468de34c7bb6cd8095db69237b1f56d5096ab8cbcd7f8a`). These values are the same as previously (but other hashes have changed). So I am surprised to see fwupd now using different hashes than in https://github.com/fwupd/firmware-lenovo/issues/178#issuecomment-996613854 when these two tools keep outputting the same.

@hughsie Are the differences in hashes between `tpm2_eventlog` and the other tools including fwupd something normal and this issue can be fixed? Or is this unexpected and still a problem? I just want to make sure that fwupd doesn't mistakenly report a valid PCR0 here because AFAIU your previous comments there should be a firmware bug and a differing PCR0 is expected.

Full info of current state if necessary:
- [fwupdtool --security --force -v](https://github.com/fwupd/firmware-lenovo/files/9453752/2022-08-30-fwupdtool-1.8.4-security-force-v.txt)
- [tpm2_pcrread](https://github.com/fwupd/firmware-lenovo/files/9453754/2022-08-30-tpm2_pcrread.txt)
- [tpm2_eventlog /sys/kernel/security/tpm0/binary_bios_measurements](https://github.com/fwupd/firmware-lenovo/files/9453753/2022-08-30-tpm2_eventlog.txt) (I copied the content of `binary_bios_measurements` to a Ubuntu 22.04 machine and ran `tpm2_eventlog` there.)

mrhpearson, 2022-08-30:
It's possible they fixed it....I did have an internal bug on this open but admittedly I never got a squeak of feedback.
I'll send out some feelers and see if it was addressed or if something else happened.

hughsie, 2022-08-30:
I wondered if this was https://github.com/fwupd/fwupd/commit/0812ce5d20cad3ba0f4a40e1be697ecf0ebe30b5

PhilippWendler, 2022-08-31:
I see, this would be an explanation. Thanks for having a look!

closed 2022-08-31 by PhilippWendler