fwupd/firmware-lenovo #179: ME on T480 not updating - "image provided is not supported"

mczuchnowski, 2021-12-19:
**Describe the bug**
fwupd shows a new version of ME available. When trying to update it, on reboot the text on splash screen says "image provided is not supported by the platform". 

**NOTE:** BIOS shows ME version as 11.8.90.3987 which seems to be the latest version as shown on Lenovo support site for this model. Last successful ME update was in fact made using Windows, not sure if this has anything to do with the issue. `get-updates` command shows the update as N24RN17W. fwupd.org site https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN24RN.firmware shows the update as N24RN18W. Downloading the .cab file manually and installing it results in the same error. `downgrade` command returns `No downgrades for Intel Management Engine: current version is 184.82.3838: 184.86.3909=newer, 184.83.3874=newer, 184.82.3838=same, 184.80.3746=lowest, 184.79.3722=lowest`.

**Steps to Reproduce**
Install new ME version using fwupdmgr and reboot

**Expected behavior**
184.86.3909 is installed and should be displayed accordingly.

**fwupd version information**
```
client version:	1.5.11
compile-time dependency versions
	gusb:	0.3.4
```

**Please note how you installed it**
apt

**fwupd device information**
> fwupdmgr get-devices --show-all-devices

```
20L5S01J00
│
├─CT1000P1SSD8:
│     Device ID:          f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Summary:            NVM Express Solid State Drive
│     Current version:    P3CR013
│     Vendor:             Micron/Crucial Technology (NVME:0xC0A9)
│     GUIDs:              f405c265-73e3-5617-a874-b9a4e1475386
│                         3b22fadf-e56e-570c-9737-1664389e2bec
│                         5651166d-6b3b-537d-8fb2-eb28d5e5796c
│                         ca0b8a88-5c41-5ac1-922d-656a356d3344
│                         00736304-a88c-58b0-b156-785f8d297fb3
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Embedded Controller:
│     Device ID:          c19580db344e418cf2b23297bcf46d75c06e432a
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             DMI:LENOVO
│     GUIDs:              b87a926d-189c-49a1-b1ff-921099b1de89
│                         646e8b00-c980-5327-b889-b518b0a1cc1c
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          ea0fd60df22477668e549841f6a22a8c31cae587
│     Current version:    184.82.3838
│     Minimum Version:    184.82.3838
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUIDs:              9e21f98b-fe98-455c-b388-da5450ab6979
│                         b4c101ff-035f-54a0-9147-23a9954dc8e0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
[... 188 lines cut ...]
Boot001B* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot001C* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0022* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0024* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```

> efivar -l | grep fw
```
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-9e21f98b-fe98-455c-b388-da5450ab6979-0
```

**Additional questions**

Operating system and version: Ubuntu 20.04
Have you tried rebooting? Yes
Is this a regression? No
Are you using an NVMe disk? Yes
Is secure boot enabled? No
Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

hughsie, 2021-12-19:
@mrhpearson a mixup perhaps?

mrhpearson, 2022-01-07:
@hughsie - I'm not sure - but the file has been downloaded 167539..with 81 success reports and been available for 8M. It would seem to me the FW file itself should be fine right? I don't have a T480 to try on myself...but I feel like I'm missing something more subtle (or obvious) here.

I wonder if this is a vPro vs non-vPro issue as there are two different updates depending on which model you have. @mczuchnowski - could you confirm which you have and I'll see what I can track down here.

Mark

johnnywalls, 2022-01-14:
I can confirm the same behavior on my T480, same specs and versions as the original reporter, the downloaded file is:

```
Upgrade available for System Firmware from 0.1.36 to 0.1.41
20L5S1HU00 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
(fwupdmgr:11007): Fwupd-DEBUG: 10:32:40.066: downloading https://fwupd.org/downloads/bba2be0e31eb23d9217e6c26e26d4596065cd61ccbb03c3564ab0292500b8b84-Lenovo-ThinkPad-T480-SystemFirmware-1.41.cab
```

Laptop model: 20L5S1HU00 (LENOVO_MT_20L5_BU_Think_FM_ThinkPad T480).  All previous updates were successfully made using linux (Ubuntu 20.04, current kernel 5.11.0-46-generic).

Let me know if I can provide additional information for diagnosing the problem.  Thanks again!

ida-greuelr, 2022-02-02:
Same here for me on model t480 (20L5-0005GE), since months. fwupd shows ME Consumer update version 184.86.3909 (N24RN17W):
```
Devices with no available firmware updates: 
 • CT480BX500SSD1
 • TOSHIBA-RC100
 • Thunderbolt host controller
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • System Firmware
 • USB-C Dock G5
20L50005GE
│
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Summary:            UEFI ESRT device
  │   Current version:    184.83.3874
  │   Minimum Version:    184.83.3874
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUIDs:              9e21f98b-fe98-455c-b388-da5450ab6979
  │                       b4c101ff-035f-54a0-9147-23a9954dc8e0 ← UEFI\RES_{9E21F98B-FE98-455C-B388-DA5450AB6979}
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T480:
        New version:      184.86.3909
        Remote ID:        lvfs
        Release ID:       8333
        Summary:          Lenovo ThinkPad T480 Consumer ME Firmware
        License:          Proprietary
        Size:             2,2 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N24RN17W
        Vendor:           Lenovo
        Flags:            is-upgrade
        Description:      
        Lenovo ThinkPad T480 ME Firmware Version 11.8.86.3909(LVFS: 184.86.3909)
        
        The computer will be restarted automatically after updating completely. Do NOT turn off your computer or remove the AC adaptor while update is in progress.
        
        This stable release fixes the following issues:
        
        • Intel CSME IPU 2021.1:
        
        Addressed several critical security vulnerabilities.
```
Update is starting after reboot with a blank screen, but stops with _Reset system_ and no error message at all. When checking the results:
`fwupdmgr get-results -v 349bb341230b1a86e5effe7dfe4337e1590227bd`
I get the following report:
```
  DeviceId:             349bb341230b1a86e5effe7dfe4337e1590227bd
  Name:                 Intel Management Engine
  Guid:                 9e21f98b-fe98-455c-b388-da5450ab6979
  Plugin:               uefi_capsule
  Flags:                internal|updatable|require-ac|needs-reboot|reported|historical|usable-during-update
  Version:              184.83.3874
  Created:              2022-02-02
  Modified:             2022-02-02
  UpdateState:          failed
  UpdateError:          failed to run update on reboot
  
  [Release]
  Version:              184.86.3909
  Checksum:             SHA1(40218561f1bf267fcfcbf57645023e6a38afb2a0)
  Flags:                none
  LastAttemptStatus:    0x0
  TpmFamily:            2.0
  LastAttemptVersion:   0xb8530f22
  RuntimeVersion(org.kernel): 5.16.2-1-MANJARO
  HostSku:              LENOVO_MT_20L5_BU_Think_FM_ThinkPad T480
  RuntimeVersion(org.freedesktop.fwupd): 1.7.4
  Pcr0_SHA1:            82ee06b1ea9123e10f86407b365a26411e09a365
  BootMgrDesc:          legacy
  UEFIUXCapsule:        Enabled
  CpuArchitecture:      x86_64
  SecureBoot:           Disabled
  HostFamily:           ThinkPad T480
  CompileVersion(com.redhat.fwupdate): 12
  HostVendor:           LENOVO
  FwupdTainted:         False
  RuntimeVersion(org.freedesktop.gusb): 0.3.10
  RuntimeVersion(com.dell.libsmbios): 2.4
  MissingCapsuleHeader: False
  KernelVersion:        5.16.2-1-MANJARO
  FwupdSupported:       True
  CompileVersion(org.freedesktop.gusb): 0.3.10
  EspPath:              /boot/efi
  HostProduct:          20L50005GE
  BootTime:             1643657489
  KernelName:           Linux
  CapsuleApplyMethod:   nvram
  LinuxLockdown:        none
  Pcr0_SHA256:          acee7afb4bec8a201cc36c20731c12eac44a505bad541e01589f0e9f46684867
  EfivarNvramUsed:      29580
  KernelCmdline:        intel_iommu=on nvidia-drm.modeset=1 apparmor=1 security=apparmor udev.log_priority=3
  DistroId:             manjaro
  CompileVersion(org.freedesktop.fwupd): 1.7.4

(null)%
```
If there's anything helpful I could provide, please let me know.

mrhpearson, 2022-04-05:
Cleaning up my inbox and realised this was still open. I believe the FW has been updated and is all correct - version N24RN19W (11.8.90.3987) is posted.

Hopefully when you update you don't hit the ESRT table update issue that is plaguing ME updates right now...

I'll go ahead and close the bug - let me know if any objections or concerns.

closed 2022-04-05 by mrhpearson

ida-greuelr, 2022-08-25:
Finally,

an update for the BIOS to version 0.1.45 fixed the wrong version reporting for Intel ME and my `fwupdmgr get-updates` is not showing any outstanding (but already applied) upgrades.

Thanks for the ongoing work on this device.