fwupd/firmware-lenovo #180: T490s - 1.74 -- What is this?

alxchk, 2021-12-25:
```
ThinkPad T490s/X390:
    New version:      0.1.74
    Remote ID:        lvfs
    Release ID:       11099
    Summary:          Lenovo ThinkPad T490s/ThinkPad X390/ThinkPad System Firmware
    Variant:          W-BIOS for Machine types: 20NX, 20NY, 20Q0, 20Q1
    License:          Proprietary
    Size:             25.1 MB
    Created:          2021-09-28
    Urgency:          High
    Vendor:           Lenovo
    Flags:            is-upgrade
    Description:      
    Lenovo ThinkPad T490s/ThinkPad X390 (W-BIOS) System Firmware 1.74
    
    The computer will be restarted automatically after updating BIOS completely. Do NOT turn off your computer or remove the AC adaptor while update is in progress.
    
    This stable release fixes the following issues:
    
    • Please add fix information here.
```

Meanwhile, at official support.lenovo.com latest one is 1.73. Is it safe to install this?

ilpianista, 2022-01-04:
[Apparently](https://forums.lenovo.com/t5/ThinkPad-T400-T500-and-newer-T-series-Laptops/T490s-Bios-1-74/m-p/5111437) it was available for a while and then they removed it :thinking:

hughsie, 2022-01-04:
It's had thousands of downloads; is the shitty update description the problem?

hughsie, 2022-01-04:
I've asked the uploader to fix the update description.

alxchk, 2022-01-04:
> I've asked the uploader to fix the update description.

Description is important, however the fact that version doesn't match latest official available looks really suspicious. So it's always better to ask

mrhpearson, 2022-01-07:
The fix in that particular version, from the internal notes, is "issue where Nuvoton TPM firmware version might not detected properly". 
But there's a new 1.75 version that is in the works with a fix on the memory size reporting - so I'm guessing a bug was found that they decided was worth pulling 1.74 for (at least based on reading the internal release notes).

The release was official for anybody who did download it. I'll see if I can get confirmation on why it was pulled but if you're not seeing any issues I'd probably just keep going until 1.75 comes out.