fwupd/firmware-lenovo #182: Samsung OEM SSD firmware missing

Genius1237, 2022-01-08:
I have a Lenovo OEM version of the very popular Samsung PM981 (MZVLB1T0HALR-000L2) and the current firmware it's running is `5L1QEXA7`. fwupdmgr puts this device in the category of "Devices with no available firmware updates". I felt that this was strange as I assumed that most OEM branded SSDs have their firmware available via fwupd.

On doing some further digging, it looks like lenovo has a version of the PM981 on LVFS [link](https://fwupd.org/lvfs/devices/com.lenovo.PM981.512GB_1TB.firmware), but it's model number is slightly different (MZVLB1T0HALR-000L7). Note that the character is a 7 instead of a 2.

I'm not sure if lenovo has released any firmware updates for the model I'm running, so I wanted to check with the leonvo/fwupd team, and to see if it could be made available.

mrhpearson, 2022-04-05:
I'm afraid I'm not sure how to check - can you let me know which platform you're using and maybe I can check via the product teams. Samsung do the updates on LVFS and seem to be doing them so I assume no updates are available currently

mrhpearson, 2022-04-05:
I'm not seeing any updates here: https://download.lenovo.com/pccbbs/mobiles/fwnva51.txt
I'll see if I can find a more official answer - but I think you're good.