fwupd/firmware-lenovo #183: Missing Bios firmware update ( 1.55/N23ET80 ) for ThinkPad X1 Carbon Gen 6 (Machine types: 20KH, 20KG)

johannbg, 2022-01-09:
Another case where bios firmware update [( 1.55/N23ET80 )](https://download.lenovo.com/pccbbs/mobiles/n23ul33w.zip) is available on Lenovo website but not found in LVFS.

Manual updating the firmware ( fwupdmgr install xxx.cab ) works fine on F35 and recommended until it's available in LVFS since the firmware contains a security update.

Note: Lenovo does not provide signed firmware on their website so it triggers the "unsigned" warning when manually updating the firmware and it's documentation does not spesifically mention that users need to update daemon.conf prior to manual update so that's another thing that Lenovo should be addressing either update it's documentation or better yet provide signed firmware or best fire the person(s) that seem to be incapable of following a simple workflow of uploading the firmware to LVFS when it's being made available on Lenovo's website.

@mrhpearson aren't you making any headway having the workflow adjusted and so when the firmware is made available on Lenovo's website, it's made available in LVFS too?

LordChaos73, 2022-02-09:
Any update on this ticket please?

mrhpearson, 2022-02-14:
It looks like the FW has been delivered: https://fwupd.org/downloads/a8248641f51f0ef240e6749429e9b62702e1f5183c8201decbd4e850f4005b6c-Lenovo-ThinkPad-X1Carbon6th-SystemFirmware-1.55P.cab
Only curious thing is I'm seeing 1.54 being displayed first...but I'm assuming that it just takes some time for things to refresh. I'll check it again later to confirm it's correct.

For workflow, it's a mixed bag. We've simplified a few things which has helped but the FW team do still take extra time to test against the distro's and it really depends on some platforms vs others. Still something I'm working on I'm afraid.

LordChaos73, 2022-02-16:
There seem to be 2 available updates for the X1 Carbon 6th gen now:
* Intel ME 184.90.3987
* System firmware (Bios 1.55)

I tried to apply them both but the updates fail; the laptop reboots but does not seem to update correctly. Then reboots again normally, but the updates haven't been applied. I tried to apply the updates using fwupmgr 1.5.11 on KDE Neon.

johannbg, 2022-02-16:
I would not apply the intel ME update since it comes with a big fat warning

Version 11.8.90.3987 Fixes the  IPU 2021.2 security vulnerabilities. please refer to Issues List.
What issue list?

This firmware is provided by LVFS community members and is not provided (or supported) by the original hardware vendor.

Which begs the question provided by whom exactly, which member(s)? 

Installing this update may also void any device warranty.

In otherwords you are fucked if this bricks your machine.

Afaik Lenovo has not stopped providing firmware updates for the carbon gen 6 so why on earth is this being proposed???

hughsie, 2022-02-16:
No, this is a fwupd bug. I got the boolean the wrong way around. I'll do a new release next week with the fix included.

johannbg, 2022-02-16:
The provided search url delivers not found and afaik is wrong for the gen 6 platform  ( N23RM18W <-- dont think this exist for gen 6 ) I'm even unsure of the gen 6 is affected by this [1] + should not the URL from Intel be the one that is provided if this is from some random member in the community release ( what ever that is supposed to mean )  
[1] https://www.intel.com/content/www/us/en/security-center/advisory/intel-sa-00470.html

johannbg, 2022-02-16:
You should probably also mention in the big fat warning that this might avoid any insurances the device might have...

hughsie, 2022-02-16:
It's this https://github.com/fwupd/fwupd/commit/9ea78077af8e6f2e5dc06f9d410aeacb68f065cb -- it was my error. Ignore the text or go back to 1.7.4 if it's troubling you.

johannbg, 2022-02-16:
I'm unsure how to interperet that response, is the 184.90.3987 a legit Lenovo security update? 
Regardless the text needs fixing ( void warranty == void insurance which should be mentioned ), missing issue reference ( CVE or an an actual issue someplace ) which lvfs community member is providing the fix ( so the source of the fix can be validated ) The provided search reference URL ( legit or not )

tuch96, 2022-02-17:
I have the same problem (x1 carbon 6th). With

`fwupdmgr verify`

I get the following info:

`Selected device: System Firmware
failed to verify System Firmware: No stored checksums for 0.1.54
`

johannbg, 2022-02-20:
@tuch96 the Intel Management Engine (184.86.3909 → 184.90.3987) fails to update just like it did for @LordChaos73  and both the firmwares that verify command detects fails 
The system firmware fails with
`failed to verify System Firmware: No stored checksums for 0.1.55`
and the UHD Graphics 620 one fails with
`failed to verify UHD Graphics 620: failed to read firmware: Error reading from file: Input/output error`

johannbg, 2022-02-20:
@hughsie could the IME update failure be related to fwupd not being able to differentiate between consumer or the corporate firmware update thus downloading the wrong one, trying to update which eventually will fail?

hughsie, 2022-02-21:
> not being able to differentiate between consumer or the corporate firmware update

Well, these **should** have different GUIDs in the ESRT, so I'd hope not. Note the word in bold. :)

tuch96, 2022-02-21:
@johannbg I was able to successfully install the Intel update today. However, the system firmware update still does not work.

johannbg, 2022-02-21:
@tuch96 hmm that makes no sense, what voodoo did you perform, did you re-download the fw ( which means something was changed )? I manually updated using the system firmware update from the Lenovo ( US ) site and it work perfectly ( see first comment in this issue for the link to the fw and directions how to do that )

tuch96, 2022-02-21:
@johannbg Voodoo is the right name for it :-). I have no idea why it suddenly worked. I had started an update process in the Terminal again, but then aborted it with N and shutdown the notebook. When I started the X1 the next day, the update was applied.

tuch96, 2022-02-21:
@johannbg ...I'll try the manual method you described to install the system update.

LordChaos73, 2022-02-22:
What's the difference between the "Absolute Persistence Module version" and the "Non-Absolute Persistence Module version" as described here: https://download.lenovo.com/pccbbs/mobiles/n23ul33w.txt ?

LordChaos73, 2022-02-22:
Ah...I think I installed the wrong cab file then. By the way, after upgrading the BIOS to 1.55 (fwupdmgr install xxxxxxx.cab) I was able to upgrade the Intel ME firmware as well using fwupdmgr.

johannbg, 2022-03-01:
@hughsie how does fwupd determine which of the updates it should install, manually downloading both consumer and corporate IME updates from Lenovo ( US ) site, the consumer get's rejected with fwupdmgr while the corporate is accepted as an update, both updates ( from lvfs and from lenovo ) fail the same way just the Lenovo logo followed by a reset system ( which is awfully similar to system firmware failing to update a while back ). Is there someway to debug this further ?

hughsie, 2022-03-02:
> how does fwupd determine which of the updates it should install

The GUID in the ESRT has to match the <provide> in the XML, e.g. 

    # cat /sys/firmware/efi/esrt/entries/entry*/fw_class

rmeissn, 2022-04-12:
I'm also having problems to install the mentioned firmware (IME and System) files from fwupd. My system reboots, takes itself some time, shows a console like error text for a second, reboots to the OS and the firmware was not flashed. fwupd complained about a known issue related to secure boot (disabled on my system) and uploaded a report.

I was able to install the newest system firmware (bios, 0.1.55) by downloading it from the Lenovo website and installing it manually, after identifying the correct file, looking at the xml files inside the cab archive and comparing these to the output of 
`# cat /sys/firmware/efi/esrt/entries/entry*/fw_class`

I am unable to install the IME update - neither fwupd repo files, nor the downloaded cab file (consumer IME) from the lenovo website work. The system reboots, shows the lenovo splash screen and finally a "Reset System" and reboots to the OS, without applying the update.

nszceta, 2022-05-17:
I am also unable to install the IME update for the ThinkPad X1 Carbon 6th