fwupd/firmware-lenovo #186: Lenovo "X380 Yoga" Intel Management Engine update does not apply

biredel, 2022-01-15:
A Lenovo "X380 Yoga" attempts to apply a Lenovo "X280" Intel ME upgrade, the expected version change is not apparent after reboot.
Attempted step is "fwupdmgr upgrade". After fwupdx64.efi, system quickly reboots twice, only vendor logo visible.

Note that vendor READMEs appear to distinguish X280 and X380 models (but lvfs does not):
https://download.lenovo.com/pccbbs/mobiles/n20rk05w.txt
https://download.lenovo.com/pccbbs/mobiles/r0srg09w.txt

UEFI was upgraded using vendor bootable media:
https://pcsupport.lenovo.com/de/de/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x380-yoga/downloads/driver-list/component?name=BIOS
Did "load setup defaults" and rebooted twice plus booted cold once. Did already try with disabled "Thunderbolt > BIOS assist" or with any earlier release via "fwupdmgr block-firmware".

Have access to this or identical devices for another week.

# fwupdmgr --version
client version:	1.5.11
compile-time dependency versions
	gusb:	0.3.4

daemon version:	1.5.11
# uname -a
Linux ews58 5.11.0-46-generic fwupd/fwupd#51~20.04.1-Ubuntu SMP Fri Jan 7 06:51:40 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
# lsb_release -d
Description:	Ubuntu 20.04.3 LTS
# mount | grep /boot/efi
/dev/nvme0n1p1 on /boot/efi type vfat (rw,nosuid,nodev,noexec,relatime,fmask=0077,dmask=0077,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro,discard,x-gvfs-hide)
# mokutil --sb-state
SecureBoot enabled
# lscpu | grep -E 'Model name'
Model name:                      Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz
# dmidecode -t "system"
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.0.0 present.

Handle 0x000C, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 20LJS13G01
	Version: ThinkPad X380 Yoga
	Serial Number: MP1DQ8ZS
	UUID: be8104cc-2a2f-11b2-a85c-844d9d50db86
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_20LJ_BU_Think_FM_ThinkPad X380 Yoga
	Family: ThinkPad X380 Yoga
...
# sudo fwupdmgr refresh
Firmware metadata last refresh: 11 minutes ago. Use --force to refresh again.
# boltctl power --timeout=255 &
acquired power guard (7)
# fwupdmgr get-devices --show-all-devices
20LJS13G01
│
├─Thunderbolt host controller:
│     Device ID:          7e31a0be1472daa80112098c84e9c2ccae45fba1
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    20.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109, TBT:0x0109)
│     GUIDs:              d0450567-8484-54fc-99fa-fbbd45c7481b
│                         0c46874f-cad3-5ef7-ae2d-c09a7c983ae8
│                         b20b3ba0-9366-57c8-99f1-e28f2a01d883
│                         84fda29b-79c1-5555-b3f3-49a88ad7ea6e
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Device stages updates
│   
├─Embedded Controller:
│     Device ID:          ca368aebcf7da847029e9f2520ec55fb7a036b31
│     Current version:    0.1.34
[... 589 lines cut ...]
                                 "KernelVersion" : "5.11.0-46-generic",
                                 "TpmEventLog" : "0x00000008 4a27364234f84079f152663ae377a8f8e76b68c9 [R.0.S.E.T.5.0.W. ...]\n0x80000008 654dc9aa5e2e94185031668a2d298b498d40955f\n0x80000008 a1c6c9e5ac09c1f985271b63e3d2f9d4dcd5ccb5\n0x80000008 551c1c2fa46c9edfa77dce2b3fa76264330cbac0\n0x80000008 ca97aac54b85c9a91c76089544c6a94b71a97b22 [..........<.....]\n0x00000001 6f346d6577a514e47971240313fcb51704d0dd6a [ACPI DATA]\n0x00000001 b0469aa139a98ebeae9693588554bab11e19724c [ACPI DATA]\n0x00000004 9069ca78e7450a285173431b3e52c5c25299e473\nPCR0: 2e9a886b0654a7cbcd7dadd224e5f3daa214b527\nPCR0: ad0f894878ecd7d5920b63a82a91eca4c4185271d1aab4a34e43830d9eb336ce",
                                 "EspPath" : "/boot/efi",
                                 "HostProduct" : "20LJS13G01",
                                 "BootTime" : "1642183481",
                                 "KernelName" : "Linux",
                                 "LinuxLockdown" : "integrity",
                                 "Pcr0_SHA256" : "ad0f894878ecd7d5920b63a82a91eca4c4185271d1aab4a34e43830d9eb336ce",
                                 "RuntimeVersion(org.freedesktop.appstream-glib)" : "0.7.14",
                                 "EfivarNvramUsed" : "41632",
                                 "KernelCmdline" : "mitigations=auto,nosmt usbcore.authorized_default=0 intel_iommu=on printk.devkmsg=on pcie_ports=native pci=assign-busses,hpbussize=0x33,realloc,hpmmiosize=128M,hpmmioprefsize=16G",
                                 "DistroId" : "ubuntu",
                                 "CompileVersion(org.freedesktop.fwupd)" : "1.5.11",
                                 "RuntimeVersion(com.redhat.fwupdate)" : "12",
                                 "HostSku" : "LENOVO_MT_20LJ_BU_Think_FM_ThinkPad X380 Yoga"
                               }
                             }
                           ]
                         }
Proceed with upload? [Y|n]: Y
Idle…                    [***************************************]
Update failure is a known issue, visit this URL for more information: https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot
Successfully uploaded 1 report
# fwupdmgr downgrade 093ef0be8328a2c4ed2fe55cd36aae3171b92ade
No downgrades for Intel Management Engine: current version is 184.55.3510: 184.86.3909=newer, 184.83.3874=newer, 184.70.3626=newer, 184.65.3590=newer, 184.60.3561=newer

hughsie, 2022-01-15:
@mrhpearson can you check if the esrt guids are different please. Thanks!

mrhpearson, 2022-01-17:
Unfortunately I don't have either of those two platforms to check myself - but I've raised a ticket to the FW team for clarification (internal reference LO-1519). Will udpate ASAP

ChiWei-Chen, 2022-07-13:
Hi,
The FW team released the BIOS to stable, please let me know if this does not work for you. Thanks!
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR0SET.firmware

closed 2022-11-16 by kmauleon