fwupd/firmware-lenovo #187: UEFI firmware update on Thinkpad P52 reset boot table to default

rugubara, 2022-01-20:
**Describe the bug**
Recent update of the UEFI firmware and Intel ME has reset the boot table to the default one.
Original boot order:
```
BootCurrent: 001F
Timeout: 0 seconds
BootOrder: 0000,0018,0019,001A,001B,001C,001D,001E,001F,0020,0021,0022
Boot0000* gentoo	HD(1,GPT,82c9b306-4587-4db0-8695-0f90e1e046e7,0x800,0x82000)/File(\EFI\gentoo\grubx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Regulatory Information	FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot0015  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0016  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0017  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0018* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0019* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot001A* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001B* NVMe1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot001C* ATA HDD1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot001D* ATA HDD2	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot001E* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001F* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot0020* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot0021  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot0022  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0023* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(15,1)
Boot0024* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(15,0)
Boot0025* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0026* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354) 
```
After update:
```
BootCurrent: 001F
Timeout: 0 seconds
BootOrder: 0018,0019,001A,001B,001C,001D,001E,001F,0020,0021,0022
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Regulatory Information	FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot0015  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0016  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0017  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0018* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0019* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot001A* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001B* NVMe1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot001C* ATA HDD1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot001D* ATA HDD2	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot001E* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001F* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot0020* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot0021  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot0022  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0023* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(15,1)
Boot0024* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(15,0)
Boot0025* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0026* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

**Steps to Reproduce**
fwupdmgr update

**Expected behavior**
The boot order table should remain intact
 
**fwupd version information**
Please provide the version of the daemon and client.
[... 584 lines cut ...]
├── System.map-5.16.0-gentoo
├── System.map-5.16.1-gentoo
├── System.map-5.4.129-rt-rt61
├── System.map-5.4.129-rt-rt61.old
├── systemrescue.iso
├── vmlinuz-5.15.11-gentoo
├── vmlinuz-5.15.13-gentoo
├── vmlinuz-5.15.3-rt-rt21
├── vmlinuz-5.16.0-gentoo
├── vmlinuz-5.16.1-gentoo
├── vmlinuz-5.4.129-rt-rt61
└── vmlinuz-5.4.129-rt-rt61.old

15 directories, 398 files

```

**Additional questions**

- Operating system and version: Gentoo
- Have you tried rebooting?: yes
- Is this a regression?: yes
- Are you using an NVMe disk?: yes
- Is secure boot enabled?: no
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?: no

mrhpearson, 2022-01-21:
Hi,
Which FW did you update? And did it update cleanly and with no errors displayed.
I can flag this to the FW team, it's not expected behaviour, but I want to make sure they know which update to focus on.
Thanks
Mark

rugubara, 2022-01-21:
I think it was the System firmware on 2022-01-19 0.1.43 to 0.1.44. If I read the history correctly, other updates didn't succeed due to boot entry reset.
```
PF16W6Y2 /var/tmp/portage/sys-devel/gcc-11.2.1_p20220115/temp # fwupdmgr get-history
20M90019RT
│
├─UEFI Device Firmware:
│ │   Device ID:          0f3c64df11304fc5711ef77a93e3e042e2738907
│ │   Previous version:   0.1.15
│ │   Update State:       Success
│ │   Last modified:      2019-12-21 13:21
│ │   GUID:               88b8ba7e-296b-4f9e-929f-dcc19318dbcf
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │
│ └─  New version:      0.1.16
│       License:          Unknown
│
├─SAMSUNG MZVLB512HAJQ-000L7:
│ │   Device ID:          c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
│ │   Previous version:   4L2QEXA7
│ │   Update State:       Success
│ │   Last modified:      2020-03-14 19:30
│ │   GUID:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │
│ └─  New version:      5L2QEXA7
│       License:          Unknown
│
├─System Firmware:
│ │   Device ID:          65b6a9dc7b7df18bdff003584b51bf21373e3aa6
│ │   Previous version:   0.1.40
│ │   Update State:       Success
│ │   Last modified:      2021-04-27 20:51
│ │   GUID:               1e1fe415-74e8-49e1-9508-106b3d13d50d
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │
│ └─  New version:      0.1.41
│       Remote ID:        lvfs
│       License:          Unknown
│
├─Intel Management Engine:
│ │   Device ID:          619057f1d0da67b3fa4e9f74e09fc9218ce32643
│ │   Previous version:   192.70.1652
│ │   Update State:       Failed
│ │   Update Error:       (null)/fwupdx64.efi cannot be found
│ │   Last modified:      2021-06-16 07:08
│ │   GUID:               971682af-21a0-46e3-91c0-825702c2ba70
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
[... 53 lines cut ...]
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad P52/P72:
        New version:      0.1.44
        Remote ID:        lvfs
        Release ID:       10904
        Summary:          Lenovo ThinkPad P52/P72 System Firmware
        License:          Proprietary
        Size:             16,2 MB
        Created:          2021-10-28
        Urgency:          Critical
        Vendor:           Lenovo
        Description:
        Lenovo ThinkPad P52P72 System Firmware Version 1.44

        Important updates:

        • Update includes a security fix.

        New functions or enhancements:

        • Updated the CPU microcode.
        • Updated the Diagnostics module to version 04.18.000.

```

mrhpearson, 2022-04-07:
Feedback from the FW team is the only time they'll be touching the efi variables is if you go into the setup and reset to defaults. Otherwise they shouldn't be touching them. 
We're trying to reproduce but so far haven't been sucessful

rugubara, 2022-05-07:
Another update I installed on my P52, and again the boot entries were reset. i suspected that it might happen and I captured the efibootmgr output just before the update and just after.
[20220507.txt](https://github.com/fwupd/firmware-lenovo/files/8645854/20220507.txt)

```
PF16W6Y2 ~ # fwupdmgr get-history
20M90019RT
│
├─UEFI Device Firmware:
│ │   Device ID:          0f3c64df11304fc5711ef77a93e3e042e2738907
│ │   Previous version:   0.1.15
│ │   Update State:       Success
│ │   Last modified:      2019-12-21 13:21
│ │   GUID:               88b8ba7e-296b-4f9e-929f-dcc19318dbcf
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │
│ └─  New version:      0.1.16
│       License:          Unknown
│       Description:
│       The vendor did not supply any release notes.
│
├─SAMSUNG MZVLB512HAJQ-000L7:
│ │   Device ID:          c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
│ │   Previous version:   4L2QEXA7
│ │   Update State:       Success
│ │   Last modified:      2020-03-14 19:30
│ │   GUID:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │
│ └─  New version:      5L2QEXA7
│       License:          Unknown
│       Description:
│       The vendor did not supply any release notes.
│
├─System Firmware:
│ │   Device ID:          65b6a9dc7b7df18bdff003584b51bf21373e3aa6
│ │   Previous version:   0.1.40
│ │   Update State:       Success
│ │   Last modified:      2021-04-27 20:51
│ │   GUID:               1e1fe415-74e8-49e1-9508-106b3d13d50d
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │
│ └─  New version:      0.1.41
│       Remote ID:        lvfs
│       License:          Unknown
│       Description:
│       The vendor did not supply any release notes.
│
├─Intel Management Engine:
│ │   Device ID:          619057f1d0da67b3fa4e9f74e09fc9218ce32643
│ │   Previous version:   192.70.1652
│ │   Update State:       Failed
│ │   Update Error:       (null)/fwupdx64.efi cannot be found
│ │   Last modified:      2021-06-16 07:08
[... 63 lines cut ...]
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad P52/P72:
        New version:      0.1.46
        Remote ID:        lvfs
        Release ID:       12990
        Summary:          Lenovo ThinkPad P52/P72 System Firmware
        License:          Proprietary
        Size:             16,2 MB
        Created:          2022-03-22
        Urgency:          Critical
        Vendor:           Lenovo
        Description:
        Lenovo ThinkPad P52P72 System Firmware Version 1.46

        New functions or enhancements:

        • Updated the Diagnostics module to version 04.22.000.

        Problem fixes:

        • Fixed an issue where always show "Configuring Thunderbolt Controller..." message on every boot.

```

mrhpearson, 2022-05-08:
Thanks for the notes - it does look like your EFI variables are completely wiped, at least that is my guess. I've forwarded the details to the FW team for their comment (as a note for myself, internal ticket LO-1716) LO-1716
In the above you also have some interesting failure logs for the ME update. I've no idea if they're related - but they do stand out.

Can you check you don't have 'Boot Order Lock' enabled in the BIOS - just in case that's responsible.

Thanks
Mark

AndriyKalashnykov, 2022-05-22:
I have had same behavior i.e. boot  table was reset. Now i want to upgrade bios to from 0.1.41 to 0.1.46 and can't afford any downtime. Can anybody provide steps to upgrade BIOS without loosing boot table or how to put it back where it was after the upgrade so i can continue from where i started but with new BIOS?

mrhpearson, 2022-06-11:
I followed up with the FW team on this and we can't reproduce and they are adamant that they don't deliberately reset them at any point.

@hughsie - I'm grasping at straws - but I assume we need to update the efiboot variable to run the linux firmware updater on the next boot? e.g from this capture

```
BootNext: 0001
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0018,0019,001A,001B,001C,001D,001E,001F,0020,0021,0022,0001
Boot0000* gentoo	HD(2,GPT,534c523b-b9fa-bb45-b5af-ac06f1a856a0,0x2000800,0xfa000)/File(\EFI\gentoo\grubx64.efi)
Boot0001* Linux-Firmware-Updater	HD(2,GPT,534c523b-b9fa-bb45-b5af-ac06f1a856a0,0x2000800,0xfa000)/File(\EFI\gentoo\fwupdx64.efi)
```

Something would have gone in and at minimum updated the BootOrder to make entry 0001 the first one?
So it's _possible_ that the boot variables were wrong before the updater was called.

Is there any way we can discount OS, fwupdx64 or shim or grub doing something wonky in this area? 
I don't know the efiboot variable storage well enough - but is it checksummed or something like that?

I agree that BIOS seems like a likely culprit...but they're telling me they don't touch this space (unless the user does something manually) so I'm a bit stuck.

Mark

jonbrenner, 2023-05-30:
I can confirm that updating to 1.53, as well as earlier firmware versions, removes my EFI boot entry. I use EFISTUB booting to directly boot a unified kernel image, discounting the idea that an intermediate bootloader could be to blame:

```
Boot0000* Arch Linux (Secure Boot)	HD(1,GPT,22c1fcf0-d66f-4484-a807-30596dc41360,0x800,0xfa000)/File(Arch\linux-signed.efi)
```