fwupd/firmware-lenovo #189: Lenovo T470: `fwupdmgr update` has effectively bricked my machine

yawpitch, 2022-01-21:
**Describe the bug**

Lenovo T470 failed to reboot after `fwupdmgr update`, bypassing the usual "Lenovo" boot screen and just going to black the first time. Waited over an hour just in case as the fan was going, but found no response to the power button; eventually tried a hardware reset via paperclip to the hardware reset button. This time I got the "Lenovo" screen, but after a few seconds "System Reset" appeared in the top left and again the machine froze, unresponsive once again to the power button, this time without fan speed. As per https://github.com/fwupd/firmware-lenovo/issues/140 only resort is to hardware reset again. On third attempt able to get to the bootloader and into Linux, however a fourth reboot has me back at a black boot screen. Attempted to boot into Windows to try the update there, but can no longer log into Windows as the machine again freezes up and becomes unresponsive to the power button as soon as the Windows 10 login splash becomes visible. Even tried downgrading to a previous version using gnome-firmware (couldn't find appropriate way to downgrade with fwupdmgr) and this appeared to work at first, but again got the black screen on reboot. Tried everything I found in fwupd/fwupd#3819 and several open issues on https://github.com/fwupd/firmware-lenovo, to no avail.

- Secure Boot is disabled
- Boot Order Lock is disabled
- LUKS Disk encryption enabled on boot
- Tried both "UEFI only" boot and "both" (legacy and UEFI), same outcome

**Steps to Reproduce**
```shell
fwupdmgr refresh
fwupdmgr get-updates
fwupdmgr update
```

**Expected behavior**
The updates to complete and, ideally, to not have to keep a straightened paperclip handy at all times.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
```

```shell
client version:	1.7.4
compile-time dependency versions
	gusb:	0.3.10

daemon version:	1.7.4
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

```shell
> pacman -Syu fwupd
```

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

```shell
20HDCTO1WW
│
├─Core™ i7-7600U CPU @ 2.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000ea
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         561403e8-143a-5071-ab09-bf5e4c146983
│                         9ca69899-3716-5857-9fd3-882a5c73236f
│     Device Flags:       • Internal device
│
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.36
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUIDs:              4b37c365-ecb2-4d89-8e15-e93c27c6ce12
│                         897f3781-238c-56a0-95fa-e2612102db0e
│     Device Flags:       • Internal device
[... 703 lines cut ...]
│       ├── videotest.mod
│       ├── xfs.mod
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
├── System Volume Information
│   ├── AadRecoveryPasswordDelete
│   └── ClientRecoveryPasswordRotation
└── vmlinuz-linux
```
**Additional questions**

- Operating system and version: Arch, fully updated
- Have you tried rebooting? YES ... many, many times. By paperclip.
- Is this a regression? Unknown.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? No.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? It is a Lenovo system, Boot Order Lock is disabled.

ChiWei-Chen, 2022-01-25:
@yawpitch thanks for sharing the information.
There are some following question from FW team, could you please check it?
>It seems system may hang after reboot. I have a question, during hang status, can you see ”Configuration change” in screen? Or the screen is in black without any word?

Thanks.

yawpitch, 2022-01-25:
@ChiWei-Chen On my first attempt I ran the update in my Linux session as normal, then when asked to reboot I hit “Y” and the machine reboots itself I get the Lenovo screen, after the “Hit Enter to enter Setup” message disappears about 10-15 seconds elapse before “Reset System” (or “System Reset”, I can’t remember for sure) appears. About 5-10 seconds later the screen then goes black (no backlight, no message) as if the machine is rebooting. Fan continues to run, but machine becomes unresponsive to power button. Power must then be disconnected and the machine force reset, or wait till battery dies.

On second boot the machine posts to my normal Arch Linux boot screen, but all three updates (EC, ME, and BIOS) show as “failed to run update on reboot” in the `fwupdmgr lget-history` output. Note that I usually have to restart the fwupd.service to get the history after a paper clip reset.

If I then run the update, or any downgrade, I will either get the first results (a Lenovo screen and “Reset System”) or a black screen with active backlight and a single underscore in the top left. This appears to happen any time the Linux Firmware Updater boot target is used.

In an attempt to get my machine useable I’ve now used Lenovo’s downloadable ISO to update to EC 1.36, ME 184.86.3909, and BIOS 0.1.69… however if I manually enter the BIOS setup and make any changes I will get another hang rather than reboot. Additionally fwupdmgr now thinks I’ve got 0.1.36, 184.77.3664, and 0.1.69 installed… the first two still being incorrect per the BIOS boot screen.

The main problem now is that my machine will very frequently hang on any attempt to reboot, and will always hang immediately upon attempting to reach the login screen of the dual-booted Windows installation.

ChiWei-Chen, 2022-03-16:
@yawpitch, I got a trial BIOS for T470, If you or someone is willing to test that it would be appreciated - please send me an email. Thanks!
(cchen50 at lenovo dot com)

Installation will be by fwupdmgr and it will involve updating the BIOS and then

trying a ME FW update
If that doesn't work, downgrading ME and then upgrading again
Thanks

yawpitch, 2022-03-19:
@ChiWei-Chen thank you for that, and sorry for the delay. I’ll email you shortly.

ChiWei-Chen, 2022-03-22:
@yawpitch, thanks for your support, I just sent you the trial BIOS.

ChiWei-Chen, 2022-04-12:
Hi,
I have a trail BIOS and MEFW for T480. If someone is willing to test that it would be appreciated - send me or @mrhpearson an email. Thanks!
(cchen50 at lenovo dot com)

Installation will be by fwupdmgr and it will involve updating the BIOS and then trying a ME FW update.

Thanks

mrhpearson, 2022-06-10:
Closing issue due to lack of updates. We can reopen if there are updates.

yawpitch, 2022-06-17:
This is still an issue. I have installed the trial BIOS provided by @ChiWei-Chen, but cannot upgrade the MEFW, either to the one provided or to the latest I get with `fwupdmgr update` (184.92.4222). Does not seem to matter which BIOS I use the MEFW update will always go through a boot that eventually says "reset system", then an automatic reboot and "failed to run update on reboot" as the error message in get-history.

Essentially I am stuck on 184.77.3664. which is sub-optimal.

mrhpearson, 2022-06-17:
I'll re-open and let Chi-Wei comment. I know the FW team are working thru the details of how we release BIOS and then ME...but I've lost track of where we are with the different platforms

ChiWei-Chen, 2022-06-21:
@yawpitch , the FW release is still in the internal verification process, it is expected to be released a few weeks later. Thanks.

yawpitch, 2022-07-14:
The BIOS update went through fine, however I still cannot update the ME FW from 184.77.3664. Any attempt to do so will only go as far as "reset system" before dumping out to a normal reboot.

The output of `fwupdmgr get-history` is:

```
LENOVO 20HDCTO1WW
│
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Previous version:   0.1.69
│ │   Update State:       Success
│ │   Last modified:      2022-07-14 14:25
│ │   GUID:               4e06f4ae-ab2e-4804-9a23-888e596a31e7
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │
│ └─ThinkPad T470 / ThinkPad 25:
│       New version:      0.1.70
│       Remote ID:        lvfs
│       Release ID:       13722
│       Summary:          Lenovo ThinkPad T470 / 25 System Firmware
│       License:          Proprietary
│       Size:             9.4 MB
│       Created:          2016-07-08
│       Urgency:          Critical
│       Vendor:           Lenovo
│       Description:
│       This stable release fixes the following issues:
│
│       • Fixed an issue where Intel Management Engine firmware update by LVFS is failed.
│
│       (Note) Drive encryption by OPAL management software needs to be disabled before
│
│       start firmware update.
│
│       Some new functionality has also been added:
│
│       • Enhancement to address security vulnerability
│       • Updated the Diagnostics module to version 04.24.000.
│       • Updated the CPU microcode.
│       • Update MEBx.
│
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Previous version:   184.77.3664
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2022-07-14 15:34
  │   GUID:               bcff5fd8-0a99-48ad-b630-40e8eadb190e
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Reported to remote server
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad T470:
        New version:      184.92.4222
        Remote ID:        lvfs
        Release ID:       12871
        Summary:          Lenovo ThinkPad T470 Corporate ME Firmware
        License:          Proprietary
        Size:             7.5 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N1QRM25W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:
        This stable release fixes the following issues:

        • Mitigated the following security vulnerabilities:
```

Anything I can do on my end?

ChiWei-Chen, 2022-07-15:
@yawpitch , thanks, could you let us know the Intel Management Engine Firmware version in your BIOS Main page (press F1 in preboot stage)? If it's 184.92.4222 it could be the same as #240, we would need FW team to merge the fix for T470.

yawpitch, 2022-07-15:
@ChiWei-Chen I've just confirmed, it is indeed 184.92.4222.

ChiWei-Chen, 2022-07-15:
@yawpitch , thanks for confirmation, I'll check the status with T470 FW team.

ChiWei-Chen, 2022-07-19:
@yawpitch , The FW team replied the official BIOS will be released few weeks later, I got a new trial BIOS for T470, If you or someone is willing to test that it would be appreciated - please send me an email. Thanks!
(cchen50 at lenovo dot com)

thomasboussekey, 2022-07-24:
Hello,

I encounter the same problem than @yawpitch on my professionnal **Lenove T580** when trying to update firmware to the same firmware version.

This upgrade seems to be rolled-back.
Yet, I encounter random session restart when I perform copy/paste operation (CTRL+C, CTRL+V or even SHIFT+INS).

It's hard to work in these conditions,
Any workaround/solution welcome :pray: 

Thanks in advance,

ChiWei-Chen, 2022-07-26:
@thomasboussekey thanks for reporting this. This thread seems to long.
Could you help to create a separate ticket for T580 and provide more details of the system? Thanks!
- fwupdmgr update --show-all
- system firmware version and Intel Management Engine firmware version in BIOS main page
- OS and fwupdmgr version

thomasboussekey, 2022-07-26:
Thanks for your feedback @ChiWei-Chen 
As cpoy/paste makes my linux session randomly restart, it might be hard to collect these information.

It was so annoying to try to work in this context, that I switched to my second professionnal laptop (on DELL hardware, without firmware issue). I will try to collect these information and open the new ticket

ChiWei-Chen, 2022-07-27:
@thomasboussekey I'm sorry to hear this, it looks really annoying. If possible could you also try the ISO method to update the BIOS and Intel Management Engine firmware? Thanks.
https://support.lenovo.com/us/en/downloads/ds502394-bios-update-utility-bootable-cd-for-windows-10-64-bit-linux-thinkpad-p52s-t580
https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t580-type-20l9-20la/downloads/ds548228-intel-management-engine-118-firmware-consumer-update-utility-for-linux-thinkpad-p52s-types-20lb-20lc-t580-types-20l9-20la?category=Motherboard%20Devices%20%28core%20chipset,%20onboard%20video,%20PCIe%20switches%29

yawpitch, 2022-07-27:
> @yawpitch , The FW team replied the official BIOS will be released few weeks later, I got a new trial BIOS for T470&T470s, If you or someone is willing to test that it would be appreciated - please send me an email. Thanks! (cchen50 at lenovo dot com)

Hi @ChiWei-Chen, I've finally had a chance to install the trial BIOS, and it does _appear_ to resolve the issue. I was able to successfully force a reinstall of the most recent ME update (184.92.4222) and that version number is now reflected both in the BIOS main page and in `fwupdmgr get-history`.

Can you please do me a favour and update this page when this BIOS version is officially released?

Thank you for your help, I shall close this issue now as resolved.

closed 2022-07-27 by yawpitch

ChiWei-Chen, 2022-07-27:
@yawpitch , thanks, it's good to know this work for you. As I confirmed with the FW team, they're in the internal process, it would take a few weeks to push to stable.

ChiWei-Chen, 2022-08-23:
Hi,
The FW was released to stable, please let me know if this doesn't work for you. Thanks!
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1WET.firmware
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1QET.firmware

gandhiRodcas, 2022-11-15:
Hi, It´s possible post de correct commands order please

ChiWei-Chen, 2022-11-16:
@gandhiRodcas please use following commands to update the firmware, thanks!
```
fwupdmgr refresh
fwupdmgr get-updates
fwupdmgr update
```