fwupd/firmware-lenovo #190: Lenovo X1G7 failed to reboot after Intel Management Engine update

zaggash, 2022-01-21:
**Describe the bug**
Lenovo X1G7 failed to reboot after Intel Management Engine update

* Secure Boot disabled
* 'Boot Order Lock' disabled
* LUKS Disk Encryption enabled on boot

It worked for a year and it is failing on this update.

**Steps to Reproduce**
```
❯ sudo fwupdmgr refresh                                                                                                                                                                               
❯ sudo fwupdmgr upgrade   
```                                                                                                                                                                               

**Expected behavior**
The laptop ask to reboot then reboot then the update is installed.

**fwupd version information**
Please provide the version of the daemon and client.
```
sudo fwupdmgr --version                                                                                                                                                                                        
client version:	1.6.4
compile-time dependency versions
	gusb:	0.3.7

daemon version:	1.6.4
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Pacman

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices

20QDCTO1WW
│
├─USB3.1 Hub:
│ │   Device ID:          da77984c82b59c6fc69516431f467fd9a8d39a7f
│ │   Summary:            USB 3.x hub
│ │   Current version:    13.24
│ │   Vendor:             VIA Labs, Inc. (USB:0x17EF)
│ │   Install Duration:   15 seconds
│ │   GUIDs:              227d0f95-a764-598e-85de-3d2bd0a7372f
│ │                       e0c50668-5c2e-54df-8e17-c7c2472f3eaa
│ │                       c6381f16-fe6d-5c45-a4ad-b619af4f0f0e
│ │                       fc6cfb35-d4e9-5467-a014-8863de104969
│ │                       7ae3100f-9b01-571c-9fc2-20cd4544123a
│ │                       ef65710a-3fa0-5330-9a61-5ced5e976be1
│ │   Device Flags:       • Updatable
│ │                       • Cryptographic hash verification is available
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │
│ ├─USB2.0 Hub:
│ │     Device ID:        0b48e60e0f5abf120fa19357ad0f627aec55d218
│ │     Summary:          USB 3.x hub
│ │     Current version:  13.23
│ │     Vendor:           VIA Labs, Inc. (USB:0x17EF)
│ │     Install Duration: 15 seconds
│ │     GUIDs:            5a743458-4929-5738-a812-84cad6574a37
│ │                       7a216856-8a97-550c-882e-8233751c7cf2
│ │                       36a4a5cc-b78b-5419-a2ba-3a692148e9f5
│ │                       575dc5a0-ff9d-5df7-8777-aa9531061b0c
│ │                       9495f83b-bee0-5484-beeb-374325c67cb6
│ │     Device Flags:     • Updatable
│ │                       • Cryptographic hash verification is available
[... 338 lines cut ...]
│   ├── Linux
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   └── arch.conf
│   ├── loader.conf
│   └── random-seed
└── vmlinuz-linux

9 directories, 11 files
```

**Additional questions**

- Operating system and version: Archlinux Kernel 5.14.8-arch1
- Have you tried rebooting? Yes
- Is this a regression? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No
- Is Secure Boot Enabled? No

mrhpearson, 2022-01-21:
Hi @zaggash 
Sorry, I wasn't aware of this issue originally. We had a whole bunch of ME FW update problems on a number of platforms.
My understanding is the FW team claims these have been fixed in the next update, though there were still some concerns around older platforms. I believe the X1C7 got the fix - I'll dig through my notes and see if I can find the details.

I guess the question is - have you been able to update the ME FW with any of the updates since?

DonnieWest, 2022-01-21:
@mrhpearson I'm also the owner of a X1C7 and I've been experiencing the same problems as described here. Including today

Do you have any good troubleshooting steps or anything I can provide to help you diagnose?

zaggash, 2022-01-21:
@mrhpearson No, the issue still exists even today.
I can't pass this failling upgrade.

I'm ok to test anything you can provide to mitigate and fix the issue.

Rgds.

deriamis, 2022-01-22:
I've been having this exact issue with the same upgrade for ages now - for several months at least. I don't have secure boot enabled (never have), and the update actually runs but never applies. Other firmware upgrades succeed just fine - I just can't seem to get off of Lenovo ThinkPad X1 Yoga 4th/X1 Carbon 7th Corporate ME Firmware version 192.68.1606 specifically no matter how I try.

kmauleon, 2022-01-24:
hi @zaggash can you please try disabling disk encryption?  For Windows case, any disk encryption tool is not supported and should be suspended or disabled first prior to update.  I guess this will have the same behavior with any type of disk encryption tool in Linux.  Thank you very much.

kmauleon, 2022-01-24:
@deriamis have you checked the MEFW version in Setup Menu after the update?  Please advise.  Thank you very much

deriamis, 2022-01-24:
> @deriamis have you checked the MEFW version in Setup Menu after the update?  Please advise.  Thank you very much

@kmauleon The version number I reported was after the update. I don't know what it was before because I didn't think to look, sorry.

kmauleon, 2022-01-24:
hi @deriamis got it. the issue you were seeing is still being looked at by BIOS Dev.  I don't think they have update yet.
hi @mrhpearson, will BIOS Dev release a probable fix for the ESRT table issue?  Please advise.  Thank you very much.

kmauleon, 2022-01-24:
@deriamis sorry forgot to mention... your system MEFW was updated as confirmed in BIOS Setup Menu.  For some reason, fwupd is not getting the correct version and this is related to ESRT table issue which is being managed by BIOS.

zaggash, 2022-01-25:
> hi @zaggash can you please try disabling disk encryption? For Windows case, any disk encryption tool is not supported and should be suspended or disabled first prior to update. I guess this will have the same behavior with any type of disk encryption tool in Linux. Thank you very much.

I can't disable LUKS encryption, it is such a pain I prefer to live without the ME update then.
The others updates are working fine, so be with it.

kmauleon, 2022-01-25:
sorry @zaggash not much of a help.  I think we can close this particular issue then.  Thank you very much.

closed 2022-01-25 by kmauleon

DonnieWest, 2022-01-25:
@kmauleon is there an issue we can follow for the ESRT table issue? Turns out I have the same problem @deriamis is having

kmauleon, 2022-01-26:
hi @DonnieWest quite a number actually but i'm afraid no update yet as of now... #65 #67 #70 
thank you very much