fwupd/firmware-lenovo #192: ECP firmware not updated with UEFI

maikzumstrull, 2022-02-01:
I was dealing with a T14 Gen1 with overheating issues, and eventually noticed that it was running an unsupported firmware combination: UEFI 1.23 (latest) with ECP 1.08 (originally paired with the 1.17 main firmware).

My understanding is that ECP updates are supposed to be applied in lockstep with UEFI updates, unless there isn't a new release of either, as per this list from the Lenovo README:

```
  The following versions of UEFI BIOS and ECP (Embedded Controller Program) have
  been released to date.

  Package      (ID)     UEFI BIOS (BIOS ID)  ECP (ECP ID)      Rev.  Issue Date
  --------------------  -------------------  ---------------   ----  ----------
  1.23      (N2XUR15W)  1.23 (N2XET33W)      1.13 (N2XHT23W)   02    2021/11/25
  1.23      (N2XUR15W)  1.23 (N2XET33W)      1.13 (N2XHT23W)   01    2021/11/08
  1.22      (N2XUR14W)  1.22 (N2XET32W)      1.13 (N2XHT23W)   01    2021/09/03
  1.21-1.12 (N2XUR13W)  1.21 (N2XET31W)      1.12 (N2XHT22W)   01    2021/08/12
  1.21      (N2XUR12W)  1.21 (N2XET31W)      1.11 (N2XHT21W)   01    Withdrawn
  1.20      (N2XUR11W)  1.20 (N2XET30W)      1.10 (N2XHT20W)   01    2021/06/04
  1.19      (N2XUR10W)  1.19 (N2XET29W)      1.09 (N2XHT19W)   01    2021/03/30
  1.17      (N2XUR09W)  1.17 (N2XET27W)      1.08 (N2XHT18W)   01    2021/01/15
  1.16      (N2XUR08W)  1.16 (N2XET26W)      1.07 (N2XHT17W)   01    2020/12/03
  1.14      (N2XUR07W)  1.14 (N2XET24W)      1.07 (N2XHT17W)   01    2020/10/05
  1.13      (N2XUR06W)  1.13 (N2XET23W)      1.07 (N2XHT17W)   01    2020/09/03
  1.10      (N2XUR05W)  1.10 (N2XET20W)      1.07 (N2XHT17W)   01    2020/08/13
  1.09      (N2XUR04W)  1.09 (N2XET19W)      1.06 (N2XHT16W)   01    2020/07/17
  1.07      (N2XUR03W)  1.07 (N2XET17W)      1.05 (N2XHT15W)   01    2020/06/16
  1.06      (N2XUR02W)  1.06 (N2XET16W)      1.04 (N2XHT14W)   01    2020/06/02
  1.05      (N2XUR01W)  1.05 (N2XET15W)      1.03 (N2XHT13W)   01    2020/05/28
```

I don't know why this would happen, but I checked a second Thinkpad with Fedora I had to hand and it also had the latest UEFI but was missing ECP updates. (It didn't seem to have given that other model any noticeable issues though.)

The bootable images released by Lenovo always include the appropriate pairing of ECP and UEFI firmware and will apply both. It seems the cab-in-zip distributions aren't following that model.

hughsie, 2022-02-01:
In general we only do one firmware update at a time on all Lenovo hardware due to this: https://github.com/fwupd/fwupd/blob/main/plugins/uefi-capsule/uefi-capsule.quirk#L13 --- if that needs to be more nuanced (e.g. to models) can someone at Lenovo let me know. Originally we switched on the quirk because of problems with UEFI _coalesce_ i.e. https://github.com/fwupd/fwupd/commit/a676a5ec70fc3e69468575d7ed0a636a4388e932

My assumption would be that we'd have some kind of requirement specified in the firmware to make the EC and system BIOS versions "match up". e.g. https://lvfs.readthedocs.io/en/latest/metainfo.html#other-firmware-version

mrhpearson, 2022-02-02:
Thanks for flagging this - I think this looks like the FW team haven't delivered the EC update for this platform...at least as far as I can tell. I can't see the 1.13 release, and that should be available. 

I've flagged it to the FW team as an urgent issue (LO-1556) but as a note with it being Chinese New Year I likely won't get an update until next week.

With the linking the two: I'll see what feedback I get. I believe we do that on the AMD systems, but it's a single package there. We've always delivered EC and BIOS separately on the Intel platforms- and I've had cases where only one needs updating. The "other firmware version" looks like a good case for that.

Mark

mrhpearson, 2022-03-11:
FW is now available. Let me know if any issues
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2XHT.firmware

closed 2022-03-11 by mrhpearson