fwupd/firmware-lenovo #193: Lenovo T460: missing latest uefi bios versions

tehcaster, 2022-02-03:
Hi,

I've noticed there have been UEFI BIOS updates 1.42 and 1.43 released by Lenovo:
https://pcsupport.lenovo.com/cz/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t460/downloads/ds112122

But the latest in LVFS is still 1.41:
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR06ET.firmware

Thanks in advance for poking the right people.

mrhpearson, 2022-02-03:
I'll look into it (for my reference, internal ticket LO-1558)

ChiWei-Chen, 2022-04-28:
@tehcaster , 
The latest FW 1.45 is updated to LVFS stable now, please let me know if this does not work for you.
Thanks for raising this topic.

https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR06ET.firmware

tehcaster, 2022-04-28:
Updated to 1.45 just fine today, thanks!

closed 2022-04-28 by tehcaster