fwupd/firmware-lenovo #194: Thinkpad X1 Gen 8 / UEFI version 0.1.24 with critical update should be pushed to stable

sukahiroaki, 2022-02-03:
So far the update for the recent critical security issues [¹]has only been pushed to testing. Especially given that those issues are now public knowledge this should be pushed to stable so that it actually gets to users.

[¹] https://www.bleepingcomputer.com/news/security/uefi-firmware-vulnerabilities-affect-at-least-25-computer-vendors/

mrhpearson, 2022-02-03:
Ack - flagged to the FW team (LO-1559). I know we have a lot of updates waiting in embargo state too.
Mark

ChiWei-Chen, 2022-02-09:
@sukahiroaki , I'm working with @mrhpearson , our FW team pushed BIOS 0.1.24 to stable. Could you give it a try then let us know how it works. Thanks!

https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2WET.firmware

sukahiroaki, 2022-02-09:
Just applied the update, went flawlessly. Looking good.

mrhpearson, 2022-02-09:
Closing. Thanks Chi-Wei

closed 2022-02-09 by mrhpearson