fwupd/firmware-lenovo #196: ThinkPad P43s, P53s, T490, T590 Thunderbolt critical update missing

persicsb, 2022-02-04:
Lenovo issued a critical update to ThinkPad P43s, P53s, T490, T590 Thunderbolt controllers on 08 Nov 2021, with ID 1.21.07.14-N2ITF15W.
This is missing from LVFS but it is really needed.

mrhpearson, 2022-04-05:
Sorry for the slow reply on this one - I had missed it.

I did an update in #160 today - and the 4 platforms are in the platforms that don't have updates available. Check the linked bug for details as to why they're not up yet.

I'm working to get them added, but recommendation for now is to install Windows and update from there.

Would you mind if we closed this bug and just track thru #160 ?

persicsb, 2022-04-05:
Sure, I'll close this.

> recommendation for now is to install Windows and update from there.
This is really sad.

closed 2022-04-05 by persicsb