fwupd/firmware-lenovo #197: New Lenovo X1 gen9 N32HT52W ECP 1.32 is out

arno01, 2022-02-09:
The new "Lenovo Thinkpad X1 Carbon 9th / X1 Yoga 6th Embedded Controller Firmware N32HT52W, version 1.32." is out, and is addressing the common problem with the primary power port issue due to which I just had my mobo replaced in https://github.com/fwupd/fwupd/issues/4267

```
Version ID: N32HT52W
Release Date: 01/28/2022
```
source https://support.lenovo.com/lu/en/downloads/ds548819-bios-update-utility-bootable-cd-for-windows-10-64-bit-thinkpad-x1-carbon-9th-gen-x1-yoga-6th-gen

But based on /var/cache/fwupd/metadata.xmlb, fwupdmgr is currently offering older ECP 1.31:

```
Version ID: N32HT51W
Release Date: 11/29/2021
```

```
root@x1gen9:~# fwupdmgr --version
client version:	1.5.11
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.11
```

## Question
how can I either modify `/var/cache/fwupd/metadata.xmlb` or update that elsewhere to instruct `fwupdmgr` to use the latest N32HT52W ECP 1.32?
(I've already tested ECP 1.32 installed via usb dongle on my previous (replaced) mobo, it works well.)

arno01, 2022-02-09:
> This was uploaded a few days ago, I assume it's undergoing testing before being pushed to stable.

Is there some way I can override some setting manually, so that I can get the updates sooner, before they get tested/QA'ed?

The reason is as I've already tried that update on my previous motherboard, it worked well. And the other reason is that I love running the latest FW [announced](https://support.lenovo.com/lu/en/downloads/ds548819-bios-update-utility-bootable-cd-for-windows-10-64-bit-thinkpad-x1-carbon-9th-gen-x1-yoga-6th-gen) at the support.lenovo.com.

They way I was usually flashing my bios was the usb dongle:

> `/dev/sdb` is the usb dongle in my case
```
geteltorito -o n32ur15w.img n32ur15w.iso 
wipefs -a /dev/sdb
sudo dd if=n32ur15w.img of=/dev/sdb
```

hughsie, 2022-02-09:
> Is there some way I can override some setting manually

No, it's currently embargoed I'm afraid. I'm sure Lenovo will push it soon.

arno01, 2022-02-10:
Looks like it got pushed! :partying_face: 
Am gonna update today :-)

```
root@x1gen9:~# fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Successfully downloaded new metadata: 6 local devices supported
root@x1gen9:~# fwupdmgr update
Devices with no available firmware updates: 
 • Integrated Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Upgrade available for Embedded Controller from 0.1.31 to 0.1.32
20XXS3JC1Q must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: N
Request canceled

root@x1gen9:~# strings /var/cache/fwupd/metadata.xmlb |grep 'Lenovo Thinkpad X1 Carbon 9th'
Lenovo Thinkpad X1 Carbon 9th / X1 Yoga 6th System Firmware N32ET75W, version 1.51.
Lenovo Thinkpad X1 Carbon 9th / X1 Yoga 6th Embedded Controller Firmware N32HT52W, version 1.32.
```

arno01, 2022-02-10:
## Success!

Gonna close this now.

And the CPU Voltage returned back to `1.0 V` (or maybe it depends on whether I started the laptop with or without the power cable connected? :thinking:)

```
root@x1gen9:~# diff -Nur dmidecode.after.mobo.replacement.plus.fwupd-4th-run dmidecode.after.mobo.replacement.plus.fwupd-5th-run
--- dmidecode.after.mobo.replacement.plus.fwupd-4th-run	2022-02-09 14:40:28.843897855 +0100
+++ dmidecode.after.mobo.replacement.plus.fwupd-5th-run	2022-02-10 11:39:48.397103562 +0100
@@ -485,7 +485,7 @@
 		TM (Thermal monitor supported)
 		PBE (Pending break enabled)
 	Version: 11th Gen Intel(R) Core(TM) i7-1185G7 @ 3.00GHz
-	Voltage: 0.8 V
+	Voltage: 1.0 V
 	External Clock: 100 MHz
 	Max Speed: 3000 MHz
 	Current Speed: 3000 MHz
@@ -536,7 +536,7 @@
 		Targeted content distribution is supported
 		UEFI is supported
 	BIOS Revision: 1.51
-	Firmware Revision: 1.31
+	Firmware Revision: 1.32
 
 Handle 0x0017, DMI type 1, 27 bytes
 System Information
@@ -740,7 +740,7 @@
 Handle 0x0035, DMI type 219, 106 bytes
 OEM-specific Type
 	Header and Data:
-		DB 6A 35 00 01 04 01 45 02 00 90 06 81 85 3B 30
+		DB 6A 35 00 01 04 01 45 02 00 90 06 81 85 30 30
 		00 00 00 00 40 00 00 03 1F 00 00 C9 03 40 C4 02
 		FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
 		FF FF FF FF FF FF FF FF 03 00 00 00 80 00 00 00
@@ -839,7 +839,7 @@
 Handle 0x0041, DMI type 141, 18 bytes
 OEM-specific Type
 	Header and Data:
-		8D 12 41 00 54 48 4E 4B 00 00 00 00 00 00 02 00
+		8D 12 41 00 54 48 4E 4B 00 00 00 00 00 00 17 00
 		00 00
 
 Handle 0x0042, DMI type 141, 18 bytes
@@ -863,13 +863,13 @@
 Handle 0x0045, DMI type 141, 30 bytes
 OEM-specific Type
 	Header and Data:
-		8D 1E 45 00 54 48 4E 4B 00 00 50 00 00 00 DB 94
-		4E 01 00 00 00 00 DF DF 85 04 00 00 00 00
+		8D 1E 45 00 54 48 4E 4B 00 00 50 00 00 00 40 57
+		51 01 00 00 00 00 6E BA 93 04 00 00 00 00
 
 Handle 0x0046, DMI type 140, 15 bytes
 ThinkPad Embedded Controller Program
-	Version ID: N32HT51W
-	Release Date: 11/29/2021
+	Version ID: N32HT52W
+	Release Date: 01/28/2022
 
 Handle 0x0047, DMI type 140, 43 bytes
 OEM-specific Type
root@x1gen9:~# 
```

closed 2022-02-10 by arno01