fwupd/firmware-lenovo #198: Missing Thunderbolt firmware for T25 (Type 20K7) Laptop (ThinkPad)

arnottcr, 2022-02-09:
As tracked in #32, the Thinkpad 25 also requires the Thunderbolt firmware update, however this is not automatically handled by `fwupdmgr update`. #32 calls out "T470 / ThinkPad 25", so I would have expected that this would just work, as that issue is closed.

Furthermore, the [lenovo support link](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t470/downloads/DS121681) from #30 indicates that it targets "ThinkPad 25, T470", however the [fwupd link](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadNNQTF.firmware) does not. As well, the cab file does not apply citing the following error:
```
# fwupdmgr install 94a7d4e7880ef6e1b40b5f1601f37d65f0609719c1e3d7feeac3747bb0162468-Lenovo-ThinkPad-T470-SystemFirmware.cab 
Decompressing…           [***************************************]
No supported devices found
```

hughsie, 2022-02-09:
> since I would expect to see a Thunderbolt Controller line

what's `fwupdmgr get-devices --show-all` say?

arnottcr, 2022-02-09:
```
20K70004US
│
├─Core™ i7-7500U CPU @ 2.70GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000ec
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         9ca69899-3716-5857-9fd3-882a5c73236f ← CPUID\PRO_0&FAM_06&MOD_8E&STP_9
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.36
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUIDs:              4b37c365-ecb2-4d89-8e15-e93c27c6ce12
│                         897f3781-238c-56a0-95fa-e2612102db0e ← UEFI\RES_{4B37C365-ECB2-4D89-8E15-E93C27C6CE12}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─GM108M [GeForce 940MX]:
│     Device ID:          5987c136e0024e0f45943927a31c5c4c711f3eb0
│     Current version:    a2
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              957ec1b4-5dea-5e00-b95d-3083d55e59b2 ← PCI\VEN_10DE&DEV_134D&SUBSYS_17AA2246&REV_A2
│                         277b6c6a-073e-532d-84c5-27d97d97df6d ← PCI\VEN_10DE&DEV_134D&SUBSYS_17AA2246
│                         00a8a8f7-ff82-5270-b8a9-967773b857e3 ← PCI\VEN_10DE&DEV_134D&REV_A2
│                         96c4a8d8-4d13-5ab3-908b-a4aa84fa3617 ← PCI\VEN_10DE&DEV_134D
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─HD Graphics 620:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    02
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              4f08c73a-6f08-5df8-9449-e0c810736339 ← PCI\VEN_8086&DEV_5916&SUBSYS_17AA2246&REV_02
│                         fdbfee18-5c87-5e9f-ab10-8f7fd88cbc5f ← PCI\VEN_8086&DEV_5916&SUBSYS_17AA2246
│                         2886e312-afd8-5510-8993-12d568a85e00 ← PCI\VEN_8086&DEV_5916&REV_02
│                         38f6c009-a25c-5b50-b3b5-fef4f9c6b846 ← PCI\VEN_8086&DEV_5916
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    184.86.3909
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUIDs:              1c0f9993-38e5-4a62-8dab-f2efee639b63
│                         f8cee9cd-9f7c-5318-b4c1-06119247b496 ← UEFI\RES_{1C0F9993-38E5-4A62-8DAB-F2EFEE639B63}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─LENSE20512GMSP34MEAT2TA:
│     Device ID:          c82980cd3fe66f7d0a8bb83b353cbc271177efd4
│     Summary:            NVM Express solid state drive
[... 67 lines cut ...]
│     Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│     Current version:    71.12.17568.4100
│     Vendor:             ST Microelectronics (TPM:STM)
│     GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
│                         84df3581-f896-54d2-bd1a-372602f04c32 ← TPM\VEN_STM&DEV_0001
│                         bfaed10a-bbc1-525b-a329-35da2f63e918 ← TPM\VEN_STM&MOD_
│                         70b7b833-7e1a-550a-a291-b94a12d0f319 ← TPM\VEN_STM&DEV_0001&VER_2.0
│                         06f005e9-cb62-5d1a-82d9-13c534c53c48 ← TPM\VEN_STM&MOD_&VER_2.0
│     Device Flags:       • Internal device
│   
└─UEFI Device Firmware:
      Device ID:          f95c9218acd12697af946874bfe4239587209232
      Summary:            UEFI ESRT device
      Current version:    4653068
      Minimum Version:    4653068
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUIDs:              1cd9297e-323b-4bc2-b6ae-235ba25fa39c
                          5a520181-9b26-501d-b090-a7475d5f5e68 ← UEFI\RES_{1CD9297E-323B-4BC2-B6AE-235BA25FA39C}
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

arnottcr, 2022-02-09:
interestingly, and this may be part of the original bug, that device does not show up unless you have a thunderbolt thing plugged in. so, upon inserting my dock, these two devices show up in the normal (not just `--show-all`):
```
├─ThinkPad Thunderbolt 3 Dock:
│     Device ID:          b2a5acb6b15a09217ce299d4edda390aff1c716d
│     Current version:    15.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0108, TBT:0x0108)
│     Update Error:       Not authorized
│     GUIDs:              fab08d10-6524-5cb6-ab20-6d365c812f9e ← THUNDERBOLT\VEN_0108&DEV_1630&REV_00
│                         0f511908-5c9f-511e-9099-06823ec33614 ← THUNDERBOLT\VEN_0108&DEV_1630
│                         df675f10-53ac-59ca-bddd-0a86ee492920 ← TBT-01081630
│                         963fd29d-5e43-582e-bdc8-d7b35ef12f05 ← TBT-01081630-controller0-1
│     Device Flags:       • System requires external power source
│                         • Device stages updates
│                         • Updatable
│   
├─Thunderbolt host controller:
│     Device ID:          e52b885782126361b6d22e0cbbe67be2ba7e4d10
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    04.00
│     Vendor:             LENOVO (THUNDERBOLT:0x0109, TBT:0x0109)
│     GUIDs:              ef30993d-3ef9-5828-9c0c-4e06623354b7 ← THUNDERBOLT\VEN_0109&DEV_1601&REV_00
│                         85098c9a-1a38-5de2-b5cf-55cad3422717 ← THUNDERBOLT\VEN_0109&DEV_1601
│                         6c2c78b7-81fa-566e-b0ff-e727dbb297a5 ← TBT-01091601
│                         ae03d412-da02-5d28-a925-bff5fc3c0b21 ← TBT-01091601-controller0-0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Device stages updates
```
at which point, things started working: (hope this helps somebody)
```
# fwupdmgr get-updates
20K70004US
│
└─Thunderbolt host controller:
  │   Device ID:          e52b885782126361b6d22e0cbbe67be2ba7e4d10
  │   Summary:            Unmatched performance for high-speed I/O
  │   Current version:    04.00
  │   Vendor:             LENOVO (THUNDERBOLT:0x0109, TBT:0x0109)
  │   GUIDs:              ef30993d-3ef9-5828-9c0c-4e06623354b7 ← THUNDERBOLT\VEN_0109&DEV_1601&REV_00
  │                       85098c9a-1a38-5de2-b5cf-55cad3422717 ← THUNDERBOLT\VEN_0109&DEV_1601
  │                       6c2c78b7-81fa-566e-b0ff-e727dbb297a5 ← TBT-01091601
  │                       ae03d412-da02-5d28-a925-bff5fc3c0b21 ← TBT-01091601-controller0-0
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Device stages updates
  │ 
  └─ThinkPad T470:
        New version:      20.00
        Remote ID:        lvfs
        Release ID:       6362
        Summary:          Lenovo ThinkPad T470 Thunderbolt Firmware
        License:          Proprietary
        Size:             254.0 kB
        Created:          2020-09-08
        Urgency:          High
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Lenovo ThinkPad T470 Thunderbolt Firmware
        
        • DO NOT FORCE UPDATE Thunderbolt Controller. This may damage the firmware.
        
        This firmware is provided by LVFS community members and is not provided (or supported) by the original hardware vendor.
        
        Installing this update may also void any device warranty.
```
[... 21 lines cut ...]
║                                                                              ║
║ Thunderbolt host controller and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Decompressing…           [***************************************]
Writing…                 [***************************************]
Writing…                 [***************************************]
Restarting device…       [***************************************]
Waiting…                 [***************************************]
Successfully installed firmware
Waiting…                 [***************************************]
Idle…                    [***************************************]
Idle…                    [***************************************]
Successfully uploaded 1 report
```

closed 2022-02-09 by arnottcr

hughsie, 2022-02-09:
> that device does not show up unless you have a thunderbolt thing plugged in

Yup, that's a limitation of the thunderbolt hardware unfortunately.

arnottcr, 2022-02-09:
that said, and again this may be more issues with lenovo's branding and model overlap, it would be nice to have a little more information about this update being for both the t470 and t25 like these two are
- https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1QET.firmware
- https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1QHT.firmware

mrhpearson, 2022-02-14:
So...are we good to close this once a thunderbolt device is plugged in?
Ack on the page should flag it being for the T25 - I can let them know that but getting the update working is more important so want to check that's resolved.

hughsie, 2022-02-14:
> So...are we good to close this once a thunderbolt device is plugged in?

Ya, It sucks that we have to have a tbt device installed to update the controller, but usb4 to the rescue I guess.