fwupd/firmware-lenovo #199: Lenovo ThinkPad T480 - Me FW Downgrade - Request MeSpiLock Failed

jvanschie, 2022-02-21:
Hi,

Ubuntu recently gave me an update which cannot be installed.
I see the following message in BIOS before startup of the system.

`(A7) Me FW Downgrade - Request MeSpiLock Failed`

What does this error mean and what can I do about this?

fwupdmgr get-history
`20L5CTO1WW
│
├─Embedded Controller:
│ │   Device ID:          c19580db344e418cf2b23297bcf46d75c06e432a
│ │   Previous version:   0.1.21
│ │   Update State:       Success
│ │   Last modified:      2021-08-06 12:17
│ │   GUID:               b87a926d-189c-49a1-b1ff-921099b1de89
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T480:
│       New version:      0.1.22
│       Remote ID:        lvfs
│       Summary:          Lenovo ThinkPad T480 Embedded Controller Firmware
│       License:          Proprietary
│       Size:             791,7 kB
│       Created:          2016-07-08
│       Urgency:          High
│       Vendor:           Lenovo
│       Description:      
│       Lenovo ThinkPad T480 Embedded Controller Firmware.
│       
│       The computer will be restarted automatically after updating BIOS completely . Do NOT turn off your computer or remove the AC adaptor while update is in progress.
│       
│       Update item in this release are applicable to Windows OS only.
│     
├─System Firmware:
│ │   Device ID:          48477ba14bd9bd0c97323a07caa54b7766f917c2
│ │   Previous version:   0.1.40
│ │   Update State:       Success
│ │   Last modified:      2021-11-24 15:30
│ │   GUID:               dc7f0308-1ef8-4774-9ba5-89a58c4d731c
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T480:
│       New version:      0.1.41
│       Remote ID:        lvfs
│       Summary:          Lenovo ThinkPad T480 System Firmware
│       License:          Proprietary
│       Size:             9,5 MB
│       Created:          2016-07-08
│       Urgency:          High
│       Vendor:           Lenovo
│       Description:      
│       Lenovo ThinkPad T480 System Firmware. Update includes a security fix. Updated the Diagnostics module to version 04.18.000.
│     
└─Intel Management Engine:
  │   Device ID:          5decfff65e3391259bca4003a090e59bf50e3bea
  │   Previous version:   184.86.3909
  │   Update State:       Needs reboot
  │   Last modified:      2022-02-18 11:52
  │   GUID:               dd31d983-cf3b-4c84-acc9-70caa94e827b
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─  New version:      184.90.3987
        Remote ID:        lvfs
        License:          Unknown`
        
Thanks in advance.

rumpumpel1, 2022-03-15:
I had the same error message with a T460. The message went away after installing "Intel Management Engine 11.8 Firmware for Windows 10".

Good luck.

jekkel, 2022-03-16:
> I had the same error message with a T460. The message went away after installing "Intel Management Engine 11.8 Firmware for Windows 10".
> 
> Good luck.

Which you did using a windows boot...?

mrhpearson, 2022-04-07:
Hi @jekkel - I'm not sure about the specific error messages, but there are a bunch of ME related failed updates around this particular version with the ESRT table not being correctly updated so fwupdmgr thinks you have a different version of ME running then you really have.

Can you boot into the BIOS setup screen (F1 during early boot) and confirm if you're running 184.90.3987. If you are then your ME FW is already updated - and you just have to ignore the suggested updates.

FW team is working on a fix for this

jekkel, 2022-04-10:
@mrhpearson I did a quick check and it seems not even bios detects the version of ME correctly:

[image]

jvanschie, 2022-04-15:
@mrhpearson I have the exact same issue as @jekkel has.

Bios does not detect the firmware version.

[image]

I've also tried to reinstall the ME firmware downloaded from the drivers page of the Lenovo website. 
This also did not work:

```
sudo fwupdmgr --allow-older --allow-reinstall install Lenovo-ThinkPad-T480-ConsumerMEFirmware-11.8.90.3987.cab
Decompressing…           [***************************************]
No supported devices found
```

What can I do about this?

54ki, 2022-04-24:
@mrhpearson Just chiming in to say that I also have the same issue as @jekkel and @jvanschie. The ME firmware update from Lenovo's website does not work for me and the ME Firmware Version is blank in my BIOS.

mrhpearson, 2022-04-25:
Ewww - that's nasty.

Raised the issue to the FW team -  LO-1709. Will update when I have feedback.

Be very careful messing with the ME - it is possible to brick a system if you try and force a downgrade in some cases. I've asked for guidance on how best to recover these.

Mark

adletaw, 2022-05-14:
I am also having this issue.

jvanschie, 2022-05-19:
@mrhpearson is there any progress made on this ticket?

mrhpearson, 2022-05-19:
Afraid not - last update I had is they wanted to collect a SPI dump to evaluate, but you need some tools to collect that which you likely don't have (if you happen to have a Dediprog let me know :))

I'm waiting on next steps from the FW team - will send them a nudge

jvanschie, 2022-08-24:
Lenovo support says that I need to install Windows :(

jvanschie, 2022-08-24:
@jekkel did you found a solution already?

jekkel, 2022-08-29:
@jvanschie no, and as the machine is my daily driver for work I probably won't go the windows install or Lenovo Support way anytime soon.

FYI Just installed Lenovo System Firmware upate to v0.1.45 whose release notes contain the following:

> Fixed an issue where an incorrect Intel Management Engine firmware version might be returned at capsule update under Linux.

But the situation remains the same.

54ki, 2022-09-13:
I managed to fix this but I had to use Windows, albeit installed on an external drive.
I installed the same firmware update that @jekkel mentioned and it did not fix the issue. What I did was install a fresh copy of Windows on the external drive as to not touch my Linux stuff, fully updated it, then downloaded "Intel Management Engine 11.8 Firmware for Windows 10". After the restart, the issue went away.

jekkel, 2022-09-14:
Just tried almost the same thing today with the following differences:
* windows 11 instead of 10
* temporarily replaced the internal SSD instead of booting from USB
* used all available update streams (windows updates, lenovo vantage, website downloads)
* explicitly installed the mgmt engine v11.8 package (which succeeded but did not want to restart the system, IMO it did not flash anything) 

The problem persists.
Notably the issue was present all the time also on all windows boots.
We were unable to find any strategy to solve the issue and eventually gave up. Next step might be to involve Customer Support as @mrhpearson proposed.

buggsi, 2023-05-02:
I had the same problem on my T470p. `ME Firmware Version` was also showing up as blank in the BIOS. I tried to install the ME Firmware in linux but it gave me a weird service error and couldn't install it. So I did the following:
- Installed Win10 temporarily.
- Installed the "Intel Management Engine Components" (it's required by the Engine 11.8 Firmware).
- Ran "Intel Management Engine 11.8 Firmware for Windows 10" (make sure you download the proper installer for your laptop).
- Rebooted, the ME Firmware is now showing in the BIOS, and the `Me FW Downgrade - Request MeSpiLock Failed` error is gone.

jekkel, 2023-06-23:
FYI I recently switched machines and were able to solve the issue for my T480 by
1. Installing Windows 11
2. Installing all Lenovo Drivers through "Lenovo System Update" utility, this also installed a BIOS Update (which did not solve the issue yet)
3. Manually installing "Intel Management Engine 11.8 Formware for Windows 11, 10 (64-bit) - ThinkPad T480, T480s, X1 Carbon 6th Gen" (installer is downloaded as `n23rg21w.exe`)

I don't know whether step 1 and 2 are actually necessary though. 
Now `ME Firmware Version` shows up in setup screen and the error message on boot is gone.

schultetwin, 2023-12-14:
I also have the same issue with a Lenovo Carbon X1 5th generation.
 * Same symptoms of no ME Firmware version reported in the BIOS.
 * Same error message reported on boot: `Me FW Downgrade - Request MeSpiLock Failed`.
 * Same error output when attempting to install the firmware manually (see at the bottom, with some SPI output if that's helpful).

As an additional piece of information, when I download the [CMSE tool from intel](https://www.intel.com/content/www/us/en/download/19392/intel-converged-security-and-management-engine-version-detection-tool-intel-csmevdt.html) I can properly read the version, but it looks like it's a different code path than I expect. 

The `fwu.py` read seems to succeed with that tool:
```
UUID = '309dcde8-ccb1-4062-8f78-600115a34327'
def req_fw_ver(mei_fd):
    '''Send FW version request'''
    buf_write = struct.pack("I", 0x00000000)
    return os.write(mei_fd, buf_write)

def get_fw_ver_raw(mei_fd):
    '''Read raw FW version'''
    rlist, _, _ = select.select([mei_fd], [], [], 5)
    if not rlist:
        raise IOError(errno.ETIME, os.strerror(errno.ETIME) + "in fwu.get_fw_ver")
    buf_read = os.read(mei_fd, VER_SIZE_MAX)
    return buf_read
```

Seems to work but:

```
UUID = '8e6a6715-9abc-4043-88ef-9e39c6f63e0f'
FIXED_UUID = '55213584-9a29-4916-badf-0fb7ed682aeb'

def connect(mei_fd):
    '''Connect to dynamic MKHI client'''
    return mei.connect(mei_fd, UUID)

def connect_fixed(mei_fd):
    '''Connect to fixed MKHI client'''
    return mei.connect(mei_fd, FIXED_UUID)

VER_SIZE_SPS = 20
VER_SIZE = 28
VER_SIZE_MAX = max(VER_SIZE_SPS, VER_SIZE)
ARB_SIZE = 1076

def req_fw_ver(mei_fd):
    '''Send FW version request'''
    buf_write = struct.pack("I", 0x000002FF)
    return os.write(mei_fd, buf_write)

def fw_ver_parse(buf_read):
    '''Parse FW version'''
    if len(buf_read) == VER_SIZE:
        str_ver = struct.unpack("4BH2B2HH2B2HH2B2H", buf_read)
    elif len(buf_read) == VER_SIZE_SPS:
        str_ver = struct.unpack("4BH2B2HH2B2H", buf_read)
    else:
        raise IOError(errno.EMSGSIZE, "Wrong version message size (%d)" % len(buf_read))
    return str_ver

def get_fw_ver_str(mei_fd):
    '''Read FW version'''
    rlist, _, _ = select.select([mei_fd], [], [], 5)
    if not rlist:
        raise IOError(errno.ETIME, os.strerror(errno.ETIME) + "in mkhi.get_fw_info")
    buf = os.read(mei_fd, VER_SIZE_MAX)
    return fw_ver_parse(buf)
```

Seems to fail.

Here's the overall output running that script.

```
$ sudo ./intel_csme_version_detection_tool
Intel(R) CSME Version Detection Tool
Copyright(C) 2017-2023, Intel Corporation, All rights reserved.

Application Version: 9.0.3.0
Scan date: 2023-12-14 17:20:15 GMT

*** Host Computer Information ***
Name: lappy
Manufacturer: LENOVO
Model: 20HRCTR1WW
Processor Name: Intel(R) Core(TM) i7-7500U CPU @ 2.70GHz
OS Version: Arch Linux (6.6.6-arch1-1)

*** Intel(R) ME Information ***
Engine: Intel(R) Converged Security and Management Engine
Version: 11.8.70.3626
```

Here's what I get when I attempt to install manually:
```
$ sudo fwupdtool install ~/Downloads/n1mrl08w.cab 7ef8a531d2413174034556f12dff8aa3bb4a8c30
Loading…                 [*********                              ]17:17:11.247 FuPluginIntelMe      failed to get public key using /fpf/OemCred: write failed on timeout with status
Loading…                 [**********                             ]SPIBAR[0x00] = 0xfff0700
SPIBAR[0x04] = 0x4e800
SPIBAR[0x08] = 0x830b22
SPIBAR[0x0c] = 0x0
SPIBAR[0x10] = 0x325c003c
SPIBAR[0x14] = 0xc53
SPIBAR[0x18] = 0x6
SPIBAR[0x1c] = 0x18d
SPIBAR[0x20] = 0x5
SPIBAR[0x24] = 0xae
SPIBAR[0x28] = 0x3
SPIBAR[0x2c] = 0x928
SPIBAR[0x30] = 0x4
SPIBAR[0x34] = 0x3490
SPIBAR[0x38] = 0xf
SPIBAR[0x3c] = 0x0
SPIBAR[0x40] = 0x9
SPIBAR[0x44] = 0x42
SPIBAR[0x48] = 0xa
SPIBAR[0x4c] = 0x5f
SPIBAR[0x50] = 0x4acf
SPIBAR[0x54] = 0x0
SPIBAR[0x58] = 0xfff0700
SPIBAR[0x5c] = 0x6ff0003
SPIBAR[0x60] = 0x20001
SPIBAR[0x64] = 0x7fff
SPIBAR[0x68] = 0x7fff
SPIBAR[0x6c] = 0x7fff
SPIBAR[0x70] = 0x7fff
SPIBAR[0x74] = 0x7fff
SPIBAR[0x78] = 0x7fff
SPIBAR[0x7c] = 0x0
SPIBAR[0x80] = 0x0
SPIBAR[0x84] = 0x889f089d
SPIBAR[0x88] = 0x8e8f08a0
SPIBAR[0x8c] = 0x8e9f0e90
SPIBAR[0x90] = 0x8fff0ea0
SPIBAR[0x94] = 0x0
SPIBAR[0x98] = 0x0
SPIBAR[0x9c] = 0x0
SPIBAR[0xa0] = 0xfe000080
SPIBAR[0xa4] = 0x0
SPIBAR[0xa8] = 0x0
SPIBAR[0xac] = 0x0
SPIBAR[0xb0] = 0x0
SPIBAR[0xb4] = 0x300c
SPIBAR[0xb8] = 0x0
SPIBAR[0xbc] = 0x0
SPIBAR[0xc0] = 0x0
SPIBAR[0xc4] = 0xf3d82084
SPIBAR[0xc8] = 0x2000
SPIBAR[0xcc] = 0x2000
SPIBAR[0xd0] = 0xfff920e5
SPIBAR[0xd4] = 0x0
SPIBAR[0xd8] = 0x0
SPIBAR[0xdc] = 0x0
SPIBAR[0xe0] = 0x0
SPIBAR[0xe4] = 0x0
SPIBAR[0xe8] = 0x0
SPIBAR[0xec] = 0x0
SPIBAR[0xf0] = 0x1
SPIBAR[0xf4] = 0x0
SPIBAR[0xf8] = 0x0
SPIBAR[0xfc] = 0x0
Decompressing…           [ -                                     ]
No supported devices found
```

Happy to test any changes!

mrhpearson, 2023-12-19:
Unfortunately I think the same is going to apply here too.
Can you try the update from Windows route please? I fully appreciate how much that sucks :(

@kmauleon - just tagging you so you're aware as this is on the X1C5

kmauleon, 2023-12-20:
@schultetwin can you please download latest MEFW 11.8.94.4494 from LVFS link as below
     [Corporate](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1MRM.firmware)
     [Consumer](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1MRN.firmware)

please install manually by executing below command

fwupdmgr --allow-older --allow-reinstall install <MEFW cab file>

BTW, have you recently updated your BIOS/EC

schultetwin, 2023-12-21:
Thank you @kmauleon! Unfortunately, that doesn't work as I don't think the ME is recognized.

```
$ fwupdmgr --allow-older --allow-reinstall install
0.	Cancel
1.	349bb341230b1a86e5effe7dfe4337e1590227bd (Embedded Controller)
2.	310f45f1f223064b5c16bf6dff31146755a64480 (MZVLB512HAJQ-000L7)
3.	a45df35ac0e948ee180fe216a5f703f32dda163f (System Firmware)
```

```
$ fwupdmgr get-devices
LENOVO 20HRCTR1WW
│
├─Core™ i7-7500U CPU @ 2.70GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f4
│     Vendor:             Intel
│     GUIDs:              561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         9ca69899-3716-5857-9fd3-882a5c73236f ← CPUID\PRO_0&FAM_06&MOD_8E&STP_9
│     Device Flags:       • Internal device
│     Device Requests:    • Message
│
├─Embedded Controller:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    0.1.22
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     Problems:           • Device requires AC power to be connected
│     GUID:               74997a6b-1adf-4b12-b994-401f06ea8c72
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│     Device Requests:    • Message
│                         • Image (custom)
│
├─HD Graphics 620 (ThinkPad X1 Carbon 5th Gen):
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    02
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              38f6c009-a25c-5b50-b3b5-fef4f9c6b846 ← PCI\VEN_8086&DEV_5916
│                         06d045aa-d32e-5383-b2ad-7f8d8600c990 ← PCI\VEN_8086&DEV_5916&SUBSYS_17AA224F
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│     Device Requests:    • Message
│
├─Intel Management Engine:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.71.12
│     Minimum Version:    0.71.12
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     Problems:           • Device requires AC power to be connected
│     GUID:               ee9df30c-5d83-4ff0-8fdd-2abddae298ed
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│     Device Requests:    • Message
│                         • Image (custom)
│
├─MZVLB512HAJQ-000L7:
│     Device ID:          310f45f1f223064b5c16bf6dff31146755a64480
[... 66 lines cut ...]
│     Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│     Current version:    71.12.17568.4100
│     Vendor:             ST Microelectronics (TPM:STM)
│     Problems:           • Device requires AC power to be connected
│     GUIDs:              84df3581-f896-54d2-bd1a-372602f04c32 ← TPM\VEN_STM&DEV_0001
│                         bfaed10a-bbc1-525b-a329-35da2f63e918 ← TPM\VEN_STM&MOD_
│                         70b7b833-7e1a-550a-a291-b94a12d0f319 ← TPM\VEN_STM&DEV_0001&VER_2.0
│                         06f005e9-cb62-5d1a-82d9-13c534c53c48 ← TPM\VEN_STM&MOD_&VER_2.0
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device can recover flash failures
│                         • Updatable
│                         • Full disk encryption secrets may be invalidated when updating
│                         • Signed Payload
│     Device Requests:    • Message
│                         • Image (custom)
```

```
fwupdmgr local-install ~/Downloads/4098588d7b16c685bcecfb7f2369c46916c9217622223ad44ec6de42bebfba69-Lenovo-ThinkPad-X1Carbon5th-ConsumerMEFirmware-11.8.94.4494.cab ee9df30c-5d83-4ff0-8fdd-2abddae298ed
Selected device: Intel Management Engine
Decompressing…           [   -                                   ]
No supported devices found
```

kmauleon, 2023-12-22:
@schultetwin  this is strange... your MEFW GUID does not fall under X1 Carbon 5th.... can I have your BIOS Setup Menu screenshot please...

schultetwin, 2023-12-22:
[image]

Of course! There’s a screenshot of the BIOS.

kmauleon, 2023-12-22:
@schultetwin your system is indeed an X1 carbon 5th and MEFW version is not displaying... my BIOS level is 1.59... I will check with my environment after updating to latest BIOS.  But can I please get back to you after the holidays.  Most of us are already on vacation.  I will update here as soon as I confirmed with BIOS owners.  thank you very much and Happy Holidays!

schultetwin, 2023-12-22:
Thank you! And yes, no rush

kmauleon, 2023-12-27:
@schultetwin can you please do me a favor.... please execute below command... i would like to compare ESRT entries of our systems....thank you very much

grep . /sys/firmware/efi/esrt/entries/*/*

schultetwin, 2023-12-27:
```
$ sudo grep . /sys/firmware/efi/esrt/entries/*/*
/sys/firmware/efi/esrt/entries/entry0/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry0/fw_class:798ffd60-f10e-4ac4-8939-c8beabfe55b4
/sys/firmware/efi/esrt/entries/entry0/fw_type:1
/sys/firmware/efi/esrt/entries/entry0/fw_version:65597
/sys/firmware/efi/esrt/entries/entry0/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry0/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry0/lowest_supported_fw_version:65559
/sys/firmware/efi/esrt/entries/entry1/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry1/fw_class:74997a6b-1adf-4b12-b994-401f06ea8c72
/sys/firmware/efi/esrt/entries/entry1/fw_type:2
/sys/firmware/efi/esrt/entries/entry1/fw_version:65558
/sys/firmware/efi/esrt/entries/entry1/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry1/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry1/lowest_supported_fw_version:1
/sys/firmware/efi/esrt/entries/entry2/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry2/fw_class:ee9df30c-5d83-4ff0-8fdd-2abddae298ed
/sys/firmware/efi/esrt/entries/entry2/fw_type:2
/sys/firmware/efi/esrt/entries/entry2/fw_version:4653068
/sys/firmware/efi/esrt/entries/entry2/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry2/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry2/lowest_supported_fw_version:4653068
```

kmauleon, 2023-12-27:
@schultetwin already forwarded to BIOS dev, they will check.  Will feedback once I receive any update.  Thank you very much.

schultetwin, 2023-12-27:
Great, thank you!

schultetwin, 2024-01-03:
@kmauleon  any updates from the BIOS dev? Happy to do any testing needed!

kmauleon, 2024-01-04:
@schultetwin I haven't heard back yet from BIOS dev... I will follow up... though most of the guys are still on vacation this week

schultetwin, 2024-01-04:
Got it, no worries! I’ll check in later next week!

kmauleon, 2024-01-16:
hi @schultetwin BIOS dev was not able to replicate the issue.  Can you please execute the following command. thank you very much.

fwupdtool hwids

schultetwin, 2024-01-17:
I'm also happy to do any dumps of the bios if that's useful!

```
$ sudo fwupdtool hwids
[sudo] password for mark:
Loading…                 [*                                      ]
Computer Information
--------------------
BiosVendor: LENOVO
BiosVersion: N1MET76W (1.61 )
BiosMajorRelease: 1
BiosMinorRelease: 61
FirmwareMajorRelease: 01
FirmwareMinorRelease: 16
Manufacturer: LENOVO
Family: ThinkPad X1 Carbon
ProductName: 20HRCTR1WW
ProductSku: LENOVO_MT_20HR_BU_Think_FM_ThinkPad X1 Carbon
EnclosureKind: a
BaseboardManufacturer: LENOVO
BaseboardProduct: 20HRCTR1WW
Hardware IDs
------------
{07a84f59-ec46-5959-9b6b-7811d09e3f7d}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{8a9bbd30-751a-5b1d-819d-f9e94663b453}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{daede04f-c724-588b-ab7a-3f1cffd9b85a}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{b57b5bf8-c08c-564f-9f24-131f5e7f98a4}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{36a75e60-fed4-5538-aa02-1ef826faf8f3}   <- Manufacturer + Family + ProductName + ProductSku
{0229dfd6-9cd9-53ef-9ae7-0e4a8d504045}   <- Manufacturer + Family + ProductName
{4794f55d-4eda-56d6-ae87-0b7dc19eddf0}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{c21fc7db-6670-575c-88ea-20bafe528c1c}   <- Manufacturer + ProductSku
{08bc2607-3ea3-545a-ac53-c24089d2c177}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{0e7ad605-9dbb-5fbf-8636-e081b052aee2}   <- Manufacturer + ProductName
{70273ad1-cc65-5924-94c6-b72c5b1a74be}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{85d2f115-5a9d-5e69-bbf6-21e4e3e77f69}   <- Manufacturer + Family
{e093d715-70f7-51f4-b6c8-b4a7e31def85}   <- Manufacturer + EnclosureKind
{cae81797-e01d-5e14-9906-991d1ece13cf}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{6de5d951-d755-576b-bd09-c5cf66b27234}   <- Manufacturer
```

kmauleon, 2024-01-18:
@schultetwin thank you very much for the information... I will forward to BIOS owner

kmauleon, 2024-01-19:
@schultetwin would you please try to downgrade BIOS version to 1.58
please download here >> https://fwupd.org/downloads/c0d5b89a368794fefd6c10d61a2f1f9ee6732bb76c1f9db6481750094dbf474f-Lenovo-ThinkPad-X1Carbon5th-SystemFirmware-1.58.cab

and then execute the command below... please make sure AC is attached and battery at least 25%

[image]

schultetwin, 2024-01-19:
Thanks! I did the downgrade:

```
fwupdmgr --allow-older --allow-reinstall install ~/Downloads/c0d5b89a368794fefd6c10d61a2f1f9ee6732bb76c1f9db6481750094dbf474f-Lenovo-ThinkPad-X1Carbon5th-SystemFirmware-1.58.cab
```

And confirmed that the downgrade worked

```
$ fwupdmgr get-devices a45df35ac0e948ee180fe216a5f703f32dda163f
LENOVO 20HRCTR1WW
│
└─System Firmware:
      Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
      Summary:            UEFI ESRT device
      Current version:    0.1.58
      Minimum Version:    0.1.23
      Vendor:             Lenovo (DMI:LENOVO)
      Update State:       Success
      GUID:               798ffd60-f10e-4ac4-8939-c8beabfe55b4
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Cryptographic hash verification is available
                          • Device is usable for the duration of the update
```

I still get the error:

```
$ fwupdmgr --allow-older --allow-reinstall install  ~/Downloads/4098588d7b16c685bcecfb7f2369c46916c9217622223ad44ec6de42bebfba69-Lenovo-ThinkPad-X1Carbon5th-ConsumerMEFirmware-11.8.94.4494.cab
Decompressing…           [   -                                   ]
No supported devices found
```

kmauleon, 2024-01-19:
Can we have your fwupdmgr get-devices and BIOS Setup Menu details again... @schultetwin thank you

schultetwin, 2024-01-19:
```
$ fwupdmgr get-devices
LENOVO 20HRCTR1WW
│
├─Core™ i7-7500U CPU @ 2.70GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f4
│     Vendor:             Intel
│     GUIDs:              561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         9ca69899-3716-5857-9fd3-882a5c73236f ← CPUID\PRO_0&FAM_06&MOD_8E&STP_9
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    0.1.22
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     Problems:           [31m[1m• Device cannot be used while the lid is closed[0m
│     GUID:               74997a6b-1adf-4b12-b994-401f06ea8c72
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─HD Graphics 620 (ThinkPad X1 Carbon 5th Gen):
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    02
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              38f6c009-a25c-5b50-b3b5-fef4f9c6b846 ← PCI\VEN_8086&DEV_5916
│                         06d045aa-d32e-5383-b2ad-7f8d8600c990 ← PCI\VEN_8086&DEV_5916&SUBSYS_17AA224F
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Intel Management Engine:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.71.12
│     Minimum Version:    0.71.12
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     Problems:           [31m[1m• Device cannot be used while the lid is closed[0m
│     GUID:               ee9df30c-5d83-4ff0-8fdd-2abddae298ed
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─MZVLB512HAJQ-000L7:
│     Device ID:          310f45f1f223064b5c16bf6dff31146755a64480
│     Summary:            NVM Express solid state drive
│     Current version:    5L2QEXA7
│     Vendor:             Samsung (NVME:0x144D)
│     Serial Number:      S3TNNF0K469691
│     GUIDs:              47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                         c9d531ea-ee7d-5562-8def-c64d0d144813 ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801
│                         79d6cfae-a5a2-5936-9248-5aebd23480f7 ← SAMSUNG MZVLB512HAJQ-000L7
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─System Firmware:
[... 78 lines cut ...]
│     Vendor:             GenesysLogic (USB:0x05E3)
│     Install Duration:   9 seconds
│     GUIDs:              953708c0-47e9-5442-af29-23193e50e792 ← USB\VID_05E3&PID_0610&IC_352310
│                         18d26dfd-d441-500d-a490-544e34ecd977 ← USB\VID_05E3&PID_0610&IC_352310&BONDING_0F
│                         1710d561-3840-5b51-8fc1-34137402cf54 ← USB\VID_05E3&PID_0610&VENDOR_GENESYSLOGIC&IC_352310&BONDING_0F&PORTNUM_23&VENDORSUP_CDB1ABD6-7C7A-5360-B47F-49A6AFE05EBF
│     Device Flags:       • Updatable
│                         • Cryptographic hash verification is available
│                         • Unsigned Payload
│   
└─Unifying Receiver:
      Device ID:          803edc5f79ff8bed710facf7e043a54b553523cc
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR24.10_B0036
      Bootloader Version: BOT03.01_B0008
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:       • Updatable
                          • Supported on remote server
                          • Unsigned Payload
    
```

And picture of BIOS coming soon!

schultetwin, 2024-01-19:
[image]

kmauleon, 2024-01-19:
@schultetwin the "no supported devices found" is expected since your MEFW GUID is mismatched.... MEFW GUID should be either of these
[image]

however, the version format shown by fwupd is incorrect as well.... BIOS owner still checking this further... thank you

kmauleon, 2024-02-07:
hi @schultetwin sorry to keep you waiting... after further checking by BIOS owner, their recommendation is to send the system for service repair as below T_T
---------------------------------------------------
This user’s machine doesn’t have OA2 marker.  
 
My guess is that this machine has cleared system data when updating the old BIOS by the new fwupd tool.
It's an attachment issue.

Unfortunately, after this problem, the data on the system is corrupt and can only be repaired by taking the machine to the service.
--------------------------------------------------

schultetwin, 2024-02-09:
Got it, thank you. Any chance you can provide the instructions on what they do in service to repair this? I’m more than comfortable opening up the laptop and flashing/programming anything. If not, where should I send it into repair?

mrhpearson, 2024-02-13:
Afraid I don't have any guidance other than to call customer support and they will help you out. We don't have the tools that we can share to be able to reprogram this outside of Lenovo.

When creating the support ticket, I recommend referencing this ticket and specifically the fact that the OA2 marker is missing. It can be useful to reference Kathereene or myself - we can sometimes save the support people doing the debug exercise that they would normally walk thru, as the issue has been analysed by the FW team.

schultetwin, 2024-02-13:
Thank you @mrhpearson. It looks like there's a minimum $109 fee for a support ticket, and this is not worth that to me.

myllynen, 2024-03-01:
FWIW, this happened also with a Lenovo ThinkPad X1 Carbon Gen 6 laptop, the symptoms are exactly as described above. I created a Windows To Go USB stick hoping that I could install the fix from the Windows exe fail that way but it took hours to boot before eventually getting stuck so that idea didn't work out. In any case, this doesn't issue seem to be related to one or two faulty units since several people are affected by it. Thanks.

zlatin, 2024-04-07:
> FYI I recently switched machines and were able to solve the issue for my T480 by
> 1. Installing Windows 11
> 2. Installing all Lenovo Drivers through "Lenovo System Update" utility, this also installed [...]

In my case updating BIOS via Lenovo System Utility is what led to the issue.
Installing the Intel management engine fixed the issue

Bohosul01, 2024-09-24:
> I also have the same issue with a Lenovo Carbon X1 5th generation.
>
> * Same symptoms of no ME Firmware version reported in the BIOS.
> * Same error message reported on boot: `Me FW Downgrade - Request [...]

Hi friend. I can't throw any light on this matter, but I am also having problems with the intel ME firmware and I would like to know where I can find the file you mention here (fwu.py)

kmauleon, 2024-09-26:
hi @Bohosul01 for X1 Carbon 5th you may check this link to install BIOS via USB
https://pcsupport.lenovo.com/ph/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-type-20hr-20hq/downloads/ds120390-bios-update-utility-bootable-cd-for-windows-10-81-64-bit-7-32-bit-64-bit-linux-thinkpad-x1-carbon-type-20hq-20hr-20k3-20k4?category=BIOS%2FUEFI