fwupd/firmware-lenovo #201: System and ME firmware fail to update on Lenovo t480s

shenki, 2022-02-28:


hughsie, 2022-02-28:
@mrhpearson the use of ShellFlash64 arguments does worry me slightly...

mrhpearson, 2022-03-01:
I've raised LO-1598 with the FW team for their input, but shellflash is (I believe) used for the BIOS update, and it looks like there's been almost 14k succesfull downloads for this update - so there is something wonky in this case.

Can you try doing just the system update by itself?

@hughsie - why is the use of arguments a concern? I know when I do updates from UEFI shell (which I do quite a bit on prototype systems before they're released) there are a number of arguments that are needed. Is there something that has changed that might be breaking that as I think it's pretty common on our platforms?

Mark

sydneymeyer, 2022-03-01:
Perhaps related: https://github.com/fwupd/firmware-lenovo/issues/155

dklementowski, 2022-03-04:
I'm affected as well on X1 Carbon 6 20KG. Firmware updates used to work without any problem and I haven't change anything, just upgrading my OS as usual. I'm on Arch Linux, my installed packages are as follows:

```
➜  ~ pacman -Qs fwupd 
local/fwupd 1.7.6-1
    Simple daemon to allow session software to update firmware
local/fwupd-efi 1.2-1
    EFI Application used by uefi-capsule plugin in fwupd
```

Just like OP, when I try to update, it hangs for a moment and then shows the error message for a short period of time. After that it disappears and reboots.

[image]

Firmware files on the EFI partition:
```
➜  ~ tree /boot/EFI/arch
/boot/EFI/arch
├── fw
│   ├── fwupd-9c9d9769-32fa-4841-b550-ea998e754e99.cap
│   └── fwupd-a4b51dca-8f97-4310-8821-3330f83c9135.cap
├── fwupdx64.efi
└── grubx64.efi

1 directory, 4 files
```

sudopluto, 2022-03-12:
i have a similar issue for the IME with my t480, but when i used the tool mentioned in this thread, the update was indeed applied: https://github.com/fwupd/firmware-lenovo/issues/140#issuecomment-1053867852

sudopluto, 2022-03-12:
another use on the t480 on the same thread got similar shellfish error: https://github.com/fwupd/firmware-lenovo/issues/140#issuecomment-981053810. i did as well but didn't think to capture it

gevhaz, 2022-04-07:
I have the same issue with the same setup as @dklementowski. Lenovo X1 Carbon 6 running Arch Linux, also with fwupd v1.7.6-1.

ChiWei-Chen, 2022-04-27:
Hi All,
I got a trial BIOS for T480, T480s and X1 carbon 6. If you're willing to test it, could you send me a mail? ( cchen50 at lenovo dot com ) or ( mpearson at lenovo dot com )

As Lenovo FW team cannot reporduce the issue, need your help to try this trial BIOS which shows more debug information. 

Here is the expected command: fwupdmgr install <*.cab>
This version couldn't update to the latest version. It would be stop at at Lenovo logo.
Please capture the Boot mode and Flash more printed on the screen for investigation as follows.
[image]

Thank you very much.

kleinph, 2022-05-03:
Somehow this issue disappeared for me. `fwupdmgr get-updates` does not show any outstanding updates on my T480s anymore.

rotanid, 2022-05-18:
on my T480s i also was able to update to system firmware 0.1.49 by specifying the ID explicitly.
the ME update still does not work, but there is no error message seen as before, it just does boot again with the old ME instead of upgrading.

benblasco, 2022-05-26:
> lenovo

What was the exact command you used to run the update successfully @rotanid?

mkock, 2022-06-02:
I see this issue as well, on Lenovo X1 Carbon 6th. gen. Using Kubuntu 22.04 but I believe the issue began before I upgraded from 21.10.

rotanid, 2022-07-01:
> > lenovo
> 
> What was the exact command you used to run the update successfully @rotanid?

first i mounted my EFI partition (if not already mounted).
i did the following commands, used the ID given for the system firmware by "get-updates", maybe another ID for your system:

```
sudo fwupdmgr refresh --force
sudo fwupdmgr get-updates
sudo fwupdmgr update ebfe8df8-dee7-4692-a721-cbcf5095c5cf
```

this way i also updated the system firmware to 0.1.50 today.
the ME update still does not work, at this time the update process starts after the system resets but it aborts soon after starting and the system boots back into normal operation. tried 3 times...

bzed, 2022-07-14:
Same issue on x390...

ChiWei-Chen, 2022-07-15:
@rotanid , I know the BIOS team is rolling out a fix with new BIOS version for T480/T480s/X1C6, it needs some weeks to complete the internal release process then push it to LVFS, thanks.

@bzed , thanks for reporting the X390 issue, could you provide more detailed information? Thanks!

- fwupdmgr update --show-all
- system firmware version and Intel Management Engine firmware version in BIOS main page
- OS and fwupdmgr version

ChiWei-Chen, 2022-07-15:
> @rotanid , I know the BIOS team is rolling out a fix with new BIOS version for T480/T480s/X1C6, it needs some weeks to complete the internal release process then push it to LVFS, thanks.
> 
> @bzed , thanks for reporting the X390 issue, could you provide more detailed information? Thanks! (LO-1598 for my tracking)
> 
> * fwupdmgr update --show-all
> * system firmware version and Intel Management Engine firmware version in BIOS main page
> * OS and fwupdmgr version

mutax, 2022-07-15:
Same error with ShellFlash64 on T470s

ChiWei-Chen, 2022-07-15:
@mutax thanks for reporting.
Could you provide more detail as my comment above of the ShellFlash64 issue on T470s? Thanks!

bzed, 2022-07-15:
x390:
```
Devices with no available firmware updates: 
 • USB2.0 Hub
 • USB2.0 Hub
 • USB3.1 Hub
 • USB3.1 Hub
 • ThinkPad Thunderbolt 3 Dock USB Audio
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 192.47.1524 to 192.85.1869?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Update Management Engine Firmware version to 12.0.85.1869.                 ║
║                                                                              ║
║ 20Q0S0P500 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

```
```
BIOS Information
        Vendor: LENOVO
        Version: N2JET92W (1.70 )
        Release Date: 08/31/2020
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                PNP is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                3.5"/720 kB floppy services are supported (int 13h)
                Print screen service is supported (int 5h)
                8042 keyboard services are supported (int 9h)
                Serial services are supported (int 14h)
                Printer services are supported (int 17h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.70
        Firmware Revision: 1.15
```

```
➜  ~ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.18.0-2-amd64
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.gusb          0.3.10
➜  ~ 
```

Debian testing/unstable

ChiWei-Chen, 2022-07-19:
@mutax , thanks for the feedback.
It seems there are 2 different issue here. 
First is ShellFlash64 issue with BIOS upgrade.
Second is ESRT ME version incorrect issue which is tracked in #189.
The FW team replied the official BIOS will be released a few weeks later, I got a new trial BIOS for T470&T470s, If you or someone is willing to test that it would be appreciated - please send me an email. Thanks!
(cchen50 at lenovo dot com)

ChiWei-Chen, 2022-07-19:
@bzed thank you for sharing the information, I'll give you back if any feedback from FW team

ChiWei-Chen, 2022-07-20:
@mutax , thanks for the assistance. I've sent you the link to trial BIOS, please let me know if this doesn't work for you.
Please try to update ME Firmware *after* the trial BIOS is installed. Thanks!

mutax, 2022-07-20:
> @mutax , thanks for the assistance. I've sent you the link to trial BIOS, please let me know if this doesn't work for you. Please try to update ME Firmware _after_ the trial BIOS is installed. Thanks!

manually installing the cab you sent me worked flawlessly, fwupdmgr shows that the ME firmware is at the latest version while at the same time recorded that the update failed last time. Should I try to manually install the same version (11.8.92.4222) again?

BIOS Info screen after only updating with the cab you provided:

```
UEFI BIOS Version                 N1WET63W (1.42 )    N1WET68T  (1.47 )
UEFI BIOS Date (Year-Month-Date)  2021-08-24          2022-07-18
Embedded Controller Version       N1WHT40W (1.22 )    same as before
ME Firmware Version               11.8.92.4222        same as before
Machine Type Model                20HFCTO1WW`

```

ChiWei-Chen, 2022-07-21:
@mutax ,
> manually installing the cab you sent me worked flawlessly, fwupdmgr shows that the ME firmware is at the latest version while at the same time recorded that the update failed last time. Should I try to manually install the same version (11.8.92.4222) again?
Thanks for the feedback.
Is it still shows 184.86.3909 in fwupdmgr? If this is the case, please manually install it again. Thanks!

mutax, 2022-07-21:
> Thanks for the feedback. Is it still shows 184.86.3909 in fwupdmgr? If this is the case, please manually install it again. Thanks!

It shows: 
```
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    184.92.4222
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               d95d3ada-eef1-464f-8a2a-a11232b8556b
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
```
But I will reinstall it because it seems weird that it complained before that it was unable to apply the update and now it is suddenly there.

ChiWei-Chen, 2022-07-21:
@mutax thanks, please let me know if you see any other problem there.

mutax, 2022-07-22:
> @mutax thanks, please let me know if you see any other problem there.

All perfect! Thanks a lot! Not noticing any issues.

mrhpearson, 2022-08-29:
I think we're good to close this one.

closed 2022-08-29 by mrhpearson