fwupd/firmware-lenovo #202: Update of ME firmware fails on Thinkpad X1 Carbon 6th gen

95440b97d, 2022-02-28:
Update of ME firmware fails on Thinkpad X1 Carbon 6th gen.  The firmware was downloaded (fwupd-9c9d9769-32fa-4841-b550-ea998e754e99.cap), the computer rebooted, nothing happened for a while, then message appears "Reset system", computer rebooted successfully.

Secure boot is on and /boot/EFI/arch/fwupdx64.efi is properly signed (however it is obviously not a problem with Secure Boot).

```shell
efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-9c9d9769-32fa-4841-b550-ea998e754e99-0
```

```shell
tree /boot/
/boot/
├── BOOT
│   └── EFI
│       ├── BOOTX64.EFI
│       
├── EFI
│   └── arch
│       ├── fw
│       │   └── fwupd-9c9d9769-32fa-4841-b550-ea998e754e99.cap
│       └── fwupdx64.efi
│       
```

```shell
pacman -Q fwupd
fwupd 1.7.5-1
```

```shell
efibootmgr -v # truncated...
BootCurrent: 0001
Boot0001* Arch Linux Zen...
Boot0003* Linux-Firmware-Updater	HD(1,GPT,9a7c71d0-9612-324d-8b29-bd67daf841d2,0x800,0x82000)/File(\EFI\arch\fwupdx64.efi)
Boot0005* Arch Linux LTS (Rescue)...
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
```

```shell
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.5
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.16.11-zen1-1-zen
compile   org.freedesktop.fwupd         1.7.5
runtime   org.freedesktop.gusb          0.3.10
```

```shell
dmidecode -s system-version
ThinkPad X1 Carbon 6th
```

```shell
dmidecode -s system-product-name
20KHCTO1WW
```

```shell
fwupdmgr get-updates

20KHCTO1WW
│
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Summary:            UEFI ESRT device
  │   Current version:    184.86.3909
  │   Minimum Version:    184.86.3909
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUIDs:              9c9d9769-32fa-4841-b550-ea998e754e99
  │                       1a4eea87-292a-5d99-82bd-fd244d4e0194 ← UEFI\RES_{9C9D9769-32FA-4841-B550-EA998E754E99}
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 6th:
        New version:      184.90.3987
        Remote ID:        lvfs
        Release ID:       11629
        Summary:          Lenovo ThinkPad X1 Carbon 6th Consumer ME Firmware
        License:          Proprietary
        Size:             2.2 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N23RN19W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        • 8 Intel Platform Update 2021.2
        
        Version 11.8.90.3987 Fixes the  IPU 2021.2 security vulnerabilities. please refer to Issues List.
        
        This firmware is provided by LVFS community members and is not provided (or supported) by the original hardware vendor.
        
        Installing this update may also void any device warranty.
```

```shell
fwupdmgr get-history
20KHCTO1WW
│
...truncated...
│
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Previous version:   184.86.3909
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2022-02-25 20:52
  │   GUID:               9c9d9769-32fa-4841-b550-ea998e754e99
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 6th:
        New version:      184.90.3987
        Remote ID:        lvfs
        Release ID:       11629
        Summary:          Lenovo ThinkPad X1 Carbon 6th Consumer ME Firmware
        License:          Proprietary
        Size:             2.2 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N23RN19W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        • 8 Intel Platform Update 2021.2
        
        Version 11.8.90.3987 Fixes the  IPU 2021.2 security vulnerabilities. please refer to Issues List.
        
        This firmware is provided by LVFS community members and is not provided (or supported) by the original hardware vendor.
        
        Installing this update may also void any device warranty.
``` 

I did some investigation. Here's what I found. I downloaded the firmware directly from Lenovos support page (link below).  After unzipping I ran:

```shell
fwupdmgr get-details Lenovo-ThinkPad-X1Carbon6th-CorporateMEFirmware-11.8.90.3987.cab

20KHCTO1WW
│
└─Unknown Device:
  │   Description:        
  │   Lenovo ThinkPad X1 Carbon 6th Corporate ME Firmware N23RM19W, 11.8.90.3987
  │   Update Error:       Device was not found
  │   GUID:               42a0a96e-c9f3-438f-9687-7826be33e4ce
  │ 
  └─ThinkPad X1 Carbon 6th:
        New version:      184.90.3987
        Summary:          Lenovo ThinkPad X1 Carbon 6th Corporate ME Firmware
        License:          Proprietary
        Size:             7.5 MB
        Created:          2016-07-08
        Urgency:          High
        Vendor:           Lenovo Ltd.
        Description:      
        11.8 Intel Platform Update 2021.1
        
        Version 11.8.90.3987 (LVFS: 184.86.3909)
        
        Problem Fixes
        
        • Intel CSME INTEL-TA-00539 IPU 2021.2:
        • Mitigated the following security vulnerabilities:
        
        This firmware is provided by LVFS community members and is not provided (or supported) by the original hardware vendor.
        
        Installing this update may also void any device warranty.
```

When I run the following on the cap file downloaded through fwupdmgr I get:

```shell
fwupdmgr get-details fwupd-9c9d9769-32fa-4841-b550-ea998e754e99.cap
Decompressing ... The input is not of cabinet format
```

Lenovo Product Page (ME firmware update):
  https://pcsupport.lenovo.com/ca/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-6th-gen-type-20kh-20kg/20kh/20khcto1ww/downloads/ds551954-intel-management-engine-118-firmware-corporate-update-utility-for-linux-thinkpad-x1-carbon-6th-type-20kh-20kg

johannbg, 2022-03-01:
I'm experiencing the same failure ( as are seemingly others on a different issue ), it does not matter if it's the update from lvfs and or Lenovo site ( tried both the consumer which got rejected and the corporate which was accepted ) so both updating via lvfs and or manually with the cab files from lenovo site fails with reset system ( which arguably is good since it should narrow it down to the cab file itself ).

johannbg, 2022-03-01:
maybe @mrhpearson can ( or have someone ) duplicate this.

mrhpearson, 2022-03-01:
Hi,
I don't have an X1 Carbon 6 unfortunately. I have raised a ticket to the FW team for their input (LO-1597)
Couple of suggestions
1) Could you check in the BIOS setup information screen (F1 during early boot) to see what that reports as the ME FW version.
2) Could you try the update with Secure Boot disabled. Even with correctly signed updaters we've had some issues with shim and fwupd in the last year and whilst I don't think you're hitting that here - but I'd like to rule it out.

I'll update when I hear back from the FW team with any other questions or suggestions
Mark

johannbg, 2022-03-01:
@mrhpearson ME FW version in bios is listed as 11.8.90.3987 but the current version as reported by fwupdmgr get-updates is 184.86.3909 and it proposes an updated versioned 184.90.3987 created on 2016-07-08  ( Release ID: 11630 ) o_O

mrhpearson, 2022-03-01:
Hi,
OK - so your FW has been updated - it's just not reporting the version correctly to the OS in the ESRT tables.

Thanks for the details. I'll pass those on to the FW team
Mark

mrhpearson, 2022-03-02:
Can we check the contents of the esrt tables please?
cat /sys/firmware/efi/esrt/entries/entry?/fw_class
cat /sys/firmware/efi/esrt/entries/entry?/fw_version

Thanks
Mark

johannbg, 2022-03-02:
> Can we check the contents of the esrt tables please? cat /sys/firmware/efi/esrt/entries/entry?/fw_class cat /sys/firmware/efi/esrt/entries/entry?/fw_version
> 
> Thanks Mark

`
cat /sys/firmware/efi/esrt/entries/entry?/fw_class
`
`
a4b51dca-8f97-4310-8821-3330f83c9135
`
`
42a0a96e-c9f3-438f-9687-7826be33e4ce
`
`
3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
`

`
cat /sys/firmware/efi/esrt/entries/entry?/fw_version
`
`
65591
`
`
3092647749
`
`
65558
`

mrhpearson, 2022-03-02:
OK - I'm officially confused.

You're ESRT table has GUID 42a0a96e-c9f3-438f-9687-7826be33e4ce as the 2nd entry - and that says the FW ID is 184.86.3909 (once you do the number munging). Which is good - if you have a vPro machine.
So that should means you should be using the Corporate/vPro version of FW - i.e 23RM:
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN23RM.firmware

@hughsie - I'm missing something. Why would fwupdmgr get-devices show the consumer non-vPro GUID (9c9d9769-32fa-4841-b550-ea998e754e99). I thought it pulled this info from the ESRT tables? What am I missing?

Mark

johannbg, 2022-03-02:
@mrhpearson yes I have a vPro machine, and yes fwup denied me to install the consumer firmware but allowed me to install the corporate one ( which is the correct behaviour I think). 

And I think you are confusing me with the guid of the original reporter since I dont have any GUID equalling to `9c9d9769-32fa-4841-b550-ea998e754e99`
[image]

johannbg, 2022-03-02:
@95440b97d could you provde the esrt tables on your machine as @mrhpearson requested

mrhpearson, 2022-03-03:
My apologies - I was trying to look at this between meetings and it seems I didn't engage my brain.

Your ESRT table is wrong. If BIOS says you have 184.90.3987 and ESRT table says 184.86.3909 then the ESRT table is incorrect and everything flows from there. 

Pretty much ignore my comment above...it's brain dead. I'll follow up with the FW team

sydneymeyer, 2022-03-03:
If it helps, here's my output too.. Same issue, also x1 6th gen, non-vPro.

`cat /sys/firmware/efi/esrt/entries/entry?/fw_class`
`a4b51dca-8f97-4310-8821-3330f83c9135`
`9c9d9769-32fa-4841-b550-ea998e754e99`
`3babca5f-b2bf-4f4b-a72e-2bdc84eb4019`

`cat /sys/firmware/efi/esrt/entries/entry?/fw_version`
`65591`
`3092647749`
`65558`

mrhpearson, 2022-03-04:
3092647749 = 0xB8560F45 = 184.86.3909
If you can confirm from the BIOS setup screen (F1 during early boot) which ME version is reported there we'll know you have the same issue

sydneymeyer, 2022-03-04:
@mrhpearson i'm not fully sure if you were asking me, but, yes, it would appear so.

BIOS says "ME Firmware Version 11.8.90.3987".

[image]

fwupd output:

```
20KHCTO1WW
│
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Summary:            UEFI ESRT device
  │   Current version:    184.86.3909
  │   Minimum Version:    184.86.3909
  │   Vendor:             DMI:LENOVO
  │   Update State:       Success
  │   GUIDs:              9c9d9769-32fa-4841-b550-ea998e754e99
  │                       1a4eea87-292a-5d99-82bd-fd244d4e0194
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad X1 Carbon 6th:
        New version:      184.90.3987
        Remote ID:        lvfs
        Release ID:       11629
        Summary:          Lenovo ThinkPad X1 Carbon 6th Consumer ME Firmware
        License:          Proprietary
        Size:             2.2 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N23RN19W
        Vendor:           Lenovo
        Flags:            is-upgrade
        Description:
        • 8 Intel Platform Update 2021.2

        Version 11.8.90.3987 Fixes the  IPU 2021.2 security vulnerabilities. please refer to Issues List.
```

dmvianna, 2022-03-06:
My BIOS looks the same, and `fwupdmgr` also keeps asking to update
``` sudo fwupdmgr update         
Devices with no available firmware updates: 
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 184.86.3909 to 184.90.3987?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ • 8 Intel Platform Update 2021.2                                             ║
║                                                                              ║
║ Version 11.8.90.3987 Fixes the  IPU 2021.2 security vulnerabilities. please  ║
║ refer to Issues List.                                                        ║
║                                                                              ║
║ 20KH000FAU must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
```
Secure boot is off and boot order lock is disabled.

mrhpearson, 2022-03-10:
Hi - looks like we'll have a test BIOS shortly, but in the meantime would you mind collectiong the remaining ESRT details for the bad entry (the one that has the ME GUID). Using this should work:
`sudo find /sys/firmware/efi/esrt/entries/entry?/ -type f -exec cat {} \; -print`
(replace the ? with the entry number if you know which entry is the one that matters)

Thanks
Mark

sydneymeyer, 2022-03-10:
> replace the ? with the entry number if you know which entry is the one that matters

I don't. Here's all of them then.

```
a4b51dca-8f97-4310-8821-3330f83c9135
/sys/firmware/efi/esrt/entries/entry0/fw_class
1
/sys/firmware/efi/esrt/entries/entry0/lowest_supported_fw_version
0
/sys/firmware/efi/esrt/entries/entry0/last_attempt_version
0
/sys/firmware/efi/esrt/entries/entry0/last_attempt_status
1
/sys/firmware/efi/esrt/entries/entry0/fw_type
65591
/sys/firmware/efi/esrt/entries/entry0/fw_version
0x0
/sys/firmware/efi/esrt/entries/entry0/capsule_flags
9c9d9769-32fa-4841-b550-ea998e754e99
/sys/firmware/efi/esrt/entries/entry1/fw_class
3092647749
/sys/firmware/efi/esrt/entries/entry1/lowest_supported_fw_version
3092647749
/sys/firmware/efi/esrt/entries/entry1/last_attempt_version
0
/sys/firmware/efi/esrt/entries/entry1/last_attempt_status
2
/sys/firmware/efi/esrt/entries/entry1/fw_type
3092647749
/sys/firmware/efi/esrt/entries/entry1/fw_version
0x8010
/sys/firmware/efi/esrt/entries/entry1/capsule_flags
3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
/sys/firmware/efi/esrt/entries/entry2/fw_class
65558
/sys/firmware/efi/esrt/entries/entry2/lowest_supported_fw_version
0
/sys/firmware/efi/esrt/entries/entry2/last_attempt_version
0
/sys/firmware/efi/esrt/entries/entry2/last_attempt_status
2
/sys/firmware/efi/esrt/entries/entry2/fw_type
65558
/sys/firmware/efi/esrt/entries/entry2/fw_version
0x0
/sys/firmware/efi/esrt/entries/entry2/capsule_flags

```

mrhpearson, 2022-03-11:
I have a TrialBIOS from the FW team for the T480. If someone is willing to test that it would be appreciated - send me an email (markpearson at lenovo dot com)
Installation will be by fwupdmgr and it will involve updating the BIOS and then
 - trying a ME FW update
 - If that doesn't work, downgrading ME and then upgrading again
Thanks
Mark

95440b97d, 2022-03-11:
>`sudo find /sys/firmware/efi/esrt/entries/entry?/ -type f -exec cat {} \; -print`

99d47dfe-68fc-4a6f-b541-32e003c886fb
/sys/firmware/efi/esrt/entries/entry0/fw_class
1
/sys/firmware/efi/esrt/entries/entry0/lowest_supported_fw_version
0
/sys/firmware/efi/esrt/entries/entry0/last_attempt_version
0
/sys/firmware/efi/esrt/entries/entry0/last_attempt_status
1
/sys/firmware/efi/esrt/entries/entry0/fw_type
65590
/sys/firmware/efi/esrt/entries/entry0/fw_version
0x0
/sys/firmware/efi/esrt/entries/entry0/capsule_flags
9c9d9769-32fa-4841-b550-ea998e754e99
/sys/firmware/efi/esrt/entries/entry1/fw_class
3092647749
/sys/firmware/efi/esrt/entries/entry1/lowest_supported_fw_version
3092647749
/sys/firmware/efi/esrt/entries/entry1/last_attempt_version
0
/sys/firmware/efi/esrt/entries/entry1/last_attempt_status
2
/sys/firmware/efi/esrt/entries/entry1/fw_type
3092647749
/sys/firmware/efi/esrt/entries/entry1/fw_version
0x8010
/sys/firmware/efi/esrt/entries/entry1/capsule_flags
3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
/sys/firmware/efi/esrt/entries/entry2/fw_class
65558
/sys/firmware/efi/esrt/entries/entry2/lowest_supported_fw_version
0
/sys/firmware/efi/esrt/entries/entry2/last_attempt_version
0
/sys/firmware/efi/esrt/entries/entry2/last_attempt_status
2
/sys/firmware/efi/esrt/entries/entry2/fw_type
65558
/sys/firmware/efi/esrt/entries/entry2/fw_version
0x0
/sys/firmware/efi/esrt/entries/entry2/capsule_flags
cea87551-1701-43fb-afbc-6e8ce9728345
/sys/firmware/efi/esrt/entries/entry3/fw_class
4784148
/sys/firmware/efi/esrt/entries/entry3/lowest_supported_fw_version
0
/sys/firmware/efi/esrt/entries/entry3/last_attempt_version
0
/sys/firmware/efi/esrt/entries/entry3/last_attempt_status
2
/sys/firmware/efi/esrt/entries/entry3/fw_type
4784148
/sys/firmware/efi/esrt/entries/entry3/fw_version
0x0
/sys/firmware/efi/esrt/entries/entry3/capsule_flags

bibz, 2022-03-12:
Same problem than OP here.

@mrhpearson A trial firmware for the T480 or for the X1C6?

My own ESRT details for the Intel ME entry:
```
% sudo find /sys/firmware/efi/esrt/entries/entry1/ -type f -exec cat {} \; -print
9c9d9769-32fa-4841-b550-ea998e754e99
/sys/firmware/efi/esrt/entries/entry1/fw_class
3092451106
/sys/firmware/efi/esrt/entries/entry1/lowest_supported_fw_version
3092451106
/sys/firmware/efi/esrt/entries/entry1/last_attempt_version
0
/sys/firmware/efi/esrt/entries/entry1/last_attempt_status
2
/sys/firmware/efi/esrt/entries/entry1/fw_type
3092451106
/sys/firmware/efi/esrt/entries/entry1/fw_version
0x8010
/sys/firmware/efi/esrt/entries/entry1/capsule_flags
```

mrhpearson, 2022-03-14:
@bibz - Trial BIOS for the T480.

Thanks to all the above for the esrt table dumps. I think I have enough for now :)

Mark

ChiWei-Chen, 2022-03-15:
I work with Mark and have a trail BIOS for X1C6.  If someone is willing to test that it would be appreciated - send me or @mrhpearson an email. Thanks!
(cchen50 at lenovo dot com)(markpearson at lenovo dot com)

Installation will be by fwupdmgr and it will involve updating the BIOS and then

trying a ME FW update
If that doesn't work, downgrading ME and then upgrading again
Thanks

nszceta, 2022-05-17:
Is there any update to this issue?

linrunner, 2022-06-21:
I can confirm that the newest update [Consumer ME Version 184.92.4222](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN23ETXXP.firmware) - together with the prerequisite [BIOS 0.1.56](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN23ETXXP.firmware) (_"Fixed an issue where Intel Management Engine firmware update by LVFS is failed"_) - can be installed by `fwupdmr` without problems.

ChiWei-Chen, 2022-06-22:
@linrunner , thanks for sharing the information, I shared it with FW team. I'll keep you posted if there is any progress.

imnotjames, 2022-06-23:
@ChiWei-Chen Hi there!  I have the same issue as linrunner

I installed [BIOS Update 0.1.56](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN23ETXXP.firmware)

Then I installed [Consumer ME Update 184.92.4222](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN23RN.firmware)

The BIOS says it's correct but the ESRT is incorrect.

I'll send you an email in case a dump of my BIOS could also be helpful

```
$ fwupdmgr get-updates

20KHCTO1WW
│
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Summary:            UEFI ESRT device
  │   Current version:    184.65.3590
  │   Minimum Version:    184.65.3590
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUID:               9c9d9769-32fa-4841-b550-ea998e754e99
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
[...]
```

```
$ fwupdmgr get-history
20KHCTO1WW
│
[...]
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Previous version:   184.65.3590
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2022-06-23 02:28
  │   GUID:               9c9d9769-32fa-4841-b550-ea998e754e99
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 6th:
        New version:      184.92.4222
        Remote ID:        lvfs
        Release ID:       12651
        Summary:          Lenovo ThinkPad X1 Carbon 6th Consumer ME Firmware
        License:          Proprietary
        Size:             2.2 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N23RN20W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        • 8 Intel Platform Update 2022.1
        
        Version 11.8.92.4222 (LVFS: 184.92.4222)
        
        Problem Fixes
        
        • Mitigated the following security vulnerabilities:
      
```

omriasta, 2022-06-27:
Also seeing the same behavior, Bios screen also confirms new version is installed:

```
20KHCTO1WW
│
├─Embedded Controller:
│ │   Device ID:          9698faabddf0d7b18925cfbbda95f8b0d0dacc53
│ │   Previous version:   0.1.21
│ │   Update State:       Success
│ │   Last modified:      2020-11-04 15:36
│ │   GUID:               3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─  New version:      0.1.22
│       Remote ID:        lvfs
│       License:          Unknown
│       Description:      
│       The vendor did not supply any release notes.
│     
├─System Firmware:
│ │   Device ID:          1c53551e7da69d896138fac1ae131c83ad46d923
│ │   Previous version:   0.1.54
│ │   Update State:       Success
│ │   Last modified:      2022-02-17 17:36
│ │   GUID:               a4b51dca-8f97-4310-8821-3330f83c9135
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─  New version:      0.1.55
│       Remote ID:        lvfs
│       License:          Unknown
│       Description:      
│       The vendor did not supply any release notes.
│     
├─Intel Management Engine:
│ │   Device ID:          29749712e93b58730ecd1b8537a5503ff5b7fb27
│ │   Previous version:   184.86.3909
│ │   Update State:       Success
│ │   Last modified:      2022-02-17 17:40
│ │   GUID:               9c9d9769-32fa-4841-b550-ea998e754e99
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─  New version:      184.90.3987
│       Remote ID:        lvfs
│       License:          Unknown
│       Description:      
│       The vendor did not supply any release notes.
│     
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Previous version:   0.1.55
│ │   Update State:       Success
│ │   Last modified:      2022-06-20 16:27
│ │   GUID:               a4b51dca-8f97-4310-8821-3330f83c9135
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad X1 Carbon 6th:
│       New version:      0.1.56
│       Remote ID:        lvfs
│       Release ID:       13903
│       Summary:          Lenovo ThinkPad X1 Carbon 6th System Firmware
│       License:          Proprietary
│       Size:             9.5 MB
│       Created:          2016-07-08
│       Urgency:          High
│       Vendor:           Lenovo
│       Description:      
│       Lenovo ThinkPad X1 Carbon 6th System Firmware
│       
│       The computer will be restarted automatically after updating BIOS completely. Do NOT turn off your computer or remove the AC adaptor while update is in progress.
│       
│       • Updated the Diagnostics module to version 04.24.000.
│       • Updated the CPU microcode.
│       • Fixed an issue where Intel Management Engine firmware update by LVFS is failed.
│       
│       (Note) Drive encryption by OPAL management software needs to be disabled before
│       
│       start firmware update.
│     
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Previous version:   184.90.3987
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2022-06-27 17:43
  │   GUID:               9c9d9769-32fa-4841-b550-ea998e754e99
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Reported to remote server
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 6th:
        New version:      184.92.4222
        Remote ID:        lvfs
        Release ID:       12651
        Summary:          Lenovo ThinkPad X1 Carbon 6th Consumer ME Firmware
        License:          Proprietary
        Size:             2.2 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N23RN20W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        • 8 Intel Platform Update 2022.1
        
        Version 11.8.92.4222 (LVFS: 184.92.4222)
        
        Problem Fixes
        
        • Mitigated the following security vulnerabilities:

```

yochaigal, 2022-07-11:
I'm having the same issue, but on an X1 Carbon 4th gen. I see there are many such issues open at the moment. I will put my diagnostic info here, in case it might help.

`efivar -l | grep fw        

0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-95f9a8f4-9531-483c-afee-c3f12774ff4b-0

tree /boot
/boot
├── config-5.11.0-40-generic
├── config-5.15.0-30-generic
├── config-5.15.0-40-generic
├── efi
│   ├── EFI
│   │   ├── BOOT
│   │   │   ├── BOOTX64.EFI
│   │   │   ├── fbx64.efi
│   │   │   └── mmx64.efi
│   │   └── ubuntu
│   │       ├── BOOTX64.CSV
│   │       ├── fw
│   │       │   └── fwupd-95f9a8f4-9531-483c-afee-c3f12774ff4b.cap
│   │       ├── fwupdx64.efi
│   │       ├── grub.cfg
│   │       ├── grubx64.efi
│   │       ├── mmx64.efi
│   │       └── shimx64.efi
│   └── initrd.img-5.3.0-19-generic

apt-cache show fwupd
Package: fwupd
Architecture: amd64
Version: 1.7.5-3
Multi-Arch: foreign
Priority: optional
Section: admin
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian EFI <debian-efi@lists.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 6648
Provides: fwupdate

efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0017,0018,0019,001A,001B,001C,0001
Boot0000* ubuntu	HD(1,GPT,d735f80f-9dc7-422b-a322-676ceeb18d7a,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* Linux-Firmware-Updater	HD(1,GPT,d735f80f-9dc7-422b-a322-676ceeb18d7a,0x800,0x100000)/File(\EFI\ubuntu\fwupdx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,1,0)
Boot001E* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,0,0)
Boot001F* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0020* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)

fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.5
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
[... 195 lines cut ...]
  │                       • Reported to remote server
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Yoga 1st:
        New version:      184.92.4222
        Remote ID:        lvfs
        Release ID:       13541
        Summary:          Lenovo ThinkPad X1 Yoga 1st Consumer ME Firmware
        License:          Proprietary
        Size:             2.2 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N1CRG70W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Intel Platform Update 2022.1 Product Version Maintenance Release
        
        Version 11.8.92.4222 (LVFS: 184.92.4222)
        
        Problem Fixes
        
        • Mitigated the following security vulnerabilities under issues. Please see fixed issues for details.

`

ChiWei-Chen, 2022-07-12:
@yochaigal , thanks for sharing the information, I've shared with FW team, we'll give your back if any progress.

kmauleon, 2022-07-12:
hi @yochaigal can you please check ME version in BIOS Setup Menu... thank you very much...

yochaigal, 2022-07-12:
> hi @yochaigal can you please check ME version in BIOS Setup Menu... thank you very much...

11.8.92.4222

kmauleon, 2022-07-13:
hi @yochaigal this means your ME is already the latest but you have the BIOS ESRT issue wherein incorrect version is referenced by fwupd, thus asking you to update it. this is still under investigation by BIOS dev. for now please disregard the update notification for ME.  thank you very much...

linrunner, 2022-08-26:
@ChiWei-Chen I consider this as finally solved for my X1 Carbon 6th. Just installed the [BIOS 1.57 cab](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN23ETXXP.firmware) via `fwupdmgr install`.

```
fwupdmgr get-devices

[...]
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    184.92.4222
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUIDs:              9c9d9769-32fa-4841-b550-ea998e754e99
│                         1a4eea87-292a-5d99-82bd-fd244d4e0194
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
[...]  
├─System Firmware:
│     Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│     Summary:            UEFI ESRT device
│     Current version:    0.1.57
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     Update Message:     The computer will be restarted automatically after updating BIOS completely . Do NOT turn  off your computer or remove the AC adaptor while update is in progress.  
│     GUIDs:              a4b51dca-8f97-4310-8821-3330f83c9135
│                         230c8b18-8d9b-53ec-838b-6cfc0383493a
│                         6b3879ff-921f-5bc5-a0c2-5cec28f31d2f
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
[...]

fwupdmgr get-updates

Devices with no available firmware updates: 
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • MZVLB512HAJQ-000L7
 • System Firmware
No updates available for remaining devices

fwupdmgr get-history

[...]
├─Intel Management Engine:
│ │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│ │   Previous version:   184.86.3909
│ │   Update State:       Failed
│ │   Update Error:       failed to run update on reboot
│ │   Last modified:      2022-06-21 09:31
│ │   GUID:               9c9d9769-32fa-4841-b550-ea998e754e99
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad X1 Carbon 6th:
│       New version:      184.92.4222
│       Remote ID:        lvfs
│       Release ID:       12651
│       Summary:          Lenovo ThinkPad X1 Carbon 6th Consumer ME Firmware
│       License:          Proprietary
│       Size:             2,2 MB
│       Created:          2016-07-08
│       Urgency:          High
│       Details:          https://pcsupport.lenovo.com/de/en/search?query=N23RN20W
│       Vendor:           Lenovo
│       Description:      
│       • 8 Intel Platform Update 2022.1
│       
│       Version 11.8.92.4222 (LVFS: 184.92.4222)
│       
│       Problem Fixes
│       
│       • Mitigated the following security vulnerabilities:
│     
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Previous version:   0.1.56
  │   Update State:       Success
  │   Last modified:      2022-08-26 07:03
  │   GUID:               a4b51dca-8f97-4310-8821-3330f83c9135
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─  New version:      0.1.57
        License:          Unknown
        Description:      
        The vendor did not supply any release notes.
```
Thanks for your support.

ChiWei-Chen, 2022-08-26:
@linrunner thanks, it's good to know this.

Just a note here for those who want to try BIOS for X1C6.

X1C6
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN23ETXXW.firmware
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN23ETXXP.firmware

bendik, 2022-08-26:
My X1C6 also took the latest update and without any issues 👍

hughsie, 2022-08-26:
@ChiWei-Chen I really appreciate all these fixes, and how you've been communicating on these tickets. It's been noticed, and passed back to people like @mrhpearson -- many thanks indeed.

mrhpearson, 2022-08-29:
Agreed - very awesome. Really happy to be able to close these ME related issues.
Thanks @ChiWei-Chen (as always)

closed 2022-08-29 by mrhpearson

LinuxOnTheDesktop, 2022-12-18:
Just a note here for those who want to try BIOS for X1C6.

> X1C6
> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN23ETXXW.firmware
> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN23ETXXP.firmware

Which of those links is for what version of X1C6, please? (I think I may have asked that question already, but by private email. Even if I have, it would be useful to have the question answered here.)

bibz, 2022-12-18:
> Which of those links is for what version of X1C6, please?

Not sure how you'd want the models to be distinguished, but I could at least see my model with `fwupdmgr`:
- Display all devices: `fwupdmgr get-devices`
- Look for the `Device ID` of the `System Firmware`
- Check out the details of this device to see which model you have with `fwupdmgr get-releases -v <Device ID> | grep -m1 AppstreamId` . I get `  AppstreamId:          com.lenovo.ThinkPadN23ETXXW.firmware`.

There might be an easier way but that one was easy to figure out.

bibz, 2022-12-18:
(Also: note that the two firmwares linked above are live and will get picked up by `fwupdmgr`.)

LinuxOnTheDesktop, 2022-12-18:
Thanks, bibz.

Before you wrote your comment, I found that the second of the listed pieces of firmware - that is, the one the name of which ends in 'P' - would and did install on my machine (whereas trying to install the other one produced an incompatibility error message). I note also, contrary to your most recent comment, the BIOS update in question was _not_ picked up by fwupdmgr. That is: I got a chance to install it only by dint of the .cab file.

bibz, 2022-12-18:
> I note also, contrary to your most recent comment, the BIOS update in question was not picked up by fwupdmgr. That is: I got a chance to install it only by dint of the .cab file.

I am not entirely sure how the update channels are set for fwupdmgr but my machine picked up the update and I did not change any settings related to firmware myself, so it either comes from fwupdmgr or somehow from Debian.

The two detail pages linked for the respective firmware versions show they were `released` as `stable`. _I assumed it meant it would be live for all_, but that could be wrong. (But what does it mean then?)

LinuxOnTheDesktop, 2022-12-19:
@bibz 

Thank you.

In further response to your earlier comment - the one that contained the commands to generate the 'AppstreamId': I get the same output as you do. That output indicates that the .cab for me is the one whose name does _not_ end with a 'P'. Yet, it was, only, the one that ended with a 'P' that actually installed on my machine. Goodness.