fwupd/firmware-lenovo #203: [ThinkPad X1 Carbon Gen 9] failed to verify System Firmware: For System Firmware 0.1.51 expected ...

arno01, 2022-02-28:


lauaviin, 2022-05-18:
Any updates on the issue?

arno01, 2023-08-07:
@mrhpearson any updates please?

## updated logs

```
root@x1:~# fwupdmgr verify
0.	Cancel
1.	5792b48846ce271fab11c4a545f7a3df0d36e00a (Alder Lake-P Integrated Graphics Controller)
2.	65a54fb6ce182f0e75edf0e43047d547a0d61f0e (Prometheus)
3.	a083ebc5138e5e071ef7270cc9a8280722cc7adf (System Firmware)
Choose device [0-3]: 1
failed to verify Alder Lake-P Integrated Graphics Controller: failed to read firmware: Error reading from file: Input/output error
root@x1:~# fwupdmgr verify
0.	Cancel
1.	5792b48846ce271fab11c4a545f7a3df0d36e00a (Alder Lake-P Integrated Graphics Controller)
2.	65a54fb6ce182f0e75edf0e43047d547a0d61f0e (Prometheus)
3.	a083ebc5138e5e071ef7270cc9a8280722cc7adf (System Firmware)
Choose device [0-3]: 2
failed to verify Prometheus: No device checksums for 10.01.3478575
root@x1:~# fwupdmgr verify
0.	Cancel
1.	5792b48846ce271fab11c4a545f7a3df0d36e00a (Alder Lake-P Integrated Graphics Controller)
2.	65a54fb6ce182f0e75edf0e43047d547a0d61f0e (Prometheus)
3.	a083ebc5138e5e071ef7270cc9a8280722cc7adf (System Firmware)
Choose device [0-3]: 3
failed to verify System Firmware: No stored checksums for 0.1.37
root@x1:~# fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.8.12
runtime   org.kernel                    6.2.0-26-generic

root@x1:~# dpkg -l fwupd
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  fwupd          1.8.12-2     amd64        Firmware update daemon
root@x1:~# fwupdmgr get-devices
LENOVO 21CB007WCK
│
├─12th Gen Intel Core™ i7-1260P:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000429
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         ab855c04-4ff6-54af-8a8a-d8193daa0cd8 ← CPUID\PRO_0&FAM_06&MOD_9A
│                         3ebbde86-d03e-549a-a8fd-02ebf9aa537a ← CPUID\PRO_0&FAM_06&MOD_9A&STP_3
│     Device Flags:       • Internal device
│   
├─Alder Lake-P Integrated Graphics Controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    0c
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              eaad9970-8e4d-56da-88ab-41a8c1e2811f ← PCI\VEN_8086&DEV_46A6
│                         ed0b9458-c2f1-54c5-9063-dea8f75b4039 ← PCI\VEN_8086&DEV_46A6&REV_0C
│                         15bf9dad-22cf-57a8-9ca1-eb3b08e0070e ← PCI\VEN_8086&DEV_46A6&SUBSYS_17AA22E7
│                         f9c988f0-e65e-5c87-9045-aa22a751c22f ← PCI\VEN_8086&DEV_46A6&SUBSYS_17AA22E7&REV_0C
│                         c4625510-a985-517c-8800-0ecfc6f68c8f ← PCI\VEN_8086&DEV_46A6&REV_00
│                         d7a0b6c6-3253-598e-9195-49093094c89a ← PCI\VEN_8086&DEV_46A6&SUBSYS_17AA22E7&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
[... 270 lines cut ...]
│     Current version:    3552440
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               6fbaaaff-982b-4b54-aae6-b82c5d89db22
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
      Summary:            UEFI ESRT device
      Current version:    18417548
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               62036a80-3968-4bf1-ab13-175eabbc4901
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
```

mrhpearson, 2023-08-17:
The original issue of the PCR changing got closed and reviewing the above it seems there is something different going on so I have a separate ticket to track that (LO-2605). I can reproduce (and see the same on other platforms)

@hughsie - I haven't dug into what fwupdmgr is actually doing under the hood but given the different failure conditions highlighted above (File IO, no device checksums, no stored checksums) it looks like there might be multiple things going on and I'm guessing the FW team might need some guidelines into what they are missing from the process. Maybe we can take offline?

mrhpearson, 2023-11-10:
The conclusion is that using PCR0 is a problem - it includes measurements of FW and the FW can change. For example the ME FW may be updated, and that is measured.
Not sure exactly who/why/when the decision of removing PCR0 was made - but I suspect it was related to getting too many tickets where the measurement was incorrect.

Not sure what the answer is - but for now I think this issue has to be closed as effectively the FW team have chosen to not support this feature.

@hughsie - if we should have an offline conversation on how to address this let me know. Especially if I'm missing something critical :)

closed 2023-11-20 by mrhpearson