fwupd/firmware-lenovo #204: Yoga Slim 7 Pro-14ACH5 Laptop (ideapad) - Type 82MS - provide payloads

colemickens, 2022-02-28:
Lenovo offers these updates for this laptop:
* https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/yoga-series/yoga-slim-7-pro-14ach5/82ms/82ms0000cd/pf2qfgdc/downloads/driver-list
* System Firmware

It would be nice if Lenovo could at least provide that via `fwupdmgr`.

Not sure why the NVME doesn't show up, maybe it's unsupported or doesn't support fw upgrades?

```
cole@raisin ~
❯ sudo fwupdmgr update
Devices with no available firmware updates:
 • USB2.0 Hub
 • USB3.0 Hub
 • Integrated Camera
 • SSD 970 EVO Plus 2TB
 • System Firmware
 • UEFI Device Firmware
 • UEFI dbx
No updatable devices
```

colemickens, 2022-02-28:
full product string from `lshw`: `product: 82MS (LENOVO_MT_82MS_BU_idea_FM_Yoga 14sACH 2021)`

mrhpearson, 2022-03-01:
Hi,

Unfortunately that platform isn't in the Linux certification program so the FW team aren't going to be posted updates to LVFS as it's not in their support scope.

Let the product team knows you want Linux support so we can add these platforms in the future. Recommendation is through the customer survey if you still have that available - or contact support.

Mark

closed 2022-03-01 by mrhpearson

colemickens, 2022-03-01:
Okay, thanks for the extra info @mrhpearson! Appreciate the details and tip on providing feedback. 💯