fwupd/firmware-lenovo #206: Thinkpad Universal Thunderbolt 4 Smart Dock Problems

LU15W1R7H, 2022-03-10:
I'm experiencing a lot of problems with my new "Thinkpad Universal Thunderbolt 4 Smart Dock", which I have connected to my "ThinkPad X1 Yoga Gen 6" with product name "20XYCT01WW" running Arch Linux (`Linux lwirth-tarch 5.16.13-arch1-1 #1 SMP PREEMPT Tue, 08 Mar 2022 20:07:36 +0000 x86_64 GNU/Linux`).

Let's talk about the problems.
The dock has two Displayport connectors. For some reason only one works at a time. Each of them work, when only one is used and if both are connected one port takes over and it's the only one working. This is the reason why I currently have only one monitor pluged in.
Using only one displayport monitor I get a lot of graphical issues. Sometimes the screen gets very distorted and keeps turning on and off. At other times it's relatively stable I get only minor graphical glitches. Sometimes graphics work fine.

Having these issues I tried to update the firmware using fwupd. Since this was my first time using fwupd there were some updates I successfully installed.
```
> fwupdmgr get-history
20XYCTO1WW
│
└─UEFI Device Firmware:
  │   Device ID:          ae1abd099407b1d95698d69b7273f7fed5c6f35c
  │   Previous version:   0.0.0.1
  │   Update State:       Success
  │   Last modified:      2022-03-09 18:55
  │   GUID:               aa096a98-94e6-479b-92f7-5771f6f2d96f
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Reported to remote server
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad X1 Carbon Gen 9 / X1 Yoga Gen 6:
        New version:      1.2.24.0
        Remote ID:        lvfs
        Release ID:       11817
        Summary:          Lenovo ThinkPad X1 Carbon Gen 9 / X1 Yoga Gen 6 Retimer Firmware
        License:          Proprietary
        Size:             285.4 kB
        Created:          2022-02-02
        Vendor:           Lenovo
        Description:
        Lenovo Intel Thunderbolt Retimer Firmware 1.2.24.0

        The computer will be restarted automatically after updating device completely . Do NOT turn off your computer or remove the AC adaptor while update is in progress.

        Fixed some bugs regarding connection issues.
```
However something isn't included in `fwupdmgr get-history` (don't know why). I tried reconstructing the text from a screenshot.
```
  └─ThinkPad X1 Carbon Gen 9 / X1 Yoga Gen 6:
        New version:      1.2.22.0
        Remote ID:        lvfs
        Release ID:       10998
        Summary:          Lenovo ThinkPad X1 Carbon Gen 9 / X1 Yoga Gen 6 Retimer Firmware
        License:          Proprietary
        Size:             285.4 kB
        Created:          2021-12-06
        Vendor:           Lenovo
        Description:
        Lenovo Intel Thunderbolt Retimer Firmware 1.2.22.0

       The computer will be restarted automatically after updating device completely. Do NOT turn off your computer or remove the AC adapter while update is in progress.

       Fixed an issue where 2nd display in the daisy chain failed to light on.

       Fixed an issue where USB devices will be lost when re-attach the USB-C Advanced Dock.
```
I'm not sure why this isn't included in the history. Maybe because it failed?
Also the `Fixed an issue where 2nd display in the daisy chain failed to light on.` sounds very similar to my problem.

The problem persisted.
So I tried to look for a firmware update again.
But this time when running `fwupdmgr refresh --force`, it caused a kernel panic (at least I think this is a kernel panic).

journalctl says this
```
[... 141 lines cut ...]
│ └─Dock Management Controller Information:
│       Device ID:        024e42ddfc4717530c7d01f01a826f93fd2a81c0
│       Current version:  10.2
│       Vendor:           Lenovo (USB:0x17EF)
│       Update Error:     Use the MCU to update the DMC device
│       GUIDs:            affa2dc2-f4be-50e9-8444-9cc4431954c8
│                         697a6a95-464e-5b90-a46e-73beab89316e
│
├─ThinkPad Universal Thunderbolt 4 Smart Dock:
│     Device ID:          bc13992accfc5ab30f068aad000d6b54e1c32411
│     Current version:    35.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0108, TBT:0x0108)
│     GUIDs:              91b4a269-c086-5312-a855-1c31dc88bb7f
│                         b86e59cf-e2de-5900-8177-82e4b6fdb7e4
│                         46e3796e-b3b2-53d7-8f42-d017452613c1
│                         d3330631-77d9-50f2-a2c3-3d98e1d86310
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Device stages updates
│                         • Signed Payload
```

Thanks for any help regarding this issue!

This issue might be related to #191

hughsie, 2022-03-11:
> Mar 10 20:07:59 lwirth-tarch kernel:  igc_get_hw_semaphore_i225+0x2f/0x170 [igc a898859c8886ad4b638108260e5b0beebc8d298b]

I think that's talking about the GigE chip in the dock. If you update the kernel, does that go away? I suspect the weirdness is due to the `BUG: unable to handle page fault for address: 0000000000005b50`.

LU15W1R7H, 2022-03-13:
Updating to the latest stable kernel 5.16.14 didn't help.

I have some further information.
When booting my laptop with the dock plugged in, I get
```
usci_acpi UDBC000:00: ucsi_handle_connector_change: GET_CONNECTOR_STATUS failed (-110)
ucsi_acpi USBC000:00 PPM init failed (-110)
```

While trying to configure the displays with xrandr with their highest resolution
```

```

I always get the error `xrandr: Configure crtc 0 failed` (it also happens for crtc 1, 2 and three). And only one monitor turns on. 

When running X11, then trying to configure the monitors and then exiting X the my screen is blank (no tty) until I remove the dock from my laptop. After that  another kernel panic is triggered:
```
```

LU15W1R7H, 2022-03-27:
@phord 
This seems to be related.
Because I was able to "reproduce" the exact same error when plugging in my dock.

phord, 2022-03-27:
@LU15W1R7H  My bug and stacktrace went away when I replaced my kernel with version 5.15.16. I read somewhere that the fix is in 5.15.13, so I'm surprised you still saw it on 5.16.14.

The bug is in the Intel Ethernet driver (igb/igc). I was able to avoid this bug on another kernel version by blacklisting this (igc) driver. This is a suitable solution for me because I'm not using the wired Ethernet connector on the dock.

LU15W1R7H, 2022-03-28:
@phord Thanks, blacklisting fixed the kernel panics!
Did you manage to get two displayport monitors working at the same time? This still doesn't work for me.

phord, 2022-03-28:
@LU15W1R7H I didn't notice that my TB4 dock has two DP connections. Why didn't they put them together at all?  I don't have DP connectors at work where the dock is, though.  I will try to bring it home this week to experiment.

I do have two 4K@60Hz monitors connected at home using HDMI connections; one is the internal one, and one is external via USB-C -> HDMI.  This is working fine.

LU15W1R7H, 2022-04-03:
@phord 
Could you try to connect two 4K monitors via displayport to your TB4 dock? (If you have them to your disposal.)
I'm having problems with this exact configuration.

taisph, 2022-04-04:
> @phord Could you try to connect two 4K monitors via displayport to your TB4 dock? (If you have them to your disposal.) I'm having problems with this exact configuration.

FWIW, I am running with two 4k monitors connected to the dock via DP. I get an occasional lost signal (audio also cuts out) for a few seconds.

LU15W1R7H, 2022-04-04:
@taisph Okay, interesting. Can you elaborate on your setup?
Are you using Xorg?
Do you have intel graphics? If yes do you use `xf86-video-intel`?
Etc...

I can run each external monitor separately at 4K. But setting both up to 4K results in xrandr telling me `xrandr configure crtc failed`. Turning both down to Full HD works.

taisph, 2022-04-04:
> @taisph Okay, interesting. Can you elaborate on your setup? Are you using Xorg? Do you have intel graphics? If yes do you use `xf86-video-intel`? Etc...

I have a ThinkPad P15 Gen 2, NVIDIA with 510.60.02 proprietary driver on Ubuntu 20.04.4 LTS, using Xorg. Both monitors are connected directly to the dock and are not daisy chained. Laptop is powered directly and not through the dock. I get an occasional (1-3 times a day) lost signal on one or both monitors for a few seconds. I have headphones connected to one monitor and the audio signal cuts out as well. Every now and then (maybe once every two weeks) I have to cold start the dock to get the monitors working with the laptop. I also had to permanently disable power saving (DPMS) for the monitors using xset -dpms as power save did a number on audio (Ubuntu misconfigures the audio output) and sometimes the monitors don't wake up again. And then there's #191 which forced me to stop using the ethernet port on the dock.

taisph, 2022-05-05:
More dock woes: Got annoyed with the lost signal today because my headphones cut out, and instead connected them directly into the dock. Every ~8 seconds it now has a kind of 1 second low-res flange effect in the audio. That was definitely not there when I got the dock.

lauaviin, 2022-05-18:
@LU15W1R7H Same issue here with connecting 2DP monitors. On P15v 2gen + Thunderbolt 4 workstation dock. Connecting second monitor via the USB-C cable will cause the screen to flicker at random intervals. 2xDP doesn't flicker but on occasion get lost signal on random intervals AND/OR the screen doesn't take in account resolution changes beyond 1080p.

tylermontneyacc, 2022-08-17:
I've been noticing similar issues on Windows across multiple laptops (typically T15) on...

- USB-C Dock Gen 2
- Thunderbolt 3 Workstation Dock Gen 2
- Universal Thunderbolt 4 Dock
- Ultra Docking Station

Been bugging me for a couple months now across various firmware versions.

fhuberts, 2022-10-24:
I have a Lenovo TB4 dock (non-smart) with 2xDP and 1x HDMI.
On Fedora 36 and 37, both my Z16 AMD and P14s Gen 2 AMD laptops can't light up either of the displays, no matter which ones I plug and which ones I do not plug.
The dock was updated to the latest firmware (1.10 I think). 
I've read some comments that from kernel 6.0 it should start working but with the rawhide kernel (6.1-rc1) I see no difference with 5.19.16-200: not functional.

vrumjantsev, 2022-11-26:
Have someone found a workaround?

tylermontneyacc, 2022-11-28:
> Have someone found a workaround?

Ever since using straight HDMI instead of DP to HDMI adapters, I've gone from multiple incidents a day to one a week (if that). Lenovo support once told me they didn't work well with those adapters, and recommended I didn't use them.

vrumjantsev, 2022-12-02:
> Ever since using straight HDMI instead of DP to HDMI adapters, I've gone from multiple incidents a day to one a week (if that). Lenovo support once told me they didn't work well with those adapters, and recommended I didn't use them.

I'm not using adapters DP to DP from both monitors to dock

silivanmunguar, 2022-12-24:
I just got the TBD workstation doc and it is a nightmare on windows 11. The graphics keep glitching, and signals cutting out especially when I open applications that have like virtual DJ or Premier pro. Is there a solution or should I just return it?

ryantm, 2022-12-24:
I gave up after spending two full days trying to get it to work. I'm having much more luck with the Plugable 4K USB C Docking Station Triple Monitor with my 12th gen Framework laptop. DisplaylinkManager uses a lot of CPU and has some barely noticeable snowlike artifacts but at least it works.

silivanmunguar, 2022-12-25:
> I gave up after spending two full days trying to get it to work. I'm having much more luck with the Plugable 4K USB C Docking Station Triple Monitor with my 12th gen Framework laptop. DisplaylinkManager uses a lot of CPU and has some barely noticeable snowlike artifacts but at least it works.

Its a shame this doesn't work. I really wanted the 230W power delivery. Can you share the dock you are using?

ryantm, 2022-12-25:
> Can you share the dock you are using?

The model number is ‎UD-ULTC4K. I said the name in the previous post.

fhuberts, 2023-01-21:
There seem to be some AMD MST fixes in the works.

CommanderRedYT, 2024-01-02:
Hi, I just noticed something weird. I am also having the two 4k monitor issue (Although with the Thunderbolt 4 Workstation dock). I found something weird tho: It looks like plugging in both displays at once with their resolution set to 4K does not work at all (they show up, but no output / they are entering standby). When I put them to 1440p, it works. When I put them to 4K again tho, it seems that the scaling increases (4K rendered, but only 1440p displayed).

mo3r31337, 2024-02-14:
I found that it is possible to run two monitors at 4k resolution on this dock if one monitor is connected via displayport and the other monitor is connected using the downstream thunderbolt port.
I managed to run two DELL U2720Q in 4k resolution
Lenovo ThinkPad Thunderbolt 4 Dock
Lenovo ThinkPad X1 nano gen 1
Arch Linux with 6.7.4 kernel
Xorg version 21.1.99.1.r669.g9c7c470b1-1
Xorg video driver: modesetting
Works in wayland too

LU15W1R7H, 2024-05-09:
Coming back to this issue after a long time...
Now I'm running NixOS with kernel 6.8.9 on my Lenovo Thinkpad Z16 Gen 1 with both an integrated GPU (AMD Radeon 680M) and a dedicated GPU (AMD Radeon RX 6500M).
I'm still using my Thinkpad Universal Thunderbolt 4 Smart Dock.
I'm using Hyprland as my wayland compositor.

When using two 4K monitors I still get weird behaviour.
When connecting to my dock while already running Hyprland, both monitors  turn on, but only one of monitors (always the same one) actually displays anything. The other just turns back off, wihtout ever showing anything. Hyprland recogizes both monitors and whatever I do, I can't get the 2nd monitor to show anything.
However if I connect the dock before starting Hyprland, my TTY is mirrored on both monitors and when I start Hyprland then I will actually be able to use both monitors (without problems).
How can this be?
Also I still get the occational other weird behaviour, when connecting or disconnecting the dock, like Hyprland crashing or even a kernel panic related to the AMDGPU driver...

It would be great to find a solution to this. This dock is giving me a lot of headaches... If I could I would just return it.

silivanmunguar, 2024-05-09:
I returned this dock and got a used Dell WD19TB which is cheaper and works
flawlessly.

LU15W1R7H, 2024-05-09:
@silivanmunguar just bought myself a new dell dock, hopefully this time it will work better!

pdinoto, 2025-01-06:
Hello, been using a 40B0 dock with Linux (Mint 22 Wilma, kernel `6.8.0-51-generic`) on a X1 Carbon (`21CB`, ThinkPad X1 Carbon Gen 10)
It was working fine (all devices on dock worked fine, they unplugged and plugged back nicely)

I updated dock to lastest versions thru Windows 11, Levono System Updater 5 (tried once using `fwupdmgr` and got no error but no progress, continuous black screen after screen for 6 or 7 minutes, then regular boot). After that, now I experience the 

```
Jan 05 20:20:15 x1sa2 kernel: igc 0000:79:00.0: PTM enabled, 4ns granularity
Jan 05 20:20:15 x1sa2 kernel: igc 0000:79:00.0 (unnamed net_device) (uninitialized): PCIe link lost, device now detached
Jan 05 20:20:15 x1sa2 kernel: igc: probe of 0000:79:00.0 failed with error -13
Jan 05 20:20:20 x1sa2 kernel: igc 0000:79:00.0: enabling device (0000 -> 0002)
Jan 05 20:20:20 x1sa2 kernel: igc 0000:79:00.0: PTM enabled, 4ns granularity
Jan 05 20:20:20 x1sa2 kernel: igc 0000:79:00.0 (unnamed net_device) (uninitialized): PHC added
Jan 05 20:20:20 x1sa2 kernel: igc 0000:79:00.0: 4.000 Gb/s available PCIe bandwidth (5.0 GT/s PCIe x1 link)
Jan 05 20:20:20 x1sa2 kernel: igc 0000:79:00.0 eth0: MAC: 08:3a:88:bb:8b:47
Jan 05 20:20:20 x1sa2 kernel: igc 0000:79:00.0 enp121s0: renamed from eth0
Jan 05 20:20:26 x1sa2 kernel: igc 0000:79:00.0 enp121s0: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Jan 05 20:56:08 x1sa2 kernel: igc 0000:79:00.0 enp121s0: PCIe link lost, device now detached
Jan 05 20:56:08 x1sa2 kernel: igc: Failed to read reg 0xc030!
Jan 05 20:56:08 x1sa2 kernel: WARNING: CPU: 2 PID: 2693 at drivers/net/ethernet/intel/igc/igc_main.c:6648 igc_rd32+0xa4/0xc0 [igc]
Jan 05 20:56:08 x1sa2 kernel:  int340x_thermal_zone int3400_thermal input_leds pmt_class sparse_keymap acpi_pad acpi_thermal_rel a>
Jan 05 20:56:08 x1sa2 kernel: Workqueue: events igc_watchdog_task [igc]
Jan 05 20:56:08 x1sa2 kernel: RIP: 0010:igc_rd32+0xa4/0xc0 [igc]
Jan 05 20:56:08 x1sa2 kernel:  ? igc_rd32+0xa4/0xc0 [igc]
Jan 05 20:56:08 x1sa2 kernel:  ? igc_rd32+0xa4/0xc0 [igc]
Jan 05 20:56:08 x1sa2 kernel:  ? igc_rd32+0xa4/0xc0 [igc]
Jan 05 20:56:08 x1sa2 kernel:  igc_update_stats+0xab/0x770 [igc]
Jan 05 20:56:08 x1sa2 kernel:  igc_watchdog_task+0xa4/0x540 [igc]
```

issue. It works on fresh boot, crashes at some point. Unplugging and plugging does not help. Doing the same while power cycling the dock is same thing. Have to reboot linux.

What kind of tests, and troubleshooing info can I get to help with the issue?

roland-bless, 2025-02-12:
I have the same issues with the Lenovo 40B0 Thunderbolt 4 Dock and a Lenovo P14sGen5: 
the dock internal ethernet NIC detaches from the PCIe and the igc crashes:

`
[18538.929552] pci 0000:49:00.0: [8086:5502] type 00 class 0x020000 PCIe Endpoint
[18538.929600] pci 0000:49:00.0: BAR 0 [mem 0x00000000-0x000fffff]
[18538.929673] pci 0000:49:00.0: BAR 3 [mem 0x00000000-0x00003fff]
[18538.930016] pci 0000:49:00.0: PME# supported from D0 D3hot D3cold
[18538.930456] pci 0000:49:00.0: Adding to iommu group 23
[18538.933392] pci 0000:49:00.0: BAR 0 [mem 0xa9e00000-0xa9efffff]: assigned
[18538.933399] pci 0000:49:00.0: BAR 3 [mem 0xa9f00000-0xa9f03fff]: assigned
[18538.942991] igc 0000:49:00.0: enabling device (0000 -> 0002)
[18538.943023] igc 0000:49:00.0: PCIe PTM not supported by PCIe bus/controller
[18539.007760] igc 0000:49:00.0 (unnamed net_device) (uninitialized): PHC added
[18539.035451] igc 0000:49:00.0: 4.000 Gb/s available PCIe bandwidth (5.0 GT/s PCIe x1 link)
[18539.035453] igc 0000:49:00.0 eth0: MAC: xx:xx:...
[18539.041980] igc 0000:49:00.0 enp73s0: renamed from eth0
....
[18541.058275] igc 0000:49:00.0 enp73s0: PCIe link lost, device now detached
[18541.058291] ------------[ cut here ]------------
[18541.058293] igc: Failed to read reg 0x5b50!
`

Unloading and reloading the igc module does not work in the sense that it crashes immediately again:
`[25435.228072] igc 0000:49:00.0 enp73s0: PHC removed
[25443.149930] Intel(R) 2.5G Ethernet Linux Driver
[25443.149933] Copyright(c) 2018 Intel Corporation.
[25443.149968] igc 0000:49:00.0: enabling device (0000 -> 0002)
[25443.150008] igc 0000:49:00.0: PCIe PTM not supported by PCIe bus/controller
[25443.150164] igc 0000:49:00.0 (unnamed net_device) (uninitialized): PCIe link lost, device now detached
`
I'm using currently Linux Kernel 6.13.2 (NixOS 24.11 with latest kernel).
I also tried the `pcie_port_pm=off pcie_aspm.policy=performance` options, but this did not change anything obvious.
I tested several docks, also upgraded to their latest firmware (the laptop has also the latest FW).
Probably it has to do with the PCIe tunneling feature of the dock, no clue.