fwupd/firmware-lenovo #208: firmware update 184.90.3987 will not install on ThinkPad T460S

arderyp, 2022-03-19:
Running Ubuntu 20.04 latest on UEFI with full disk encryption and SecureBoot.

```bash
sudo fwupdmgr refresh
sudo fwupdmgr update

Firmware metadata last refresh: 2 hours ago. Use --force to refresh again.
Devices with no available firmware updates: 
 • THNSF5256GPU7 TOSHIBA
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 184.86.3909 to 184.90.3987?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ • 8 Intel Platform Update 2021.2 Product Version Maintenance Release         ║
║                                                                              ║
║ Version 11.8.90.3987 (LVFS: 184.90.3987)                                     ║
║                                                                              ║
║ Problem Fixes                                                                ║
║                                                                              ║
║ • Mitigated the following security vulnerabilities under issues. Please      ║
║ see fixed issues for details.                                                ║
║                                                                              ║
║ Note: Please suspend or disable any disk encryption tool prior to update.    ║
║                                                                              ║
║ 20F9CTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
```

- I answer "y"
- it prompts me to reboot
- I answer "Y"
- machine reboots
- reboot interrupt occurs
- spinning line saying "please wait while we install firmware update"
- it spins for a while then loads my encryption password screen
- I unencrypt, then log in
- re-run the above commands and it tells me the same update needs to be installed

I've installed firmware on this machine multiple times through this method.  Normally the spinning line goes away and then a real progress bar shows with a green advancing fill to indicate the new firmware is being installed.  In fact, I just installed this same firmware update on another machine just fine this morning (Dell OptiPlex running ubuntu 20.04 latest).

I tried disabling secure boot and trying again, no luck.  The instructions do say I need to disable encryption before rebooting, but this is not easily done, and I've never had problems flashing firmware updates on this exact machine through this exact method for the past 6 years. Any help would be greatly appreciated.  I couldn't find any relevant logs, but I don't know what I'm looking for.  Happy to provide log/command output as necessary.

kmauleon, 2022-03-21:
hi @arderyp Please check BIOS Setup Menu and see if MEFW version is updated.  If the version is already 11.8.90.3987, then the update was successful and you are seeing a known issue with BIOS ESRT table similar to this one >> #89 
If MEFW was not indeed updated, you have to disable your encryption as this is a known limitation thus the note: "Please suspend or disable any disk encryption tool prior to update."
thank you very much...

arderyp, 2022-03-21:
Thanks for the comment.  I am AFK for a week, but will check after that and get back to you.

arderyp, 2022-03-28:
I haven't managed to check BIOS yet, but I am now no longer getting the same prompt to re-install the same update... I am now getting:

```
Firmware metadata last refresh: 5 minutes ago. Use --force to refresh again.
Devices with no available firmware updates: 
 • UEFI dbx
No updatable devices
```

Which seems wrong, as again I've been updating the firmware on this machine for years.

arderyp, 2022-03-28:
```
$ fwupdmgr get-devices --show-all-devices
20F9CTO1WW
│
├─Core™ i7-6600U CPU @ 2.60GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000ec
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         28d7f31c-1663-5b15-a900-94a71cbcd53f
│                         e4bc7423-9fec-55a6-8f6c-827c40992491
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.14
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     Update Error:       Cannot install update when not on AC power
│     GUIDs:              ffaf3e12-a6ea-4352-851b-76c6455913c6
│                         1445f630-3409-5ac3-aca4-8e59c36f7846
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    184.86.3909
│     Minimum Version:    184.86.3909
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     Update Error:       Cannot install update when not on AC power
│     GUIDs:              63451986-0c42-42db-9a18-342a8455ac68
│                         0e559cf4-f432-5cd4-9611-bd774afb417a
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─Skylake GT2 [HD Graphics 520]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    07
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              3ed2bd10-18ff-52c2-b2f9-a0676606c8b5
│                         1ba26452-aa80-5c89-872e-245836b1bfa2
│                         2f7ef0d3-15f0-571c-b852-de72f1c7c730
│                         bdcb90a0-e7cd-5d77-995d-d79b7183aa29
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.54
│ │   Minimum Version:    0.1.27
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   Update Error:       Cannot install update when not on AC power
│ │   GUIDs:              42f29620-3d63-4f09-950b-9b7055570f28
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       68b8d4d4-88dc-5c69-8013-67b942a77dd3
[... 48 lines cut ...]
│                         • Updatable
│   
└─TPM:
      Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
      Summary:            TPM 2.0 Device
      Current version:    303.12.0.0
      Vendor:             Intel (TPM:INTC)
      GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6
                          34801700-3a50-5b05-820c-fe14580e4c2d
                          8e1cbc5d-5a11-5149-bfea-b6065d5296ba
                          03f304f4-223e-54f4-b2c1-c3cf3b5817c6
                          52d7b679-db28-5bf7-bd87-41d77aeec600
      Device Flags:       • Internal device
    
________________________________________________

Devices that were not updated correctly:

 • Intel Management Engine (184.86.3909 → 184.90.3987)

Devices that have been updated successfully:

 • Intel Management Engine (184.83.3874 → 184.86.3909)
 • System Firmware (0.1.53 → 0.1.54)
```

arderyp, 2022-03-29:
now it says:

```
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Downloading…             [***************************************]
Successfully downloaded new metadata: 3 local devices supported
Devices with no available firmware updates: 
 • UEFI dbx
No updatable devices
```

So, maybe this has somehow resolved itself?

mrhpearson, 2022-03-29:
So I think you may have hit the ESRT table update issue, but interestingly yours has resolved. I know the BIOS team are rolling out a potential fix for this (fingers crossed).
Are we OK to close this now?

arderyp, 2022-03-30:
sure, lets close for now and I will report back if it resurfaces

arderyp, 2022-03-30:
sigh.  Of course, right after closing this, I ran updated, the same firmware update resurfaced:

```
Firmware metadata last refresh: 11 hours ago. Use --force to refresh again.
Devices with no available firmware updates: 
 • THNSF5256GPU7 TOSHIBA
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 184.86.3909 to 184.90.3987?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ • 8 Intel Platform Update 2021.2 Product Version Maintenance Release         ║
║                                                                              ║
║ Version 11.8.90.3987 (LVFS: 184.90.3987)                                     ║
║                                                                              ║
║ Problem Fixes                                                                ║
║                                                                              ║
║ • Mitigated the following security vulnerabilities under issues. Please      ║
║ see fixed issues for details.                                                ║
║                                                                              ║
║ Note: Please suspend or disable any disk encryption tool prior to update.    ║
║                                                                              ║
║ 20F9CTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
```

I applied it, rebooted, and the same firmware update is still pending.

0xOMEGA, 2022-03-30:
I have the same issue, running fedora on t480.

apawelec, 2022-04-02:
Same thing on T480s / Mint 20.3

arderyp, 2022-04-02:
@mrhpearson looks like there are some other confirmations

arderyp, 2022-04-03:
now I am back to getting:

```
Firmware metadata last refresh: 11 hours ago. Use --force to refresh again.
Devices with no available firmware updates: 
 • UEFI dbx
No updatable devices
```

mrhpearson, 2022-04-04:
I think this is the same as the some other ME update issues raised on the forum - the ESRT table (which fwupd uses to determine what the version of ME FW running is) is advertising the incorrect ME FW version.

You can confirm by going into the BIOS setup screen (F1 during early boot) and confirming the ME FW version there. If it's 184.90.3987 then your FW is already updated and it's the ESRT table that is wrong. If it's not then we have a different issue and let me know.

For the ESRT issue - the FW team have a fix in the works that they have told me they'll release before the next ME update. For now I'm afraid the answer is just to ignore those prompted ME FW updates. Unfortunately there is, apparently, no way to downgrade the ME to force it to update correctly.

Mark

arderyp, 2022-04-05:
I've just confirmed in my BIOS that the ME Firmware Version is 11.8.90.3987.

The version numbers don't match up exactly (reported firmware version is 184.90.3987).  But, since they both end in 90.3987, I'm guessing they are the same and thus that @mrhpearson is correct.

So I guess the only thing to do here is wait for the ESRT table fix to roll out.  Do you have an issue number/link for that @mrhpearson?

arderyp, 2022-04-11:
@mrhpearson do you have another issue number we can track for this?

mrhpearson, 2022-04-12:
I've made sure the T460s is specifically added to the LO-1247 case for tracking. For updates I'm afraid you'll just have to nag me - we don't have a public facing bug system I'm afraid

ssilenzi, 2022-04-13:
Hello,

I think I am in a similar situation, the same firmware update is still pending.

Computer:
Lenovo Thinkpad P15v gen 2

Current version of Intel ME Firmware:
15.0.35.1951 Corporate
installed from Windows 11 by the Lenovo Utility

OS:
Ubuntu 20.04 (in dual boot with Windows 11)
Secure boot enabled
Disk encrypted by LUKS

Commands executed:
sudo fwupdmgr refresh
sudo fwupdmgr update

Output of the command:
═════════════════════════════════════════════════════════════════════════════
 Upgrade Intel Management Engine **from 0 to 240.35.1951**?                                                                                                
═════════════════════════════════════════════════════════════════════════════
 Lenovo ThinkPad P15 Gen 2i/ P17 Gen 2i/ P15g Gen 2i Corporate ME Firmware                                                                   
 Version 15.0.35.1951                                                                                                                                                                 
                                                                                                                                                                                                    
 The computer will be restarted automatically after updating completely . Do                                                                            
 NOT turn off your computer or remove the AC adaptor while update is in                                                                                 
 progress.                                                                                                                                                                                    
                                                                                                                                                                                                    
 21A90006IX must remain plugged into a power source for the duration of the                                                                         
 update to avoid damage.                                                                                                                                                          
═════════════════════════════════════════════════════════════════════════════

Update error from log:
**failed to update to 0: authentication signing error**

Summarizing: it is trying to install the same version that is already installed.
The update is proposed because the version detected is **0**.

mrhpearson, 2022-04-14:
I've added a note to the same LO-1247 bug about this, but this is the first platform of that generation that I've seen this issue reported against and it does seem different with version '0' being mentioned.

Just to avoid confusing the issue on this thread - would you mind raising this one as a separate issue and give the contents of your ESRT table by running:

`grep -r . /sys/firmware/efi/esrt/entries/*`

And also go into the BIOS setup screen (F1 during early boot) and seeing what version of FW is being reported there please?

If it's different I'll need to flag it internally as a separate issue.

arderyp, 2022-04-14:
Agreed, it's sounds like a separate issue to me as well.

Thanks for all the help @mrhpearson!

arderyp, 2022-04-14:
Agreed, @ssilenzi's sounds like a separate issue to me as well.

Thanks for all of the assistance @mrhpearson!

arderyp, 2022-04-27:
hey @mrhpearson, any update on this?

mrhpearson, 2022-04-27:
The updates and fixes are supposed to be rolling out - but I'm not sure what the status is for these two platforms. I'll go and check

ChiWei-Chen, 2022-04-28:
Hi 
I have a trail BIOS and ME for T480 on this issue. If you're willing to test that it would be appreciated - send me or @mrhpearson an email. Thanks!
(cchen50 at lenovo dot com) (mpearson at lenovo dot com)

Installation will be by fwupdmgr and it will involve updating the BIOS and then trying a ME FW update.

arderyp, 2022-04-30:
My machine is T460s, so I don't think I can test @ChiWei-Chen 

For what it is worth, `fwupdmgr` is still saying I need to update from `184.86.3909` to `184.90.3987`, despite my boot menu saying I am running `ME Firmware Version: 11.8.90.3987`

palonsoro, 2022-05-08:
Hi. I am hitting this also on a Lenovo ThinkPad P50. I already have the indicated `11.8.90.3987` version but it still says that I need to upgrade from `184.86.3909`.

ChiWei-Chen, 2022-05-09:
@arderyp , @palonsoro , Thanks for reporting this, I'll keep you update once any progress from FW team.

arderyp, 2022-05-09:
Great, thanks @ChiWei-Chen.  The issue is still occurring on my machine sporadically (sometimes I'm told I need to update to the newest version, which I'm already on, other times I'm told there is no update, other times I'm told there are no updatable devices).

mmerklinger, 2022-05-16:
Hey, I have the same issue on a ThinkPad X1 Carbon 6th Generation. The UEFI setup shows ME Firmware version `11.8.90.3987`, but fwupd still offers me to upgrade from `184.86.3909`.

arderyp, 2022-05-17:
@ChiWei-Chen and @mrhpearson, any update?

nschloe, 2022-05-20:
Same on my Thinkpad X280. BIOS reports 11.8.90.3987, still fwupdmgr asks me to
```
Upgrade Intel Management Engine from 184.86.3909 to 184.90.3987?
```

tomelliot, 2022-05-30:
I'm seeing this issue also
Lenovo Carbon X1 7th Gen.
Pop OS
fwupdmgr says ME ver 192.81.1753
BIOS says ME ver 192.85.1869

I stumbled on this GitHub issue because I'm having another problem and I'm not sure if it's related: My power management is completely wrong. It's reporting that the system is on AC power and charging, without any AC plugged in. Is the ME responsible for power management or is that a different driver?

mrhpearson, 2022-05-30:
I believe that's usually handled by the EC. Best to raise a different ticket - or probably better to post this on the Lenovo forums (forums.lenovo.com/linux) as it's not really lvfs related

tomelliot, 2022-05-30:
Thanks for the quick response @mrhpearson .

ChiWei-Chen, 2022-06-08:
@arderyp I can see the BIOS is testing in the embargo, I'll keep you update once it's stable.

arderyp, 2022-06-10:
great, thanks!

arderyp, 2022-06-16:
@beaukinstler is the problem still solved for you?  I haven't done anything to mittigate it, and sometimes I'm told there is a pending update when I query for updates, and sometimes I am not.  And other times I'm told there is no updatable device :stuck_out_tongue: 

Are you confident that whatever you did permanently resolved the problem?  If so, can you please provide a little more detail on what you did in BIOS?

@ChiWei-Chen , any update by chance?

ChiWei-Chen, 2022-06-16:
@arderyp , the BIOS just released the fix of T480s to sable, they're working on applying the fix to T460s. I'll give you an update once it's available.

arderyp, 2022-06-17:
awesome, glad to hear there's movement!

beaukinstler, 2022-06-17:
> is the problem still solved for you? I haven't done anything to mittigate it, and sometimes I'm told there is a pending update when I query for updates, and sometimes I am not. And other times I'm [...]

@ChiWei-Chen , I'm very sorry. I was incorrect. On first reboot, I was no longer getting the GUI pop up message that I had been getting each time. However, I didn't get the upgrade after all. 
I ran `sudo fwupdmgr update --show-all` and still see these.

```
Devices with no available firmware updates: 
 • INTEL SSDPEKKF256G8L
 • UEFI Device Firmware
 • UEFI dbx
 • Unifying Receiver
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 184.90.3987 to 184.92.4222?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Mitigated the following security vulnerabilities:                          ║
║                                                                              ║
║ 20JMS0Q300 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
```

ChiWei-Chen, 2022-06-21:
@beaukinstler , thanks for the update, I'm checking the status with FW team, I'll keep you posted if there is any progress.

jpseara, 2022-06-22:
@ChiWei-Chen 

> I'm seeing this issue also Lenovo Carbon X1 7th Gen. Pop OS fwupdmgr says ME ver 192.81.1753 BIOS says ME ver 192.85.1869
> 
> I stumbled on this GitHub issue because I'm having another problem and I'm not sure if it's related: My power management is completely wrong. It's reporting that the system is on AC power and charging, without any AC plugged in. Is the ME responsible for power management or is that a different firmware?

Reset your laptop using the small reset button, after shutting it down. That should fix it.

jpseara, 2022-06-22:
@mrhpearson @ChiWei-Chen

Reporting the same issue others have.

My T480s (Ubuntu 22.04, fully updated both packages and firmware), keeps saying I need to upgrade Intel ME from 184.90.3987 to 184.92.4222, no matter how many times I upgrade it and let it reboot and do its thing (it says it's upgrading, etc etc and apparently it finishes successfully).

From the BIOS, I can see that ME is already upgraded to 184.92.4222, so clearly the problem is in fwupgmdr thinking it's in a previous version.

Other details from my laptop:

Model: 20L7001LPG
BIOS: N22ET73W (1.50)

Also I've seen you asking this output before, so here it is:

```
# grep -r . /sys/firmware/efi/esrt/entries/*

/sys/firmware/efi/esrt/entries/entry0/fw_class:ebfe8df8-dee7-4692-a721-cbcf5095c5cf
/sys/firmware/efi/esrt/entries/entry0/lowest_supported_fw_version:65568
/sys/firmware/efi/esrt/entries/entry0/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry0/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry0/fw_type:1
/sys/firmware/efi/esrt/entries/entry0/fw_version:65586
/sys/firmware/efi/esrt/entries/entry0/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry1/fw_class:676af093-2a5c-4238-9c29-db8063a33532
/sys/firmware/efi/esrt/entries/entry1/lowest_supported_fw_version:3092909971
/sys/firmware/efi/esrt/entries/entry1/last_attempt_version:3092909971
/sys/firmware/efi/esrt/entries/entry1/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry1/fw_type:2
/sys/firmware/efi/esrt/entries/entry1/fw_version:3092909971
/sys/firmware/efi/esrt/entries/entry1/capsule_flags:0x8010
/sys/firmware/efi/esrt/entries/entry2/fw_class:4d254d6e-cd67-477b-97d5-bc3048af45c4
/sys/firmware/efi/esrt/entries/entry2/lowest_supported_fw_version:65558
/sys/firmware/efi/esrt/entries/entry2/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry2/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry2/fw_type:2
/sys/firmware/efi/esrt/entries/entry2/fw_version:65558
/sys/firmware/efi/esrt/entries/entry2/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry3/fw_class:b7913cf4-ba8b-6972-af81-eff6ba94f5c6
/sys/firmware/efi/esrt/entries/entry3/lowest_supported_fw_version:121572633
/sys/firmware/efi/esrt/entries/entry3/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry3/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry3/fw_type:2
/sys/firmware/efi/esrt/entries/entry3/fw_version:121572633
/sys/firmware/efi/esrt/entries/entry3/capsule_flags:0x0
```

ChiWei-Chen, 2022-06-22:
@jpseara , thanks. I was told by FW team the previous issue should be fixed by BIOS 1.50. Could you let us know the fwupdmgr version on your system? 
If it still happened with the latest version, the FW team need to get your BIOS dump for investigation. If you're willing to help on this, could you drop me a mail to: ( cchen50 [at] lenovo dot com ). I'll send you the instruction and the efi tool.
Thanks!

jpseara, 2022-06-22:
@ChiWei-Chen 

I will be more than happy to help.

Here are the fwupd versions:

```
$ sudo dpkg -l | grep fwup
ii  fwupd                                      1.7.5-3                                    amd64        Firmware update daemon
ii  fwupd-signed                               1.44+1.2-3                                 amd64        Linux Firmware Updater EFI signed binary
ii  fwupdate                                   12-7                                       amd64        Transitional package for fwupd
ii  libfwupd2:amd64                            1.7.5-3                                    amd64        Firmware update daemon library
ii  libfwupdplugin5:amd64                      1.7.5-3                                    amd64        Firmware update daemon plugin library
```

I will drop you an email.

mrhpearson, 2022-06-22:
Hi - that shouldn't matter. The update is done the same way (UEFI capsule) and the FW should update the esrt table with the correct version and isn't doing that here.

If you're interested - from the above esrt table entry 0 you collected - it has FW version 3092909971. First convert this to hexadecimal = 0xB85A0F93
0xB8 = 184
0x5A = 90
0x3987 = 3987
So it's reporting the previous version.

This has been a common problem on a number of platforms and the FW team are issuing new BIOS updates that fix it. There's a bit of a chicken and egg situation as you need to update your BIOS, and then do an ME FW update to correct everything...and it takes a while for everything to roll out, especially as multiple platforms are involved

Chi-Wei has a better idea on the details for the T460s specifically, but in summary it's a known issue and there is a fix in the works.

jpseara, 2022-06-22:
> Hi - that shouldn't matter. The update is done the same way (UEFI capsule) and the FW should update the esrt table with the correct version and isn't doing that here.
>
> If you're interested - from [...]

Thanks. I should reinforce that my system is a T480s and not a T460s. @ChiWei-Chen told me it should already be fixed in my BIOS revision, but it clearly isn't.

I've already sent Chi-Wei a dump of my BIOS, so let's see what happens. Thanks!

arderyp, 2022-06-24:
sigh... I just retried answering Y to the fwupdmgr prompt, reboot, watching spinning icon,then auto-reboot.  Same problem.  I'm running the latest everything under Ubuntu 20.04.

ChiWei-Chen, 2022-06-24:
@arderyp , the instruction is expected like:
1. update BIOS [Lenovo Thinkpad T460s System Firmware N1CET87W, version 1.55](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1CET.firmware)  first
2. Reboot automatically
3. Update [T460s Consumer ME Firmware N1CRG70W, 11.8.92.4222](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1CRN.firmware)
4. Reboot, press F1 in preboot to check BIOS(System Firmware)  and ME Firmware version.
5. Boot into OS the check BIOS and ME Firmware with "fwupdmgr get-devices"

Please let us know how it works. Thanks!

arderyp, 2022-06-28:
@ChiWei-Chen I don't know how to do step 1 on a ThinkPad running ubuntu 20.04.  I am looking into it, but the site you linked doesn't (as far as I can see) provide a link to the BIOS update package. If you know explicitly what to do, please let me know.

I am currently behind:

```
$ sudo dmidecode -s bios-version
[sudo] password for arderyp: 
N1CET86W (1.54 )
```

arderyp, 2022-06-28:
I found the BIOS update here: https://support.lenovo.com/us/en/downloads/ds112117-bios-update-utility-bootable-cd-for-windows-10-81-7-64-bit-7-32-bit-linux-thinkpad-t460s

I unzipped it and tried installing, and failed:

```
$ fwupdmgr install N1CET87W.cab 
Decompressing…           [***************************************]
firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer
```

Given the stern warning, I am reluctant to change that conf setting.

arderyp, 2022-06-28:
I see this related comment on a similar issue: https://github.com/fwupd/firmware-lenovo/issues/155#issuecomment-1048528780

and yet:

```
$ apt info libjcat
N: Unable to locate package libjcat
N: Unable to locate package libjcat
E: No packages found
```

arderyp, 2022-06-28:
```
$ sudo apt install libjcat1
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libjcat1 is already the newest version (0.1.4-0ubuntu0.20.04.1).
libjcat1 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

arderyp, 2022-06-28:
seems ubuntu's libjcat1 is woefully out of date, which isn't an uncommon story.  So perhaps my only option is to temporarily update the conf to allow an untrusted install now, then set it back. I've checked the checksums, and they are good, but I want to get your input @ChiWei-Chen and @mrhpearson before proceeding.

Also, I am running Ubuntu 20.04, and the [lenovo site](https://support.lenovo.com/us/en/downloads/ds112117-bios-update-utility-bootable-cd-for-windows-10-81-7-64-bit-7-32-bit-linux-thinkpad-t460s) that provides the 1.55 BIOS update file only explicitly mentions Ubuntu 16.04... should I be concerned?

ChiWei-Chen, 2022-06-28:
@arderyp , it is suggested to install the BIOS from LVFS. Because the BIOS from support.lenovo doesn't contain this Linux issue fix yet.
You can do it with following instructions:
1. Download [Lenovo-ThinkPad-T460s-SystemFirmware-1.55.cab](https://fwupd.org/downloads/9192e972b3f60fe70e2ad499fbfdc955078bcb55b39f28e38448e62570f1f232-Lenovo-ThinkPad-T460s-SystemFirmware-1.55.cab)
2. fwupdmgr install 9192e972b3f60fe70e2ad499fbfdc955078bcb55b39f28e38448e62570f1f232-Lenovo-ThinkPad-T460s-SystemFirmware-1.55.cab
3. Download [Lenovo-ThinkPad-T460s-ConsumerMEFirmware-11.8.92.4222.cab](https://fwupd.org/downloads/311a1ab421ffc0ea95f0e4e43f2626876e78a2cfdc27382f0df9162fd8123e04-Lenovo-ThinkPad-T460s-ConsumerMEFirmware-11.8.92.4222.cab)
4. fwupdmgr install 311a1ab421ffc0ea95f0e4e43f2626876e78a2cfdc27382f0df9162fd8123e04-Lenovo-ThinkPad-T460s-ConsumerMEFirmware-11.8.92.4222.cab

Please let me know if this doesn't work for you.

ChiWei-Chen, 2022-06-28:
@arderyp , sorry for the confusion, I just confirmed with FW team, it seems the T460s doesn't fix the issue yet, please ignore my instruction.

arderyp, 2022-07-10:
any update @ChiWei-Chen ?

ChiWei-Chen, 2022-07-12:
@arderyp I see the BIOS team are rolling out a potential fix for this on other platforms, I'm checking if they can implement the fix in T460s. I'll give you back if any progress.

marex, 2022-07-12:
P51 same problem.

ChiWei-Chen, 2022-07-13:
@marex , thanks for reporting the issue, could you provide more detailed information? Thanks!
- fwupdmgr update --show-all
- system firmware version and Intel Management Engine firmware version in BIOS main page
- OS and fwupdmgr version

arderyp, 2022-07-14:
Maybe on a separate issue?  This one is getting pretty long and still no end in sight

[119 more replies]

closed 2022-11-10 by arderyp