fwupd/firmware-lenovo #209: fwupd sometimes removes GRUB from the UEFI boot order

mgedmin, 2022-03-21:


mrhpearson, 2022-03-21:
I'm not sure we have any X390's in our team...they're from before when I joined.
How easy is this to reproduce?
I've raised an internal ticket for tracking: LO-1636

dkwo, 2022-03-22:
@mrhpearson This happens on many other thinkpads as well, e.g. on my t460s

mrhpearson, 2022-03-30:
FW team are unable to reproduce the problem. They've tried on the X390 as well as some other platforms (X13/T14s Gen1/2 and T490/T590)

I'm personally a bit confused about the steps to reproduce above. I'm probably being thick but they didn't add up to me. The Ubuntu bug linked to is more useful - and that looks like the boot variables just got lost in a regular update with nothing to do with USB sticks and grub involved - but we unfortunately have limited information there.

I think the "not realised ME updated" issue above noted is very different and related to the ESRT table not getting update which they have a (hopefully) fix in the pipeline for. I don't want to confuse that issue with the efiboot variables disappearing as it's not the same.

@hughsie - any recommendations on what else we can do here or can I mark this unreproducible?

Mark

hughsie, 2022-03-30:
@mrhpearson I think it's highly a requirement of having the correct *old* firmware version, i.e. the one you're upgrading **from**.

dkwo, 2022-03-30:
@mrhpearson Please look into this more carefully: this issue has to do with grub entry being deleted by a lenovo fw update, as title says.
It happens for sure on my t460s, for which I provided relevant logs above.

It seems to be related to some sort of reset that the firmware update performs on the boot order, which erases the grub from boot entries,  see also this comment https://github.com/fwupd/firmware-lenovo/issues/209#issuecomment-1074111005

dkwo, 2022-03-30:
@mrhpearson I appreciate the effort. In my case, there's no Windows to boot into, so the simplest thing is to chroot from a usb stick and reinstall grub, which adds back the appropriate efi boot variables I guess. The efibootmgr output is from *after* doing that.

dkwo, 2022-04-04:
@mrhpearson I think this is the same as https://github.com/fwupd/firmware-lenovo/issues/187
The firmware update removes the boot efi variable pointing to grub

ChiWei-Chen, 2022-07-01:
Hi, 
According to the FW team, this seems to be a known grub issue in old version, seems already fixed in later version. (Sorry, we don't have specific version of the fix)
Please help to provide the grub and fwupd version if you still see the problem.
Thanks.

dkwo, 2022-07-01:
@ChiWei-Chen I'm using grub 2.06, the current released version, and I had this issue a few days ago, with fwupd 1.8.1. Is the fw team referring to some specific commit in grub master?

hughsie, 2022-07-01:
I think it could be referring to the *old* firmware version, i.e. the thing you're updating _from_.

dkwo, 2022-07-01:
I see, then last update (Released 2022-06-15) Version 1.55 still had the issue for my T460s.

ChiWei-Chen, 2022-07-05:
Hi @dkwo, the FW team need to know following detail of the machine, could you comment on it? Thanks

> What is their partition scheme of their SSD?
> Also, did they set secure boot to DISABLED before the update?

ChiWei-Chen, 2022-07-05:
@dkwo , thanks! This is very appreciated. 
I'll give you back if any progress from FW team.

dkwo, 2022-08-08:
This is not specific to grub, as the same happens with just an efistub created using efibootmgr:
when updating firmware, it resets/removes the corresponding boot efi variables, and the only option left in the list is nvme... (the hard drive).

ChiWei-Chen, 2022-08-10:
@dkwo , I shared your feedback to the FW, since they're in the holiday season these 2 weeks, we may not get an update soon. 
I'll keep you posted if there is any progress. Thanks.

liskin, 2023-02-12:
@ChiWei-Chen Assuming holidays are over; any updates? Just had this happen to me on a T25 (T470-ish motherboard, afaik). This is the second time I tried fwupd on this machine. First time was sometime in 2018, the same thing happened. Resorted to using [downloaded isos](https://pcsupport.lenovo.com/cz/cs/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t25-type-20k7/20k7/20k70000ge/downloads/driver-list/component?name=BIOS%2FUEFI) and/or Windows to perform firmware updates, which worked flawlessly for years. Today decided to give fwupd another chance, and lost my boot entries again. Seems a bit weird that it's only happening with fwupd, doesn't it?

ChiWei-Chen, 2023-02-14:
> Today decided to give fwupd another chance, and lost my boot entries again. Seems a bit weird that it's only happening with fwupd, doesn't it?
@liskin , I'm afraid I haven't get feedback from FW team yet, do you remember what fwupd version was used on this T25 machine?

ChiWei-Chen, 2023-02-14:
@liskin the FW team has no clues on this issue. They did not add BIOS modification for GRUB missing version issue.
As @hughsie commented previously, this could be related to the old firmware version, could you let us know the old and new firmware versions?
> I think it could be referring to the _old_ firmware version, i.e. the thing you're updating _from_.

liskin, 2023-02-14:
@ChiWei-Chen This was an update from 0.1.70 to 0.1.73. Secure boot is disabled, because Debian's shim doesn't currently support fwupd at all (https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=968997, https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=995155). My boot entries are therefore just `HD(1,GPT,…,0x800,0x80000)/File(\EFI\debian\grubx64.efi)`, no shim.

dkwo, 2023-02-14:
Once again, this has nothing to do with GRUB.
(I see this just using efibootmgr with an efistub.)
The firmware updates from Lenovo clear the EFI boot variables.

mrhpearson, 2023-08-21:
Looks like we missed updating this ticket, but they reproduced the issue on the X390 and believe the issue is related to the BIOS flash driver.
There is a BIOS update released - N2JETA2W (1.80). They can't reproduce the issue with this. If anybody can confirm it's fixed for them with this update it would be appreciated.

dkwo, 2023-08-21:
@mrhpearson I have a T460s. Can you push the update for that model as well?

mrhpearson, 2023-08-21:
ack - I've asked for confirmation

mrhpearson, 2023-09-01:
FW for X390 is released on LVFS to stable. Closing ticket.

closed 2023-09-01 by mrhpearson