fwupd/firmware-lenovo #211: Lenovo X1 carbon 0.1.44 EC / 0.1.41 System Firmware update fail

pgc, 2022-03-28:


hughsie, 2022-03-28:
@mrhpearson if i understand correctly the capsule updater needs to be updated to allow flashing the firmware either with a low battery (or no battery) -- I don't think there is anything we can do in userspace to affect this check, right?

mrhpearson, 2022-03-29:
Hmmm - yeah...that's awkward. Let me check with the FW team for their recommendations - internal ticket is  LO-1656
Mark

pgc, 2022-04-06:
Now 18%. Slightly concerned this 1% seems to have wicked away a bit quicker than the last. Anyway it would seems we have ~36 days until my laptop implodes in on itself forming a mini black hole.

mrhpearson, 2022-04-06:
I've escalated - but please call customer support and get a ticket logged there too. Did you buy your system with the Linux preload?

We don't have battery firmware updates on Linux (first time I've ever seen a battery update - it's on the list of devices we want to add, but not done yet). I'm pretty sure you're going to get a "install windows and update the battery FW" response, and I'm afraid that's probably the best way to get your system back up and running at this point - much as it sucks. 

If you brought the Linux preload you probably have a good case for getting a replacement unit - please reference this ticket and give support my name as having looked into this.

Mark

pgc, 2022-04-06:
Thanks Mark - I did not buy this system preloaded with Linux (this appeared to be a step too far for corporate purchasing and our reseller). I will figure out my support channel to log a ticket. I figured that reinstalling Windows and updating was my "pull in case of emergency" option - I have other machines I can use while I do that. Apologies, I didn't intend to and will not continue with the countdown.

mrhpearson, 2022-04-06:
Sorry - hit the button by mistake when reading the update... :/

ChiWei-Chen, 2022-04-07:
@pgc , I work with @mrhpearson and go following question from FW team, could you help to clarify it? Thanks!

Which product is what customer is facing the problem? 
from previous discussion, I’m not sure the machine type of customer.
I can see “Lenovo X1 carbon” as a page title, but on the other hand, 
I can also see 20UQS0TD00 that first 4digit “20UQ” stands for X1 nano.
Firstly can you clarify it? 

Thanks

pgc, 2022-04-07:
@ChiWei-Chen my apologies, I had forgotten the nano part. To confirm it is an X1 Nano Carbon.

ChiWei-Chen, 2022-04-08:
@pgc , your feedback is apprecited.
Our battery FW team need more detail of the battery, could you help to check this? Thanks.
- output of "upower -d"
- capture of usage history of battery (power statics)
FYI
https://help.gnome.org/users/gnome-power-manager/stable/statistics.html.en

pgc, 2022-04-08:
Device: /org/freedesktop/UPower/devices/line_power_AC
  native-path:          AC
  power supply:         yes
  updated:              Mon 28 Mar 2022 15:59:35 BST (956879 seconds ago)
  has history:          no
  has statistics:       no
  line-power
    warning-level:       none
    online:              yes
    icon-name:          'ac-adapter-symbolic'

Device: /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               Celxpert
  model:                5B10W13964
  serial:               5068
  power supply:         yes
  updated:              Fri 08 Apr 2022 17:46:44 BST (50 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               pending-charge
    warning-level:       none
    energy:              8.42 Wh
    energy-empty:        0 Wh
    energy-full:         45.11 Wh
    energy-full-design:  48.2 Wh
    energy-rate:         0.000480766 W
    voltage:             11.271 V
    percentage:          18%
    capacity:            93.5892%
    technology:          lithium-polymer
    icon-name:          'battery-caution-charging-symbolic'

Device: /org/freedesktop/UPower/devices/keyboard_hid_04o69of8od3oe3o89_battery
  native-path:          hid-04:69:f8:d3:e3:89-battery
  model:                pgcooper magickeyboard
  power supply:         no
  updated:              Fri 08 Apr 2022 17:46:11 BST (83 seconds ago)
  has history:          yes
  has statistics:       yes
  keyboard
    present:             yes
    rechargeable:        yes
    state:               discharging
    warning-level:       none
    percentage:          91%
    icon-name:          'battery-full-symbolic'

Device: /org/freedesktop/UPower/devices/touchpad_hid_64oa5oc3oefob9o34_battery
  native-path:          hid-64:a5:c3:ef:b9:34-battery
  model:                pgcooper magictrackpad
  power supply:         no
  updated:              Fri 08 Apr 2022 17:47:20 BST (14 seconds ago)
  has history:          yes
  has statistics:       yes
  touchpad
    present:             yes
    rechargeable:        yes
    state:               discharging
    warning-level:       none
    percentage:          62%
    icon-name:          'battery-full-symbolic'

Device: /org/freedesktop/UPower/devices/DisplayDevice
  power supply:         yes
  updated:              Fri 08 Apr 2022 17:36:44 BST (650 seconds ago)
  has history:          no
  has statistics:       no
  battery
    present:             yes
    state:               pending-charge
    warning-level:       none
    energy:              8.42 Wh
    energy-full:         45.11 Wh
    energy-rate:         0.000480766 W
    percentage:          18%
    icon-name:          'battery-caution-charging-symbolic'

Daemon:
  daemon-version:  0.99.13
  on-battery:      no
  lid-is-closed:   yes
  lid-is-present:  yes
  critical-action: PowerOff

pgc, 2022-04-08:
I've installed gnome-power-manager but it will take some time to collect data AFAICT. Will post later.

pgc, 2022-04-11:
gnome-power-manager doesn't seem to be able to capture any data.

ChiWei-Chen, 2022-04-11:
@pgc thank you for the feedback. Our battery team wants to know the battery firmware version, could you provide the output of "sudo fwupdtool get-devices –verbose"? Thanks!

ChiWei-Chen, 2022-04-12:
Thanks, @pgc , I'll give you reply if any feedback from battery team.

pgc, 2022-04-13:
@ChiWei-Chen many thanks for your help, this is great news.

ChiWei-Chen, 2022-05-13:
@pgc , please wait for few more days.
The battery firmware is under internal review process. I'll keep you updated once the FW is available on LVFS.
Thank you for your understanding.

pgc, 2022-05-13:
@ChiWei-Chen ok great, thanks for the update.

ChiWei-Chen, 2022-05-20:
@pgc , thanks for your patience.
Here is the battery firmware on LVFS. Could you give it a try and let us know how it works?
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2TBU.firmware
Any comment would be very appreciated.

pgc, 2022-05-20:
Yay, it worked! Many thanks to everyone on the thread and in the background that made this happen. 

Just as an FYI in case people arrive at this bug in future, I had to install the battery update by specifying the device-id i.e.

sudo fwupdmgr update <device-id>

because there were other (older) firmware updates waiting to be installed that required battery >25% so when those got queued up the process failed. Now my battery is charged up I can do those other updates as normal.

pgc, 2022-05-20:
Ugh, that got mangled

sudo fwupdmgr update device-id

ChiWei-Chen, 2022-05-20:
@pgc thanks for the feedback. Can we close this is no further concerns? 😉

pgc, 2022-05-20:
Yes, I have close the issue.

closed 2022-05-20 by pgc

hughsie, 2022-05-20:
Thanks @pgc and @ChiWei-Chen -- team work!

mmohaveri, 2022-05-20:
@ChiWei-Chen Is there a solution for X1 Yoga as well?
I'm having the same issue on my X1 Yoga 4th gen, my battery has suddenly dropped to 0% and says 'Not Charging' and at the same time I can't update any of my firmwares due to the same error regarding 25% charge threshold.