fwupd/firmware-lenovo #214: ThinkPad X1 Carbon 6th System Firmware upgrade failure

ausil, 2022-04-14:
my ThinkPad X1 Carbon 6th has an update available Upgrade System Firmware from 0.1.51 to 0.1.55 

with SecureBoot enabled the update is quickly skipped, if I turn SecureBoot off the process seems to hang. Eventually, I get a message on the screen,

```
SCT Flash Utility for Lenovo
 for shell V1.0.4.9
Copyright (c) 2011-2019Phoenix Technologies Ltd.
Copyright (c) 2011-2019 Lenovo Group Limited.

Usage: ShellFlash64 [COMMAND]...

Try "ShellFlash64 -help" for more commands.

ERROR 105 - Command line error!
Reset System
```
followed by the system rebooting

mrhpearson, 2022-04-14:
Hi
I think this is the same as reported in of https://github.com/fwupd/firmware-lenovo/issues/201

I have a ticket open with the FW team for this: LO-1598
Still waiting for updates from them. It seems common to that generation of platforms :(

ChiWei-Chen, 2022-04-21:
@ausil , the FW team haven't reporduced the issue yet, they want know whether you can see this same issue if you update to previous version? for example:  0.1.51-0.1.54 or 0.1.53?
Thanks

ausil, 2022-04-21:
@ChiWei-Chen I can try, I just got a new 9th Gen Carbon X1 and will only have the 6th gen one about a week more. I believe I was not able to update off of 0.1.51 to anything newer which is why I am where I am now

ChiWei-Chen, 2022-04-21:
@ausil , this information was really appreciated. I'll keep you posted if any progress from FW team.

ChiWei-Chen, 2022-04-22:
@ausil , the FW team suspects this issue may be related to the parameter used by fwupd, they want to know if you can still see the same problem if you do "fwupdmgr install <*.cab>" locally.
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN23ETXXW.firmware

BTW, could you let us know the current version of the fwupd?
Thanks.

ausil, 2022-04-22:
manually applying the update worked, I am using fwupd-1.7.7-1.fc36.x86_64 so the problem is not an older version

ausil, 2022-04-22:
after updating manually to 0.1.54 the upgrade to 0.1.55 failed the same. though there is also a second update "Upgrade Intel Management Engine from 184.86.3909 to 184.90.3987?"

ausil, 2022-04-22:
and infact manually running "fwupdmgr install <*.cab>" for the 0.1.55 firmware from 0.1.54 fails. so I think it is the firmware and not fwupd

ChiWei-Chen, 2022-04-26:
@ausil , these findings are appreciated. I'll keep you posted if any feedback from the FW team.

ChiWei-Chen, 2022-04-27:
@ausil I got a trial BIOS for X1 carbon 6. If you're willing to test it, could you send me a mail? ( cchen50 at lenovo dot com )
Thank you very much.

LinuxOnTheDesktop, 2022-06-20:
I too have an X1C6 on which 'system firmware' will not install. I notice that Lenovo seems to have released a revised version of the update. For, I see (my bold):

─ThinkPad X1 Carbon 6th:
│       New version:      0.1.56
│       Remote ID:        lvfs
│       Release ID:       13903
│       Summary:          Lenovo ThinkPad X1 Carbon 6th System Firmware
│       License:          Proprietary
│       Size:             9.5 MB
│       Created:          2016-07-08
│       Urgency:          High
│       Vendor:           Lenovo
│       Release Flags:    • Is upgrade
│       Description:      
│       Lenovo ThinkPad X1 Carbon 6th System Firmware

[. . .]
│       • **Fixed an issue where Intel Management Engine firmware update by LVFS is failed.**

Yet, the update _still_ does not install. I note also the 'high' urgency declared by Lenovo.

ChiWei-Chen, 2022-08-26:
Hi, Here are the BIOS releases for X1C6, please feel free to give it a try and let us know how it works. Thanks!

X1C6
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN23ETXXW.firmware
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN23ETXXP.firmware

I also commented this link to #140

LinuxOnTheDesktop, 2022-08-28:
I managed to install this BIOS on my X1CG6. However, the TPM upgrade that I have long been offered, and that would not install, _still_ does not install.

ChiWei-Chen, 2022-08-29:
@LinuxOnTheDesktop , thanks, this seems a different issue, could you provide the error log and `fwupdmgr get-devices`? We need FW team to check this.

LinuxOnTheDesktop, 2022-08-29:
@ ChiWei-Chen: thank you. Here is the `fwupdmgr get-devices`. As to 'the error log', I am afraid that I do not know what you mean: please explain.

```
20KH001UKP
│
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUIDs:              3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│                         7090c94a-ee40-593e-91bc-8cfd5a8c0a5d
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    184.92.4222
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUIDs:              9c9d9769-32fa-4841-b550-ea998e754e99
│                         1a4eea87-292a-5d99-82bd-fd244d4e0194
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─LENSE20512GMSP34MEAT2TA:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express solid state drive
│     Current version:    2.8.8341
│     Vendor:             Lenovo (NVME:0x17AA)
│     GUIDs:              22eade43-1bc6-5eb2-ae69-b7d59beedeb8
│                         762c02f3-333f-5f89-a0c6-289659ae3cb6
│                         963522d7-bc73-51fa-a70c-cf615f42bcf6
│                         b0cd8903-04e1-51f7-a69a-f08b77e4a6ad
│                         ef93a17b-97f6-550e-a5bc-32c22ff6cdf8
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.57
│ │   Minimum Version:    0.0.1
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   Update Message:     The computer will be restarted automatically after updating BIOS completely . Do NOT turn off your computer or remove the AC adaptor while update is in progress.
│ │   GUIDs:              a4b51dca-8f97-4310-8821-3330f83c9135
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       6b3879ff-921f-5bc5-a0c2-5cec28f31d2f
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  238
│       Minimum Version:  238
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            14503b3d-73ce-5d06-8137-77c68972a341
│                         5971a208-da00-5fce-b5f5-1234342f9cf7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│     
├─UEFI Device Firmware:
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    0.71.12
│     Minimum Version:    0.71.12
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUIDs:              cea87551-1701-43fb-afbc-6e8ce9728345
│                         9ce4f5c8-e787-5f82-a7ef-a22c071c9423
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:          fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR24.10_B0036
      Bootloader Version: BOT03.01_B0008
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                          279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:       • Updatable
                          • Supported on remote server
```

ChiWei-Chen, 2022-08-31:
@LinuxOnTheDesktop sorry for not being clear. 
Since I cannot find the TPM offered in the `fwupdmgr get-devices` above. And BIOS(system firmware) and ME firmware have already been updated successfully.
Could you provide steps or instructios on how the system getting TPM not installed? Any video or screenshot would be very appreciated.
Thanks!

LinuxOnTheDesktop, 2022-08-31:
@ChiWei-Chen 

```
$ fwupdmgr refresh
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Successfully downloaded new metadata: 6 local devices supported
$ fwupdmgr get-updates
Devices with no available firmware updates: 
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • LENSE20512GMSP34MEAT2TA
 • System Firmware
 • Unifying Receiver
20KH001UKP
│
└─UEFI Device Firmware:
  │   Device ID:          f95c9218acd12697af946874bfe4239587209232
  │   Summary:            UEFI ESRT device
  │   Current version:    0.71.12
  │   Minimum Version:    0.71.12
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUIDs:              cea87551-1701-43fb-afbc-6e8ce9728345
  │                       9ce4f5c8-e787-5f82-a7ef-a22c071c9423
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 6th:
        New version:      0.73.20
        Remote ID:        lvfs
        Release ID:       8342
        Summary:          Lenovo ThinkPad X1 Carbon 6th STM TPM Firmware
        License:          Proprietary
        Size:             439.6 kB
        Created:          2020-03-03
        Urgency:          High
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Lenovo ThinkPad X1 Carbon 6th STM TPM Firmware Version 73.20
        
        • Do NOT turn off your computer or remove the AC adaptor while update is in progress
      
$ fwupdmgr update
Devices with no available firmware updates: 
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • LENSE20512GMSP34MEAT2TA
 • System Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI Device Firmware from 0.71.12 to 0.73.20?                        ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad X1 Carbon 6th STM TPM Firmware Version 73.20                 ║
║                                                                              ║
║ • Do NOT turn off your computer or remove the AC adaptor while update is     ║
║ in progress                                                                  ║
║                                                                              ║
║ 20KH001UKP must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
Downloading…             [***************************************]
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Scheduling…              [***************************************]
Decompressing…           [***************************************]
Restarting device…       [***************************************]
Waiting…                 [***************************************]
Successfully installed firmware
 • Unifying Receiver

An update requires a reboot to complete. Restart now? [y|N]: n
```

When I do reboot, a firmware screen flashes up briefly - 'please wait while we install a system update' - and then the PC reboots, and then next time I run fwupdmg, I see the same text as above. When I tried the procedure with a newer version of fwupd (using flatpak), `fwupdmgr update` gives me a message about a missing or invalid capsule header.

LinuxOnTheDesktop, 2022-08-31:
Ah: here is some of what happens on the newly released fwupd 1.8.4:

```
$ sudo -i flatpak run org.freedesktop.fwupd get-updates
Loading…                 [**************************             ]04:38:30:0973 FuBiosSettings       KERNEL BUG: 'type' attribute not exported: (failed to load type: Failed to open file “/sys/class/firmware-attributes/thinklmi/attributes/SecureBoot/type”: No such file or directory)
$ sudo -i flatpak run org.freedesktop.fwupd update
Loading…                 [**************************             ]04:40:09:0953 FuBiosSettings       KERNEL BUG: 'type' attribute not exported: (failed to load type: Failed to open file “/sys/class/firmware-attributes/thinklmi/attributes/SecureBoot/type”: No such file or directory)
Loading…                 [-                                      ]
WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates: 
 • UEFI dbx
 • Unifying Receiver
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • LENSE20512GMSP34MEAT2TA
 • System Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI Device Firmware from 0.71.12 to 0.73.20?                        ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad X1 Carbon 6th STM TPM Firmware Version 73.20                 ║
║                                                                              ║
║ • Do NOT turn off your computer or remove the AC adaptor while update is     ║
║ in progress                                                                  ║
║                                                                              ║
║ LENOVO 20KH001UKP must remain plugged into a power source for the duration   ║
║ of the update to avoid damage.                                               ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
WARNING: This package has not been validated, it may not work properly.
Writing…                 [***************************************]
Downloading…             [***************************************]
Waiting…                 [***************************************]
Writing…                 [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Scheduling…              [***************************************]
Decompressing…           [***************************************]
04:40:18:0814 FuPluginUefiCapsule  missing or invalid embedded capsule header
Restarting device…       [***************************************]
Waiting…                 [***************************************]

An update requires a reboot to complete. Restart now? [y|N]: n

An update requires a reboot to complete. Restart now? [y|N]: n [sic, i.e., I really did get this prompt twice.]
```

ChiWei-Chen, 2022-08-31:
@LinuxOnTheDesktop , thank you for sharing the information, I'll give you back if any feedback from the FW team. (LO-1990 for my internal tracking)

LinuxOnTheDesktop, 2022-10-05:
@ChiWei-Chen 

Please note that the update in question is from _March 2020_ (_sic_) and is given the urgency level of . . 'high'.

ChiWei-Chen, 2022-10-05:
Hi @LinuxOnTheDesktop thanks, I know the FW team is working on fixing this by re-upload the new FW with metadata, I will keep you posted if any updates from the team.

ChiWei-Chen, 2022-11-08:
@LinuxOnTheDesktop , sorry for the late update.
I got feedback from the FW team, it looks like this machine is a development sample that can only upgrade to the 71.16 version.
It should only allow the shipping machine upgrade to 0.73.20 from 0.73.04, they are editing metadata to restrict this.
Please send me a mail(cchen50 [at] lenovo [dot] com), I would send you the 71.16 cap firmware once it's available.

Thank you very much.