fwupd/firmware-lenovo #216: P15v gen 2 / Intel Management Engine firmware version detected is 0

ssilenzi, 2022-04-16:
This issue follows the previous discussion with @mrhpearson, in https://github.com/fwupd/firmware-lenovo/issues/208#issuecomment-1098319861 and https://github.com/fwupd/firmware-lenovo/issues/208#issuecomment-1099616770.

### Brief description

I have a dual boot computer (Windows / Ubuntu).
I updated the Intel ME Firmware from Windows and everything is ok.
The firmware version shown in the BIOS is correct.
Instead Ubuntu detects **version 0**.

### Configuration

Computer:
Lenovo Thinkpad P15v gen 2

Current version of Intel ME Firmware:
15.0.35.1951 Corporate
installed from Windows 11 by the Lenovo Utility

OS:
Ubuntu 20.04 (in dual boot with Windows 11)
Supervisor password set
Secure boot enabled
Disk encrypted by LUKS

### Screenshot from BIOS

[image]

### ESRT table:
`sudo grep -r . /sys/firmware/efi/esrt/entries/*`

> /sys/firmware/efi/esrt/entries/entry0/fw_class:28155bd3-6555-4913-830d-c4533ae85e92
> /sys/firmware/efi/esrt/entries/entry0/lowest_supported_fw_version:0
> /sys/firmware/efi/esrt/entries/entry0/last_attempt_version:0
> /sys/firmware/efi/esrt/entries/entry0/last_attempt_status:0
> /sys/firmware/efi/esrt/entries/entry0/fw_type:1
> /sys/firmware/efi/esrt/entries/entry0/fw_version:65549
> /sys/firmware/efi/esrt/entries/entry0/capsule_flags:0x0
> 
> **/sys/firmware/efi/esrt/entries/entry1/fw_class:666705ce-0624-4943-8738-09829a26c8ee**
> **/sys/firmware/efi/esrt/entries/entry1/lowest_supported_fw_version:0**
> **/sys/firmware/efi/esrt/entries/entry1/last_attempt_version:0**
> **/sys/firmware/efi/esrt/entries/entry1/last_attempt_status:5**
> **/sys/firmware/efi/esrt/entries/entry1/fw_type:2**
> **/sys/firmware/efi/esrt/entries/entry1/fw_version:0**
> **/sys/firmware/efi/esrt/entries/entry1/capsule_flags:0x8010**
> 
> /sys/firmware/efi/esrt/entries/entry2/fw_class:2f45f824-5964-4cf7-a20e-6b2a1e3e69f0
> /sys/firmware/efi/esrt/entries/entry2/lowest_supported_fw_version:65547
> /sys/firmware/efi/esrt/entries/entry2/last_attempt_version:0
> /sys/firmware/efi/esrt/entries/entry2/last_attempt_status:0
> /sys/firmware/efi/esrt/entries/entry2/fw_type:2
> /sys/firmware/efi/esrt/entries/entry2/fw_version:65547
> /sys/firmware/efi/esrt/entries/entry2/capsule_flags:0x0
> 
> /sys/firmware/efi/esrt/entries/entry3/fw_class:d60f4073-924e-4a20-8a7c-36580a61c6b3
> /sys/firmware/efi/esrt/entries/entry3/lowest_supported_fw_version:1
> /sys/firmware/efi/esrt/entries/entry3/last_attempt_version:0
> /sys/firmware/efi/esrt/entries/entry3/last_attempt_status:0
> /sys/firmware/efi/esrt/entries/entry3/fw_type:2
> /sys/firmware/efi/esrt/entries/entry3/fw_version:16973834
> /sys/firmware/efi/esrt/entries/entry3/capsule_flags:0x8010
> 
> /sys/firmware/efi/esrt/entries/entry4/fw_class:e01209cf-ace0-4647-b3c6-790ca0e26a8c
> /sys/firmware/efi/esrt/entries/entry4/lowest_supported_fw_version:1
> /sys/firmware/efi/esrt/entries/entry4/last_attempt_version:0
> /sys/firmware/efi/esrt/entries/entry4/last_attempt_status:0
> /sys/firmware/efi/esrt/entries/entry4/fw_type:2
> /sys/firmware/efi/esrt/entries/entry4/fw_version:0
> /sys/firmware/efi/esrt/entries/entry4/capsule_flags:0x8010
[... 217 lines cut ...]
> │                         • Updatable
> │                         • System requires external power source
> │                         • Needs a reboot after installation
> │                         • Device is usable for the duration of the update
> │   
> ├─USB4 host controller:
> │     Device ID:          a3aea46a5dc6e7931f538ef8eb42d560c0219d42
> │     Summary:            Unmatched performance for high-speed I/O
> │     Update Error:       No vendor ID set
> │     GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
> │     Device Flags:       • Internal device
> │                         • System requires external power source
> │                         • Updatable
> │   
> └─Unifying Receiver:
>       Device ID:          7c6ba6685c28dbcaafaf7d69e00bea41ff638392
>       Summary:            Miniaturised USB wireless receiver
>       Current version:    RQR24.09_B0033
>       Bootloader Version: BOT03.02_B0009
>       Vendor:             HIDRAW:0x046D|USB:0x046D
>       Install Duration:   30 seconds
>       GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
>                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
>                           279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
>       Device Flags:       • Updatable

ssilenzi, 2022-05-06:
May I ask for a feedback regarding this issue? Are there any updates?

mrhpearson, 2022-05-10:
Hi - apologies for the slow update, holidays in Asia and I was off sick.

As a reference for myself - internal ticket is  LO-1721

Waiting for an update from the FW team but my guess looking at the details above is this is a similar issue as reported in the linked issues which are being actively worked.
Not sure why you have version zero, instead of the previous version - that's what I'm waiting for an update from the FW team on.

Functionality-wise I don't think you should have any impacts - it looks like it's just the FW mis-reporting the ME FW version to Linux.

hughsie, 2022-05-11:
> it looks like it's just the FW mis-reporting the ME FW version to Linux.

Mark, one thing I thought (although it's a huge hack) is maybe we should overwrite the ESRT reported value of the ME entry after we've enumerated the AMT interface? Maybe that only works for vpro, but a `fwupdmgr get-devices --show-all` on an affected machine might prove that it's possible.

ssilenzi, 2022-05-11:
> Functionality-wise I don't think you should have any impacts - it looks like it's just the FW mis-reporting the ME FW version to Linux.

@mrhpearson, I think you are right

ssilenzi, 2022-05-11:
> Maybe that only works for vpro, but a `fwupdmgr get-devices --show-all` on an affected machine might prove that it's possible.

@hughsie 
`fwupdmgr get-devices --show-all`

21A90006IX
│
├─11th Gen Intel Core™ i7-11850H @ 2.50GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000003c
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         a6bd4ca5-75a6-5796-b564-66b5cab1b11b ← CPUID\PRO_0&FAM_06&MOD_8D
│                         d9dd5e77-df17-5bab-b5ec-22827598bfed ← CPUID\PRO_0&FAM_06&MOD_8D&STP_1
│     Device Flags:       • Internal device
│   
├─CT2000P5PSSD8:
│     Device ID:          310f45f1f223064b5c16bf6dff31146755a64480
│     Summary:            NVM Express solid state drive
│     Current version:    P7CR402
│     Vendor:             Micron/Crucial Technology (NVME:0xC0A9)
│     Serial Number:      213631650368
│     GUIDs:              c02c9d4b-c3b3-5f73-ad7d-f98d9f643733 ← NVME\VEN_C0A9&DEV_5407&SUBSYS_C0A90100&REV_00
│                         cd36ee17-e270-50f3-9468-7bb0b99b5fc5 ← NVME\VEN_C0A9&DEV_5407&SUBSYS_C0A90100
│                         4dabbbed-5335-5c11-8b19-de91122047e9 ← NVME\VEN_C0A9&DEV_5407&REV_00
│                         5b5825f0-2a84-54df-b60b-8ee09786de93 ← NVME\VEN_C0A9&DEV_5407
│                         827004e7-828b-5c9b-b0f7-ea564d50e76f ← CT2000P5PSSD8
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Display controller:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              d0daff4b-4010-5790-9901-fd572936b035 ← PCI\VEN_10DE&DEV_1FBA&SUBSYS_17AA22D9&REV_A1
│                         136f3331-260a-53d8-99e5-4960f268c9a4 ← PCI\VEN_10DE&DEV_1FBA&SUBSYS_17AA22D9
│                         9068af5d-dd9f-5c1b-8aab-3a0a9d06d0c2 ← PCI\VEN_10DE&DEV_1FBA&REV_A1
│                         ac0a29fb-9491-5887-86f6-3d4eb4f96a8c ← PCI\VEN_10DE&DEV_1FBA
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Display controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    01
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              1cae4095-38b9-58a8-be09-6bdadf78d6b8 ← PCI\VEN_8086&DEV_9A60&SUBSYS_17AA22D9&REV_01
│                         65b6195b-72a4-5e58-96ac-dd2f2cc8e7a7 ← PCI\VEN_8086&DEV_9A60&SUBSYS_17AA22D9
│                         91ff7032-68d2-5f2b-aa22-cf9d9a6beab6 ← PCI\VEN_8086&DEV_9A60&REV_01
│                         542e165e-ec6d-59dd-b7c2-60892a04d76a ← PCI\VEN_8086&DEV_9A60
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Integrated Camera:
│     Device ID:          3fa281ddf80d8a06b8ee5d8beb48d38ee95a9627
│     Current version:    60.4
│     Vendor:             Azurewave (USB:0x13D3)
│     Serial Number:      0000
│     GUIDs:              96607700-e5f3-59e1-a572-3987639781b1 ← USB\VID_13D3&PID_56FB&REV_6004
│                         e43e8c39-23e2-55bd-83fb-ccb0264df2d8 ← USB\VID_13D3&PID_56FB
│     Device Flags:       • Updatable
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    0
│     Vendor:             Lenovo (DMI:LENOVO)
[... 191 lines cut ...]
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─USB4 host controller:
│     Device ID:          d1806b0616fc6f13efdc72aa949b02ba4a32d5fe
│     Summary:            Unmatched performance for high-speed I/O
│     Update Error:       No vendor ID set
│     GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Updatable
│   
└─Unifying Receiver:
      Device ID:          7c6ba6685c28dbcaafaf7d69e00bea41ff638392
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR24.09_B0033
      Bootloader Version: BOT03.02_B0009
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:       • Updatable

mrhpearson, 2022-05-11:
@hughsie - they're supposed to have the fix for this in the next round of BIOS's though I don't think we're going to know if it really works until they've rolled out and ME updates have been tried subsequently.
There are a few trial BIOS's that went out to some people but I don't believe we've gotten any feedback yet
I'd rather hold off doing a huge hack in fwupd for what to me looks like an obvious Lenovo FW issue. Just feels like the wrong way forward even though it's taking us a long time to fix this. It is being looked at and not ignored if that helps :)

hughsie, 2022-05-11:
@mrhpearson yes, you're correct of course. I'm just getting dinged about it on a daily basis these days. I'll pass on the good news about the next round of updates.

ssilenzi, 2022-05-24:
Update: the firmware update is also offered on Windows, not just Linux. But it is already installed as already reported.

Also, maybe unrelated to ME but related to FW: on Windows, if the screen is off and no processes are running, the fans start at full speed. Unlocking the screen they stop. Instead Linux is fine.

mrhpearson, 2022-05-25:
Thanks for the update - I've not heard back from the FW team so have sent a nag.

mrhpearson, 2022-05-26:
Hi - I got some feedback from the FW team.

They believe this is a different issue from the ESRT update issues we're seeing on other platforms
They've given me a vPro and non vPro ME updated FW that think think will resolve this to test with if that would be possible for you. If you can drop me a note (markpearson at lenovo dot com) with which one you would need I can get you it to try them out

Thanks
Mark

ssilenzi, 2022-06-02:
My computer is a **vPro** device. Tried the update but **failed**:
```
21A90006IX
│
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    0
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Failed
│     Update Error:       failed to update to 0: authentication signing error
│     GUIDs:              666705ce-0624-4943-8738-09829a26c8ee
│                         d6163bf5-ce22-5650-a944-51e6421118e5 ← UEFI\RES_{666705CE-0624-4943-8738-09829A26C8EE}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update

```
Steps followed:

- touched the flag OnlyTrusted=false inside /etc/fwupd/daemon.conf
- launched the command
`sudo fwupdmgr install Lenovo***.cab 349bb341230b1a86e5effe7dfe4337e1590227bd`
- the computer reboots and hangs on the Lenovo logo, without starting the update
- after few seconds, it reboots again
- launched the command
`sudo fwupdmgr get-devices`
and got the result above.

I am available to try any possible future firmware upgrade.

Thanks @mrhpearson

ssilenzi, 2022-06-06:
More details: I retried the update with the binaries which @mrhpearson provided to me.

I get this error after reboot:
`Error 354: Update image with lower VCM is not allowed`
`Reset the system`

In fact the Lenovo Tool has installed an update to ME Firmware yesterday. But the problem of version 0 has not been resolved yet.

mrhpearson, 2022-06-06:
Drat...I've forwarded to the FW team and asked for recommendations on next steps.

mrhpearson, 2022-06-07:
Feedback from FW team is:

> If the MEFW version has been update to 15.0.41.214 they need to wait for next BIOS and MEFW version update (Plan in July ) to fix this problem.

I think you're running an older version though (from the above). Can you confirm if you're still on 15.0.35.1951?

ssilenzi, 2022-06-08:
The BIOS displays 15.0.41.2142, while Linux still display 0. If you have another unofficial firmware to test, it will be great.

mrhpearson, 2022-06-08:
OK - so if you're on 2142 you're already at the latest and that explains the VCM error. Guidance from the FW team is please wait for the next update of BIOS and ME which is planned for July.

ssilenzi, 2022-08-22:
The last BIOS update, installed today, solved the problem:

├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    240.41.2142
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUIDs:              8d023cbb-b57d-4c49-bdf4-7c563c36aad6
│                         d044385c-6abc-50f4-9a10-61e71b8e323d ← UEFI\RES_{8D023CBB-B57D-4C49-BDF4-7C563C36AAD6}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update