fwupd/firmware-lenovo #217: P1 Gen 3 cannot update intel management engine

jmontleon, 2022-04-20:
A message flashes briefly `Error 20: Cannot find ME client.` in the upper left corner of the screen for an instant before rebooting after about 1 minute 20 seconds of sitting at `Please wait while we install a system update`

[image]

```
$ sudo fwupdmgr get-updates
Devices with no available firmware updates: 
 • Integrated Camera
 • Thunderbolt host controller
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • MZVLB1T0HBLR-000L7
 • Prometheus
 • Prometheus IOTA Config
 • System Firmware
20TJS53A00
│
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Summary:            UEFI ESRT device
  │   Current version:    224.36.1158
  │   Minimum Version:    0.0.1
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUIDs:              5695cc48-4f4f-4677-8ffb-9f496d3ad9d3
  │                       3ef35d3b-ceeb-5e27-8c0a-ac25f90367ab ← UEFI\RES_{5695CC48-4F4F-4677-8FFB-9F496D3AD9D3}
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  ├─ThinkPad P1 Gen3 X1 Extreme 3rd:
  │     New version:      225.53.1649
  │     Remote ID:        lvfs
  │     Release ID:       9252
  │     Summary:          Lenovo ThinkPad P1 Gen3 X1 Extreme 3rd Corporate ME Firmware
  │     License:          Proprietary
  │     Size:             12.2 MB
  │     Created:          2016-07-08
  │     Urgency:          High
  │     Details:          https://pcsupport.lenovo.com/de/en/search?query=N2VRM12W
  │     Vendor:           Lenovo
  │     Release Flags:    • Is upgrade
  │     Description:      
  │     Lenovo ThinkPad P1 Gen3 X1 Extreme 3rd Corporate ME Firmware
  │     
  │     Version : 14.1.53.1649 (LVFS : 225.53.1649)
  │     
  │     Problem fixes
  │     
  │     • Several Intel known issues were fixed in this release. Refer to Intel.
  │     • Closed Issues at the bottom portion of Intel's release note.
  │   
  ├─ThinkPad P1 Gen3 X1 Extreme 3rd:
  │     New version:      224.48.1605
  │     Remote ID:        lvfs
  │     Release ID:       8307
  │     Summary:          Lenovo ThinkPad P1 Gen3 X1 Extreme 3rd Corporate ME Firmware
  │     License:          Proprietary
  │     Size:             12.2 MB
  │     Created:          2016-07-08
  │     Urgency:          High
  │     Details:          https://pcsupport.lenovo.com/de/en/search?query=N2VRM11W
  │     Vendor:           Lenovo
  │     Release Flags:    • Is upgrade
  │     Description:      
  │     Lenovo ThinkPad P1 Gen3 X1 Extreme 3rd Corporate ME Firmware Version 14.0.48.1605 (LVFS:224.48.1605)
  │   
  └─ThinkPad P1 Gen3 X1 Extreme 3rd:
        New version:      224.45.1389
        Remote ID:        lvfs
        Release ID:       7266
        Summary:          Lenovo ThinkPad P1 Gen3 X1 Extreme 3rd Corporate ME Firmware
        License:          Proprietary
        Size:             12.2 MB
        Created:          2020-10-15
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N2VRM10W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        • 0 Maintenance Release
        
        Version 14.0.45.1389 (LVFS:224.45.1389)
```

jmontleon, 2022-04-20:
fwupd was installed from Fedora packages:
```
$ rpm -qa | grep fwupd | sort -u
fwupd-1.7.6-1.fc36.x86_64
fwupd-efi-1.3-1.fc37.x86_64
fwupd-plugin-flashrom-1.7.6-1.fc36.x86_64
fwupd-plugin-modem-manager-1.7.6-1.fc36.x86_64
fwupd-plugin-uefi-capsule-data-1.7.6-1.fc36.x86_64
```

After attempting to update I verified the firmware is not updated in the system setup.

jmontleon, 2022-04-20:
For troubleshooting I also disabled Secure Boot and after find the error message in lns code for AMT I also tried turning on AMT with no change in behavior when trying to update.

https://github.com/intel/lms/blob/master/UNS/PartialFWUpdateService/errorlist.h#L119

mrhpearson, 2022-04-20:
Hi @jmontleon - can you go into the BIOS setup screen (F1 during early boot) and check the ME version there please?
Thanks
Mark

jmontleon, 2022-04-20:
@mrhpearson thanks for the quick response. The BIOS setup screen says `ME Firmware Version    14.0.36.1158`.

I saw issues filed for other laptops, but nothing with the error message I encountered. It's fine if you'd like to add this to one of the existing issues if one is appropriate.

jmontleon, 2022-04-22:
I ended up using rufus to create a bootable windows thumb drive to run the windows firmware update executable and it worked fine. I'll give fwupdmgr another try when there is a new update available.

halfsquatch, 2022-07-14:
I have ths aem issue. I am hoping not to have to create a a Windows thumb drive. Is there any information I can provide to help troubleshoot this?

jmontleon, 2022-07-23:
I received an email for and saw in Lenovo Vantage on a Windows system a new update for the IME firmware.
https://download.lenovo.com/pccbbs/mobiles/n2vrg06w.txt

I have not seen the update available via fwupd yet, but I'll try to keep an eye out and if/when it is I'll give it a try and see if it works any better coming from an updated version.

jmontleon, 2022-08-05:
I saw the update this morning:
```
LENOVO 20TJS53A00
│
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Summary:            UEFI ESRT device
  │   Current version:    225.60.1807
  │   Minimum Version:    0.0.1
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUID:               5695cc48-4f4f-4677-8ffb-9f496d3ad9d3
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad P1 Gen3 X1 Extreme 3rd Corporate:
        New version:      225.65.1969
        Remote ID:        lvfs
        Release ID:       14698
        Summary:          Lenovo ThinkPad P1 Gen3 X1 Extreme 3rd Corporate ME Firmware
        License:          Proprietary
        Size:             11.0 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N2VRM14W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Lenovo ThinkPad P1 Gen3 X1 Extreme 3rd Corporate ME Firmware Version : 14.1.65.1969 (LVFS : 225.65.1969)
        
        The computer will be restarted automatically after updating MEFW completely . Do NOT turn off your computer or remove the AC adapter while update is in progress.
```

And it updated normally, without issue:
```
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    225.65.1969
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               5695cc48-4f4f-4677-8ffb-9f496d3ad9d3
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
```

kmauleon, 2022-11-16:
Closing as already fixed.  please reopen if needed.  thank you very much.

closed 2022-11-16 by kmauleon