fwupd/firmware-lenovo #218: UEFI 1.15 not offered by fwupdmgr for ThinkPad X1 Extreme Gen 4

ginkel, 2022-04-24:
Hi there,

despite being listed [1] on LVFS, the 1.15 firmware update does not seem to be offered by `fwupdmgr update` for my X1E4.

```
$ sudo fwupdmgr update 
Devices with no available firmware updates: 
 • Prometheus IOTA Config
 • Qualcomm Snapdragon X55 5G
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • MZVL22T0HBLB-00BL7
 • Prometheus
 • System Firmware
 • Unifying Receiver
```

Any ideas?

[1] https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN40ET.firmware

--

```
$ sudo fwupdmgr get-devices

20Y5CT01WW
│
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.15
│     Minimum Version:    0.1.15
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               ab5a2de4-75ef-4906-9533-143d2acad54e
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─MZVL22T0HBLB-00BL7:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express solid state drive
│     Current version:    6L2QGXB7
│     Vendor:             Samsung (NVME:0x144D)
│     Serial Number:      S64SNF0R314779
│     GUIDs:              4d7a2791-106b-5e72-9cfb-8ea3d89f5421 ← NVME\VEN_144D&DEV_A80A&SUBSYS_144DA801&REV_00
│                         310f81b5-6fce-501e-acfb-487d10501e78 ← NVME\VEN_144D&DEV_A80A&SUBSYS_144DA801
│                         60c89aac-f321-515b-b419-3cf02aa9d375 ← NVME\VEN_144D&DEV_A80A&REV_00
│                         bec63ed7-a95f-54fe-b8cc-8e9fee64ba5a ← NVME\VEN_144D&DEV_A80A
│                         75877a5d-2070-57b7-ba80-feb6c937df57 ← SAMSUNG MZVL22T0HBLB-00BL7
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─Prometheus:
[... 144 lines cut ...]
│     Summary:            UEFI ESRT device
│     Current version:    24596
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               7c8c8f35-13d3-5271-5e44-5a045511746b
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:          3801e1f7043b9abdf1f873bf0bf0b945402e4238
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR24.11_B0036
      Bootloader Version: BOT03.02_B0009
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:       • Updatable
                          • Supported on remote server
                          • Signed Payload
```

mrhpearson, 2022-04-25:
That's strange - it should work - the GUID matches. Looks like it's had over 1000 downloads too.

Are you plugged in, with battery > 50% and the lid open? 

I just did the update on my P1G4 as I hadn't done it yet...it all went smoothly.

mrhpearson, 2022-04-25:
Not sure what you mean by 'in contrast to the firmware versions currently offered on lenovo.com' - do they not work? If the issue is FCC unlock related I can't unfortunately help you with that but I hear the modem manager forum has some useful tips.

I'm afraid I also don't have WWAN on any of my systems so limited experience with it. I think that's the first FW update we've offered on LVFS (finally - woohoo!). If it helps - it has 87 downloads with 5 success reports and 1 failure report (which was due to "failed to detach using modem_manager"...not sure how bad that is)

Can we close this bug?

Mark

hughsie, 2022-04-26:
> Can we close this bug?

I think so. Thanks Mark!

closed 2022-04-26 by hughsie

ginkel, 2022-04-26:
> Can we close this bug?

Yes, thank you!