fwupd/firmware-lenovo #219: [Qualcomm Snapdragon X55 5G] since firmware update -> not working anymore

j5k, 2022-04-26:
Since I made a firmware update from [1] to [2] the modem can't establish a connection anymore. For Windows exists a newer version [3]. Can you put this on LVFS and is the problem with this version solved?

[1] T99W175.F0.0.0.5.7.SC.004.076 (old)
[2] T99W175.F0.1.0.0.9.SC.005.067 (current)
[3] T99W175.F0.1.0.0.9.SC.005.075 (windows)

ginkel, 2022-04-27:
I can confirm that the update to T99W175.F0.1.0.0.9.SC.005.067 effectively soft-bricks the device as ModemManager's FCC unlock no longer works.

@j5k Are you sure that T99W175.F0.1.0.0.9.SC.005.075 solves the issue? I previously had to downgrade to T99W175.F0.1.0.0.9.DT.003.055 to get it working.

muelllu9, 2022-05-03:
I can also confirm that the update to T99W175.F0.1.0.0.9.SC.005.067 has broken the FFC Unlock, and is now effectively soft-bricked.

hughsie, 2022-05-03:
@mrhpearson should we pull this one back from stable?

ginkel, 2022-05-03:
> I'm checking with the WWAN team what happened here - but it's a holiday in Japan this week so won't know until next week the answer. I did ask last week and the FCC unlock mechanism wasn't supposed to have changed - but it certainly seems to have so I've been surprised by what has happened

My hypothesis is: The FCC unlock mechanism did not change from the (Windows) firmware published at pcsupport.lenovo.com. So the statement by the firmware team is most likely factually correct... It did, however, change from the v48 that is no longer available officially.

@mrhpearson Thanks for keeping us updated! While this is pretty frustrating from an end-user perspective, it must be even more so from your end...

j5k, 2022-05-04:
@ginkel No, I was not sure that T99W175.F0.1.0.0.9.SC.005.075 will solve the issue. I was asking about it.

But today I installed Windows on a second SSD to test this (newest) version. Same issue.

@mrhpearson If I can help in some way, please let me know.

mrhpearson, 2022-05-04:
Thanks, Waiting for feedback from the WWAN team on the next steps. We have pulled the FW back to testing in the meantime.
Will update ASAP

ginkel, 2022-05-30:
We made some progress adding support for the new unlock mechanism to ModemManager / libqmi (thanks, Aleksander and Bjørn!).

Would anyone volunteer to help testing the fix before it gets merged into ModemManager? You need to compile and install the development version of libqmi from https://gitlab.freedesktop.org/mobile-broadband/libqmi and replace the fcc-unlock script in ModemManager with the version from https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/806...

j5k, 2022-05-30:
This week I'm very busy, but on the weekend I maybe find some time. Thanks a lot for your effort!

j5k, 2022-05-30:
Ok, my wonder has won ;-) It works!! But I can nothing say about the stability. I will test this the next few days...

Thanks!

j5k, 2022-05-30:
```
May 30 23:42:34 testpc systemd[1]: Starting Modem Manager...
May 30 23:42:34 testpc ModemManager[485982]: <info>  ModemManager (version 1.18.8-1) starting in system bus...
May 30 23:42:34 testpc ModemManager[485982]: [qrtr] socket lookup from 1:0
May 30 23:42:34 testpc ModemManager[485982]: [qrtr] initial lookup finished
May 30 23:42:34 testpc systemd[1]: Started Modem Manager.
May 30 23:42:34 testpc audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=ModemManager comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
May 30 23:42:34 testpc kernel: audit: type=1130 audit(1653946954.228:2231): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=ModemManager comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
May 30 23:42:34 testpc NetworkManager[3251]: <info>  [1653946954.2516] modem-manager: ModemManager now available
May 30 23:42:35 testpc ModemManager[485982]: opening device...
May 30 23:42:35 testpc ModemManager[485982]: cannot connect to proxy: Could not connect: Connection refused
May 30 23:42:35 testpc ModemManager[485982]: spawning new mbim-proxy (try 1)...
May 30 23:42:35 testpc ModemManager[485982]: [/dev/wwan0mbim0] Couldn't find descriptors file, possibly not using cdc_mbim
May 30 23:42:35 testpc ModemManager[485982]: [/dev/wwan0mbim0] Fallback to default max control message size: 4096
May 30 23:42:36 testpc ModemManager[485982]: <info>  [wwan0mbim0/mbim] MBIM device is not QMI capable
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0] channel destroyed
May 30 23:42:36 testpc ModemManager[485982]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:0d.0/usb2/2-3/2-3.4/2-3.4.4/2-3.4.4.3': not supported by any plugin
May 30 23:42:36 testpc ModemManager[485982]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:14.3': not supported by any plugin
May 30 23:42:36 testpc ModemManager[485982]: <info>  [device /sys/devices/pci0000:00/0000:00:1c.0/0000:08:00.0] creating modem with plugin 'foxconn' and '4' ports
May 30 23:42:36 testpc ModemManager[485982]: <info>  [base-manager] modem for device '/sys/devices/pci0000:00/0000:00:1c.0/0000:08:00.0' successfully created
May 30 23:42:36 testpc ModemManager[485982]: opening device...
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0] Couldn't find descriptors file, possibly not using cdc_mbim
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0] Fallback to default max control message size: 4096
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0] Opening device with flags 'version-info, proxy, mbim, expect-indications'...
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0] created endpoint
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0] creating MBIM device...
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0] MBIM device created
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0] opening MBIM device...
May 30 23:42:36 testpc ModemManager[485982]: opening device...
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0] Couldn't find descriptors file, possibly not using cdc_mbim
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0] Fallback to default max control message size: 4096
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0] MBIM device open
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0] Checking version info (15 retries)...
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0] QMI Device supports 35 services:
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    ctl (1.5)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    wds (1.193)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    dms (1.79)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    nas (1.25)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    qos (1.18)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    wms (1.10)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    auth (1.14)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    at (1.6)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    voice (2.1)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    cat2 (2.24)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    uim (1.77)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    pbm (1.4)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    test (1.0)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    loc (2.131)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    sar (1.0)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    ims (1.91)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    ts (1.0)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    tmd (1.0)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    wda (1.24)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    csvt (1.6)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    imsa (1.44)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    coex (1.0)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    pdc (1.0)
May 30 23:42:36 testpc ModemManager[485982]: [/dev/wwan0mbim0]    rfrpe (1.0)
[... cut ...]
May 30 23:42:47 testpc ModemManager[485982]: <info>  [modem0] simple connect state (6/8): bearer
May 30 23:42:47 testpc ModemManager[485982]: <info>  [modem0] simple connect state (7/8): connect
May 30 23:42:47 testpc ModemManager[485982]: <info>  [modem0] state changed (registered -> connecting)
May 30 23:42:47 testpc ModemManager[485982]: <info>  [modem0] state changed (connecting -> connected)
May 30 23:42:47 testpc ModemManager[485982]: <info>  [modem0] simple connect state (8/8): all done
```

ginkel, 2022-05-30:
> Ok, my wonder has won ;-) It works!! But I can nothing say about the stability. I will test this the next few days...

Excellent, thanks for your feedback! :-)

cowboyrushforth, 2022-05-30:
Works here!  This is what windows installed

T99W175.F0.1.0.0.9.TO.006.079

davide125, 2022-07-08:
As a datapoint, I'm on T99W175.F0.0.0.5.7.VZ.006 on an X1 Nano with Fedora 36 and it's failing with 'Invalid transition' even after installing the latest libqmi and the fcc-unlock mentioned above.

plumlis, 2022-09-20:
Fedota 37 beta with x55 modem, Still  failing with 'Invalid transition'

xapienz, 2022-11-07:
@plumlis Please check https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/806, newer version of ModemManager is needed for "Invalid transition" fix, which is not in Fedora 37 yet

murad0522, 2023-01-27:
could anyone plz confirm if this will work in my desktop? or only lenovo? thank you