fwupd/firmware-lenovo #221: Thinkpad T14 Gen1 AMD specified firmware is older than installed

j13m126, 2022-05-02:
I noticed that there is a new firmware on lvfs that doesn't show up in gnome-software and also in terminal.
when I download the cab file and run fwupdmgr install fe0c..71w.cab I get:

**Specified firmware is older than installed '0.1.40 < 65573**'

downgrading is also not an option: **No releases found**

history: 
I downgraded to firmware version 1.37 from the build in testing branch which was pulled.

Is there a problem on my part or is this a bug?

I tried fwupd version 1.7.6. 1.7.7 and 1.8.0 on fedora 36
Thanks!

kpfleming, 2022-05-02:
I had a similar problem on a P15s Gen2i which was resolved by upgrading to 1.7.7 (Debian Bookworm packages). Odd that you have the same issue but using newer fwupd didn't resolve it.

mrhpearson, 2022-05-02:
That is strange. I've got Fedora35 with version 1.7.7 on my T14 AMD and it updated correctly on my system.

I checked and the FW here: 
https://fwupd.org/downloads/fe0c3dfba0a6f032fa39d59d1bb8b2191166bf385f7232c2dede3dcf83916a5d-r1bul71w.cab

and it has the version format set in the metainfo correctly - which is normally the problem.

Can you try 'fwupdmgr refresh' to update the metainfo fwupdmgr is using (the files in /var/lib/fwupd/remotes.d/lvfs). Is it possible they're old and out of date?

Can you also try just doing a 'fwupdmgr update' instead of installing the cab file manually? Not sure why that would make a difference...grasping at straws a bit.

Mark

mrhpearson, 2022-05-02:
Do just check with 'fwupdmgr get-devices' that the update isn't blocked for other reasons (low battery is the main one; but I've also seen 'lid closed' and there may be others)

mrhpearson, 2022-05-02:
I'm stumped - that all looks correct - except the version number is shown as a string and not a triplet (on my system it's shown correctly).

Want to try forcing it? Create /usr/share/fwupd/quirks.d/lenovo-triplet.quirk with this as the contents:
```
[01453b71-da0c-4832-9f4f-e378245339c7]
VersionFormat = triplet
```

And (you might have to do a fwupdmgr refresh first) see if that changes the Current version display for the BIOS to be a triplet format. It's not fixing the root cause though... 

Hoping somebody more competent that me can comment as my bag of tricks is empty after this.

Mark

j13m126, 2022-05-02:
sure, I had to restart to see the triple though.
```
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.37
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              01453b71-da0c-4832-9f4f-e378245339c7
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update

```

mrhpearson, 2022-05-02:
OK - and the update should now be offered correctly now right?

Not sure why fwupd isn't getting this itself though - @hughsie, any suggestions?

Mark

j13m126, 2022-05-02:
no, no update. Could it be that the leading 0 in the version number is the problem? shouldn't it say 1.37?

mrhpearson, 2022-05-02:
No - that should be fine (it's what I see on my system).
And if you do the manual install of the cab file now?

j13m126, 2022-05-02:
yes, installing it manually worked!! :)
```
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.40
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              01453b71-da0c-4832-9f4f-e378245339c7
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 

```
thanks Mark!! :+1:

mrhpearson, 2022-05-02:
Glad we got the install done - but I think there's something wonky with your fwupd install or state of metadata which I'd love to understand. Looking at the above you're going to hit the same problem with the EC and ME FW updates.

j13m126, 2022-05-02:
do you mean the UEFI Devices? Should I create a quirk file for each UEFI Device with the GUID in brackets?

mrhpearson, 2022-05-02:
I wouldn't. Let's wait and see if someone smarter than me can figure out what is wrong with your system and why it's not picking up the update and metaversion automatically - it should work.

dimisjim, 2022-08-13:
Stumbled upon this issue with a T16 too, while trying to upgrade to a newer UEFI firmware which was available via the lenovo support website, but was not picked up via the `fwupdmng` utility in ubuntu 22.04 (running v1.7.5). Doing the quirk thingy @mrhpearson suggested worked! Thanks!

(I also had to reboot, just refreshing didn't help)