fwupd/firmware-lenovo #224: Intel ME 192.85.1869 fails to install on my P52

rugubara, 2022-05-07:
```
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Previous version:   192.70.1652
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2022-05-07 20:17
  │   GUID:               971682af-21a0-46e3-91c0-825702c2ba70
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Reported to remote server
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad P52/P72:
        New version:      192.85.1869
        Remote ID:        lvfs
        Release ID:       11618
        Summary:          Lenovo ThinkPad P52/P72 Corporate ME Firmware
        License:          Proprietary
        Size:             12,2 MB
        Created:          2021-11-10
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N2CRG30W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:
        • 0 Intel Platform Update

        Version 12.0.85.1869 (LVFS: 192.85.1869)
```

I can't get past this update. Other updates are not offered for installation.

mrhpearson, 2022-05-10:
Can you check in the BIOS setup screen (F1 during early boot) which ME version is installed please?

There is a known issue being worked on where the version reported to linux is incorrect - so you may be trying an update that is already done (so gets reported as a failure)

dicer, 2022-05-30:
I see that error on a P72 as well. I can confirm that the update seems to install fine, because the ME version in my bios is indeed 12.0.85.1869. So I'll just ignore that it keeps showing up for now.

VladimirJelincic, 2022-07-20:
Same here on my P53s, it tries to continue install after reboot but fails.

VladimirJelincic, 2022-07-20:
Oddly enough, my bios displays version: 12.0.90.2072
Is there any way to remove this update out of the "Software Update" que?

dicer, 2022-07-20:
@agurenko Yeah, like I said above, it is already installed. Just a problem of the update which keeps showing up.

@mrhpearson Could you point me to where the reporting issue is being discussed/worked on? Maybe I can help.

mrhpearson, 2022-07-26:
@ChiWei-Chen - sorry, I'm getting lost with all the different ME update threads so tagging you on this one.

Do you have the details on the P52 fixes for the ME update issue and when they will be rolled out? (and there's a P53 and P72 user on this thread too)

ChiWei-Chen, 2022-07-27:
@mrhpearson , thanks this seems the same as #140. It should be fix with new BIOS, I'll check with BIOS team.
(LO-1884 for internal tracking)

ChiWei-Chen, 2022-08-02:
Hi, BIOS team confirmed this will be fixed in new BIOS release, it's under internal process, expected to release a few weeks later.
Thanks!

agurenko, 2022-08-02:
@ChiWei-Chen does it require new BIOS for all affected models? (I'm having similar issues on P1 gen2)

ChiWei-Chen, 2022-08-04:
@agurenko thanks for reporting the issue, could you provide more detailed information?
Please raise a separate ticket for P1 Gen 2 since we need to address it with separate FW team. Thanks!

- fwupdmgr update --show-all
- system firmware version and Intel Management Engine firmware version in BIOS main page
- OS and fwupdmgr version

rugubara, 2022-08-05:
> Can you check in the BIOS setup screen (F1 during early boot) which ME version is installed please?
> 
> There is a known issue being worked on where the version reported to linux is incorrect - so you may be trying an update that is already done (so gets reported as a failure)

I checked the version in my BIOS - 12.0.25.1869 - so the update has happened, but the version is incorrectly reported to linux.

ChiWei-Chen, 2022-08-10:
@rugubara , thanks for the information, please wait for the next BIOS release. :)

kmauleon, 2022-11-16:
Closing this as BIOS ESRT fix has been rolled out already.  Please reopen if you still see the problem.  Thank you very much.

closed 2022-11-16 by kmauleon