fwupd/firmware-lenovo #226: Failed ME Firmware on Lenovo T14s Gen 2

vwalek, 2022-05-17:
I recently had unsuccessful firmware update from my Fedora box on Lenovo T14s Gen2.

The update didn't went through and now whenever I reboot the system, I get into the attempt update which always fails.
Showing messages:

" Reading ME Firmware...Please do not power off! 18% Complete"
" Failed to recover ME Firmware..."

Eventually it fails around 60% of update.

Here is some info:

```
[root@fedora fw]# fwupdmgr get-updates
Devices with no available firmware updates: 
 • ELAN0676:00 04F3:3195
 • USB2.0 Hub
 • USB2.0 Hub
 • USB3.1 Hub
 • USB3.1 Hub
 • Integrated Camera
 • MSP430
 • Prometheus IOTA Config
 • ThinkPad Thunderbolt 3 Dock
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
 • Unifying Receiver
Devices with the latest available firmware version:
 • Intel Management Engine
 • Micron <>
 • Prometheus
20WNS1F63S
│
├─Embedded Controller:
│ │   Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.35
│ │   Minimum Version:    0.1.35
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUID:               e6c0d74d-98a2-489b-920d-f8e8c9418cb3
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T14s Gen 2 / X13 Gen 2:
│       New version:      0.1.38
│       Remote ID:        lvfs
│       Release ID:       12076
│       Summary:          Lenovo ThinkPad T14s Gen 2 / X13 Gen 2 Embedded Controller Firmware
│       License:          Proprietary
│       Size:             767.3 kB
│       Created:          2021-05-27
│       Vendor:           Lenovo
│       Release Flags:    • Is upgrade
│       Description:      
│       Embedded Controller Firmware N35HT38W version 1.38. The computer will be restarted automatically after updating EC completely. Do NOT turn off your computer or remove the AC adapter while the update is in progress
│     
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Summary:            UEFI ESRT device
  │   Current version:    0.1.41
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUIDs:              363864c7-1a44-4247-ae16-f2315d3855ea
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  ├─ThinkPad T14s Gen 2i / X13 Gen 2i:
  │     New version:      0.1.44
  │     Remote ID:        lvfs
  │     Release ID:       12405
  │     Summary:          Lenovo ThinkPad T14s Gen 2i / X13 Gen 2i System Firmware
  │     License:          Proprietary
  │     Size:             18.3 MB
  │     Created:          2021-05-11
  │     Vendor:           Lenovo
  │     Release Flags:    • Is upgrade
  │     Description:      
  │     Lenovo ThinkPad T14s Gen2 /ThinkPad X13 Gen2 System Firmware Version 1.44
  │     
  │     The computer will be restarted automatically after updating BIOS completely . Do NOT turn off your computer or remove the AC adaptor while update is in progress.
  │     
  │     • Updated the E-Labeling applcation to version 5.0.0.30.
  │     • Supported 5.15-5.35 GHz band in Indonesia.
  │   
  └─ThinkPad T14s Gen 2i / X13 Gen 2i:
        New version:      0.1.43
        Remote ID:        lvfs
        Release ID:       11577
        Summary:          Lenovo ThinkPad T14s Gen 2i / X13 Gen 2i System Firmware
        License:          Proprietary
        Size:             18.3 MB
        Created:          2021-05-11
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        The computer will be restarted automatically after updating BIOS completely . Do NOT turn off your computer or remove the AC adaptor while update is in progress.
        
        • Update includes a security fix.
        • Changed power LED Sleep mode behavior when system is going to Modern Standby.
        • Fixed an issue where device firmware update via Windows Update failed when OPAL management software is installed.
        • Fixed an issue where unlock message isn't correct after sign out in Windows,after check "Require user to press Ctrl+Alt+Delete".
    
```

```
[root@fedora fw]# fwupdate -i
Information for the update status entry 0:
  Information Version: 7
  Firmware GUID: {363864c7-1a44-4247-ae16-f2315d3855ea}
  Capsule Flags: 0x00000000x
  Hardware Instance: 0
  Update Status: attempt-update
  Capsule File Path: /EFI/fedora/fw/fwupd-363864c7-1a44-4247-ae16-f2315d3855ea.cap

Information for the update status entry 2:
  Information Version: 7
  Firmware GUID: {e6c0d74d-98a2-489b-920d-f8e8c9418cb3}
  Capsule Flags: 0x00000000x
  Hardware Instance: 0
  Update Status: attempt-update
  Capsule File Path: /EFI/fedora/fw/fwupd-e6c0d74d-98a2-489b-920d-f8e8c9418cb3.cap
```

```
[root@fedora fw]# ls -l
total 18664
-rwx------. 1 root root 18335664 May 15 12:24 fwupd-363864c7-1a44-4247-ae16-f2315d3855ea.cap
-rwx------. 1 root root   771440 May 15 12:24 fwupd-e6c0d74d-98a2-489b-920d-f8e8c9418cb3.cap
```

Additionally to that, the audio stopped working as well - the internal speakers or audio over hdmi. I would say it could be issue of the SOF drivers.

Any idea how to revert it?

vwalek, 2022-05-17:
@hughsie hey, reading through the issues below, I can say I had exactly the same experience.

https://github.com/fwupd/firmware-lenovo/issues/131
https://github.com/fwupd/firmware-lenovo/issues/129

The screen was black and a lot noise from the fan. Hit the power button to force off, then started again. It didn't start at the beginning, juts glowing power indicator. After couple of tries, it booted back, but now shows Failed to recover the ME...

makaai, 2022-05-19:
Same problem here with a T15 Gen 2 on Ubuntu 22.04. I thought the problem was my encrypted drive setup, and secure boot. But retried with disabled secure boot, and a clean install of Ubuntu without the encrypted drive, and got to the same result. Is there perhaps a way to disable the message at boot?

@vwalek for the audio I solved with #2 here https://www.linuxuprising.com/2018/06/fix-no-sound-dummy-output-issue-in.html

vwalek, 2022-05-19:
> Same problem here with a T15 Gen 2 on Ubuntu 22.04. I thought the problem was my encrypted drive setup, and secure boot. But retried with disabled secure boot, and a clean install of Ubuntu without [...]

Did you reinstall the system and it went back to normal? Like you are not getting this failed message at the boot?

vwalek, 2022-05-19:
@mrhpearson any news?

makaai, 2022-05-19:
> Did you reinstall the system and it went back to normal? Like you are not getting this failed message at the boot?

Sorry for the confusion. With a fresh Ubuntu install nothing changed for me. I forced the reinstall of the firmware updates but still with the message at startup.

vwalek, 2022-05-19:
@makaai - did you try the bootable BIOS upgrade image? 

https://pcsupport.lenovo.com/ca/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t14s-gen-2-type-20wm-20wn/downloads/driver-list/

makaai, 2022-05-19:
> @makaai - did you try the bootable BIOS upgrade image?
> 
> https://pcsupport.lenovo.com/ca/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t14s-gen-2-type-20wm-20wn/downloads/driver-list/

Not yet (also I'd need the one for the T15 in my case). I was hoping to get at least the message disabled from startup in the meanwhile.

vwalek, 2022-05-19:
@makaai same here :) I am waiting for the team to reply what to do next. Really what would be great is to get rid of the message, or force to reset the attempt for update and just continue.

mrhpearson, 2022-05-24:
_Feedback from FW team:_

Could the fail system successfully boot to linux when have the fail ?
If ok, could customer re-install the MEFW LVfs file by local update, to see if can recover it.

https://pcsupport.lenovo.com/us/zh/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t14s-gen-2-type-20wm-20wn/downloads/driver-list/component?name=%E6%99%B6%E7%89%87%E7%B5%84

After download the files (n35rk01.cab for Corporate system, n35rl01.cab for Consumer system), run below command to local update (for Corporate system):
`fwupdmgr --allow-older --allow-reinstall install n35rk01.cab`

_(Note - easiest way to check which FW you need is probably to stop in the BIOS setup screen (F1 during early boot) and see what version is reported there)_

makaai, 2022-05-24:
Hi Mark, could you share with me the .cab file link for a Lenovo t15 gen 2 type 20W4? Many thanks

vwalek, 2022-05-25:
> (n35rk01.cab for Corporate system, n35rl01.cab for Consumer system),

Hi Mark, will try it. Just quick question, how would you define which is Corporate or Consumer? I just want to verify. Thx

mrhpearson, 2022-05-26:
@makaai - could you send me an email (markpearson at lenovo dot com) and I'll forward the FW files for the T15

@vwalek - this is whether your system is vPro (corporate) or not (consumer). Looks like you figured it out :) Thanks for the note.

vwalek, 2022-05-26:
> @vwalek - this is whether your system is vPro (corporate) or not (consumer). Looks like you figured it out :) Thanks for the note.

Thx @mrhpearson. Yeah, it would be great to add it to the README notes, so people can check which system they have. 
https://download.lenovo.com/pccbbs/mobiles/n35rk01w.txt

vwalek, 2022-05-27:
@mrhpearson Great news! It worked and the message is gone.

Just couple of hints that could be mentioned:
- the file's name is actually `n35rk01w.cab` for Corporate system
- to install the package you need to set `OnlyTrusted=false` in `/etc/fwupd/daemon.conf`
- the installation went ok and asked to reset the system.
- the system rebooted to the message again `Failed to recover ME firmware...`
- in about 10-20 seconds the `fwudp-efi 1.3` message showed on left corner
- nothing else was shown, thinking it didn't work
- after around 5-6 minutes another line poped up `Resetting system`
- after that, the system booted normally, fixing also the audio issues.

Thank you for fixing the issue! @mrhpearson++

mrhpearson, 2022-05-30:
Awesome - that's good news. That 5-6 minute wait completely sucks when that happens with nothing to show progress.

Glad you're up and running again. I'll let the FW team know

makaai, 2022-06-02:
Hi @mrhpearson any news oh the T15 files? Thanks

mrhpearson, 2022-06-06:
No update - I've sent another request for them

mrhpearson, 2022-06-07:
Hi @makaai - I now have the T15 files. If you can email me (markpearson at lenovo dot com) I'll send you the link to them
Thanks
Mark

makaai, 2022-06-09:
Thanks @mrhpearson that fixed it for me too!

pkese, 2022-07-13:
I'm having the same experience as https://github.com/fwupd/firmware-lenovo/issues/226#issuecomment-1128272230 (didn't boot, forced power-off, now "Failed to recover ME Firmware" and re-upgrading firmware doesn't help).

Is there a set of instructions to be followed to get this working?

```
# dmidecode | grep SKU
	SKU Number: LENOVO_MT_20W0_BU_Think_FM_ThinkPad T14 Gen 2i
```
... on Ubuntu 22.04.

I'm looking into downloading the .cab files.  
How do I know if my notebook is Corporate or Consumer?

ExecuteRestart66, 2022-07-25:
Hey there. new here. I am going to make my own thread on this as I am being plagued by this right now. And find it insane that lenovo hasn't issued a recall on these machines given all the issues- screen flicker, charging completely quits until bios update, and now this- bricking of the laptops from a damn update that Lenovo is pushing out over Lenovo system update. 

I have had this happen to 5 computers in my fleet already. within 2 months of having them...

I was under the impression that this only happens if you update IME first before updating the BIOS. But I have a machine that I was testing this whole flashing of external monitors when on a TB4 dock. I basically updated everything but the IME and GPU. Then today I decided to update the IME- mind you the BIOS was already updated. As soon as i did, now im getting a black screen boot loop. which led me to this post. 

I just tried booting to a linux flash drive, and it will not boot into it. so i dont think this will help me at all. although i will say I have no idea how to find these CAB files. all i get is the exe from the link posted. and when i extract the files from that, there is no CAB files. 

I am almost to the point where I want to sound the alarm and return my entire fleet. Lenovo has been so silent about this and I've spent the last 2 months compiling all this data on these issues. 
1. laptop ceases to charge AT ALL. which then requires you to update the bios. but you cant use the system update tool to do this because that requires power. the EXE does not. however if you get too low battery you run the risk of bricking. 
2. The screen flicker and/or monitors going off/on when you are using 2 monitors on a TB4 or TB3 dock. all firmware and drivers updated. still happens. 
3. this insane bricking of a laptop if you install the IME drivers using system update. and in the notes I now see that it says you should disable bitlocker. that is insanity to have to do such a manual thing in a corporate environment. then worry about re-enabling it and updating your BL codes in AD manually. Not only that but if lenovo is saying disable bitlocker or else it will be bricked, then whyTF are they allowing this to be pushed out using system update? 

sorry for the rant here. but I dont think i have dealt with anything so horrid that almost feels criminal to not recall.

grobz, 2022-07-27:
I am experiencing almost exactly the same scenario described by [vwalek](https://github.com/vwalek) but unfortunately, after downloading the right CAB for my hardware (LENOVO_MT_20W1_BU_Think_FM_ThinkPad T14 Gen 2i) and running the fwupdmgr re-installation, the behavior is still the same. The "Reading ME Firmware...Please do not power off!" and the "Failed to recover ME Firmware..." persists. 

@mrhpearson do you have any other tip I can try? I also tried the bootable USB drive FW installation also. No luck. All my restarts take ages and I can't seem to have the sound enable properly in Linux (I am an experienced user).

@pkese did you manage to solve your issue?

Thanks in advance.

kleops9, 2022-08-01:
Similar issue with my p15 gen2 [#250](https://github.com/fwupd/firmware-lenovo/issues/250)
Same as @grobz re-flashing does not work unfortunately.

mrhpearson, 2022-08-04:
Tagging @ChiWei-Chen for this one. Afraid I've lost track of where we are with the ME FW update issues...

ChiWei-Chen, 2022-08-05:
@grobz , @kleops9 , thanks for reporting this issue. Could you open a separate ticket for T14 Gen2 and P15 Gen2 provide more detailed information? We need to address this with separate FW team. Thanks!

- fwupdmgr update --show-all (if possible)
- system firmware version and Intel Management Engine firmware version in BIOS main page
- OS and fwupdmgr version

tv-nieri, 2022-08-22:
Team, where can i get the files for Lenovo T14 GEN2 Corp?

I can see that Mark shared a procedure that seems to resolve the issue and it was complemented by other amazing people here, but i still did not find this files...

ChiWei-Chen, 2022-08-23:
@tv-nieri , thanks. Could you open a separate ticket for T14 Gen2 and provide more detailed information? We need to address this with different FW team. Thanks!

- fwupdmgr update --show-all (if possible)
- system firmware version and Intel Management Engine firmware version in BIOS main page
- OS and fwupdmgr version

xfilestogo, 2022-08-31:
Same happened to me on Ubuntu 22.04, but I could still but the system.
I solved it by reinstalling the ME Firmware from the terminal.

# fwupdmgr reinstall

Then selected the appropriate number for the ME Firmware from the list.

sufanek1, 2022-10-10:
same happened to me with p17gen2

fwupdmgr reinstall helped

MartinSMadsen, 2023-10-06:
@xfilestogo

> Same happened to me on Ubuntu 22.04, but I could still boot the system. I solved it by reinstalling the ME Firmware from the terminal.
> 
> fwupdmgr reinstall

This did it for me, thanks!