fwupd/firmware-lenovo #230: Lenovo X1 Yoga 4th gen battery firmware issue prevents update of other firmwares.

mmohaveri, 2022-05-24:
# Describe the bug

I've got a X1 Yoga 4th gen with Debian Sid (I installed Debian myself, instead of the stocked Windows 10). Due to what I'm assuming is battery firmware issues with power management software (like TLP & Power-deamon) my battery reports 0% charge. It started by jumping from high battery (between %100 - %70) to %0 and jumping back to the original amount. But after a while it got stock in %0 longer and longer. Currently it has been on %0 for longer than a month.

P.S: I'm using my Laptop mostly plugged in (say %95 of the times its plugged in), although 

For the same reason that has been mentioned in [this](https://github.com/fwupd/firmware-lenovo/issues/211) issue, the %0 battery is preventing me from upgrading any firmware.

mrhpearson, 2022-05-25:
Hi @mmohaveri 
I think your issue is different - the X1Y4 isn't listed here:
https://support.lenovo.com/eg/en/solutions/ht512607

Unfortunately I don't think a FW update is going to fix your battery. If the system is under warranty you could try calling customer support.

mmohaveri, 2022-05-27:
I'm pretty sure that it's a FW issue, two days ago I turned off my laptop and unplugged everything. Now that I turned it on, it started on %70 battery, it shows me that it's not a hardware issue but a software one.

I first got the idea of a firmware issue from [this part of TLP manual](https://linrunner.de/tlp/faq/battery.html#faq-erratic-battery-behavior):
> ### Erratic battery behavior on ThinkPad T420(s)/T520/W520/X220 (and all later models)
> **Symptom**: some users report severely reduced battery capacity or sudden drops of the charge level from around 30% to zero when employing charge thresholds.
> **Probable cause**: conflict with dualmode battery firmware.

Also [here in TPL manual](https://linrunner.de/tlp/faq/battery.html#faq-good-battery-thresholds) it refers to a lenovo employe's comment about ThinkPad dualmode battery firmware:

> Newer ThinkPad models may not need charge thresholds due to dualmode battery firmware – a Lenovo staff member states at [Lenovo Forums [1]](https://forums.lenovo.com/t5/Windows-10/Power-Manager-for-Windows-10/m-p/2129075#M794): “The battery firmware itself will recognize the scenario where the battery is ALWAYS fully charged 100% (over a period of many weeks) and adjust the full charge capacity downwards in a way to maintain maximum battery health. This is something that happens automatically in the battery firmware. There is nothing that a user needs to do manually, to maximize battery health on these batteries. For this reason, we don’t provide any utility to manually manage battery charge thresholds […]”

So I'm pretty sure that it's a firmware issue or a software issue from Linux power management tools that conflicts with firmware's inner workings.

mmohaveri, 2022-06-04:
@mrhpearson Since my last comment, I've been careful to unplug my laptop as soon as it reaches a high battery charge (anything above 80%), and since that day I haven't seen any 0% battery issue! (after about 2 month of having this issue) so I'm 99.99% sure that the issue has something to do with the combination of Linux's power management tools (currently I'm using power-profile-deamons) and battery's firmware.

Should I open another issue for this?

mrhpearson, 2022-06-06:
Hi,

Interesting! I've raised an internal ticket so I can have a discussion with the battery FW team about what is happening. For my reference - LO-1759

Would you mind raising this issue on forums.lenovo.com/linux? It's not really a LVFS issue (not related to FW update other than impacting the update) but I do think it's worth digging into.

(I assume you're able to update your FW by staying at 80% or lower?)

Have you used TLP to set battery threshold limits?

Thanks
Mark

mmohaveri, 2022-06-08:
> Would you mind raising this issue on forums.lenovo.com/linux?
Sure, I just did. Thanks for raising the issue with the battery FW team.

> (I assume you're able to update your FW by staying at 80% or lower?)

Yeah, my issue with FW update has been fixed.

> Have you used TLP to set battery threshold limits?

In the past yes, but I've removed the battery threshold about 5 month ago & removed TLP from my system completely with the release of GNOME 42 (I think about 3 month ago).

closed 2022-06-08 by mmohaveri