fwupd/firmware-lenovo #233: T460p: No new BIOS and Intel ME updates?

uklotzde, 2022-06-08:
- The last BIOS update 2.31 dates back to 2020
- The Intel ME update is from 2021 and outdated according to the Windows counterpart

s-vincent, 2023-02-07:
Any news on this issue ?

At time of writing, there are bios (2.37) and Intel ME  (11.8.93.4323) firmwares updates available on Lenovo website.

kmauleon, 2023-02-08:
hi... owner said will release asap... thank you

kmauleon, 2023-02-08:
owner uploaded the latest as below.... can you please try?  thank you very much.
**please update firmware one at a time BIOS first....**
BIOS >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR07ET.firmware
ME >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR07RM.firmware

uklotzde, 2023-02-09:
Thank you! I can confirm that both updates are offered for installation, thereby resolving this issue.

Will probably install them in a few days. Can't afford bricking my machine now ;)

closed 2023-02-09 by uklotzde

hughsie, 2023-02-09:
> please update firmware one at a time BIOS first....

@kmauleon if you need this behavior you need to depend on `org.freedesktop.fwupd >= 1.7.5` in the metainfo (or add it on the LVFS) as that;'s when we started only doing one update at a time on Lenovo. If the ME requires a specific BIOS version to be installed first then the requirements need to be set to say that, e.g. https://lvfs.readthedocs.io/en/latest/metainfo.html#other-firmware-version

kmauleon, 2023-02-10:
@hughsie i am not the owner of this one... just taking a precaution here just in case.... but thank you will take note of this for my systems

uklotzde, 2023-04-03:
Worked flawlessly. Missing progress feedback during the ME update increases the suspense ;)

kmauleon, 2023-04-03:
thanks for the update @uklotzde glad it worked!