fwupd/firmware-lenovo #234: Battery not charging issue on Thinkpad X1 Carbon Gen9

tweippert, 2022-06-12:
Since yesterday, my Thinkpad X1 Carbon Gen9 doesn't charge anymore. It run's on battery yesterday fine and shuts down after fully uncharged, afterwards it doesn't charge anymore:

```
Battery 0: Discharging, 0%, rate information unavailable
Battery 1: Unknown, 0%
Battery 2: Discharging, 0%, rate information unavailable
```

Maybe i get hit by the Battery Firmware issue? Can't find any direct and also looking on the closed issues, i don't see a direct fix for my model. As the systems says the battery is under 25% (0% currently) i can't install any firmware updates with that requirement from Linux (Manjaro)/fwupdmgr.

Output of `upower -d`:
```
Device: /org/freedesktop/UPower/devices/line_power_AC
  native-path:          AC
  power supply:         yes
  updated:              So 12 Jun 2022 12:40:52 CEST (313 seconds ago)
  has history:          no
  has statistics:       no
  line-power
    warning-level:       none
    online:              yes
    icon-name:          'ac-adapter-symbolic'

Device: /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               Celxpert
  model:                5B10W13974
  serial:               2032
  power supply:         yes
  updated:              So 12 Jun 2022 12:45:53 CEST (12 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               charging
    warning-level:       none
    energy:              0 Wh
    energy-empty:        0 Wh
    energy-full:         52,21 Wh
    energy-full-design:  57 Wh
    energy-rate:         0 W
    voltage:             14,16 V
    charge-cycles:       14
    percentage:          0%
    capacity:            91,5965%
    technology:          lithium-polymer
    icon-name:          'battery-caution-charging-symbolic'

Device: /org/freedesktop/UPower/devices/battery_hidpp_battery_1
  native-path:          hidpp_battery_1
  model:                M570
  serial:               1028-c8-76-99-4a
  power supply:         no
  updated:              So 12 Jun 2022 12:45:53 CEST (12 seconds ago)
  has history:          yes
  has statistics:       yes
  keyboard
    present:             yes
    rechargeable:        yes
    state:               discharging
    warning-level:       none
    percentage:          45%
    icon-name:          'battery-good-symbolic'

Device: /org/freedesktop/UPower/devices/battery_hidpp_battery_0
  native-path:          hidpp_battery_0
  model:                Wireless Keyboard K270
  serial:               4003-fb-f7-c7-17
  power supply:         no
[... 512 lines cut ...]
├─USB4 host controller:
│     Device ID:          d9ab19962a3d06e91ff8176df9c297e62f83bf63
│     Summary:            Unmatched performance for high-speed I/O
│     Update Error:       [31m[1mCannot install update when system battery is not at least 25%[0m
│     GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Signed Payload
│   
└─Unifying Receiver:
      Device ID:          5dceabb952abcbeba20a9efe726666c5aac57c7f
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR12.10_B0032
      Bootloader Version: BOT01.02_B0014
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      Update Error:       [31m[1mCannot install update when system battery is not at least 25%[0m
      GUIDs:              9d131a0c-a606-580f-8eda-80587250b8d6
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
      Device Flags:       • Supported on remote server
                          • Updatable
                          • Unsigned Payload
 
```

tweippert, 2022-06-12:
Found an workaround to install the firmware (https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN32BL.firmware) which fixes charging issue, but i had to remove upower from my system. Without upower fwupd will not check the Battery State and i can install updates without the above update error.

Will check the system for the next day and close the ticket if no further issues occure.

closed 2022-08-10 by tweippert

Jasontoronto, 2022-09-14:
Hi Tweippert,

I got exactly same issue as you did on my X1 Carborn 9.  How did you remove the upower from the system?  can you please share your steps.

Thanks a lot.

t-8ch, 2022-09-14:
@Jasontoronto You can probably just mask the service with `sudo systemctl mask upower.service`

tweippert, 2022-09-17:
> Hi Tweippert,
> 
> I got exactly same issue as you did on my X1 Carborn 9. How did you remove the upower from the system? can you please share your steps.
> 
> Thanks a lot.

Hi Jasontoronto, 
as i write, i removed the upower package from my system and restart fwupd. If fwupd doesn't recognize upower on the system it doesn't check the battery status. Haven't tried with the service masking t-8ch mentioned above, maybe it is sufficient. Afterwards i installed upower package again and all is fine.

HTH!